hoylabs/OpenDTU-OnBattery #1016: Stromzähler Einstellungen Problem

Omermann, 2024-06-01:
### What happened?

Ich habe gestern die neue Version heruntergeladen

### To Reproduce Bug

Die Messung ergibt immer das Gleiche

### Expected Behavior

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

2024.05.07

Omermann, 2024-06-01:
HM1500
shelly3em (nicht Pro Version)

spcqike, 2024-06-01:
Die Einheit im Stromzähler ist wahrscheinlich falsch.

Omermann, 2024-06-01:
Es funktionierte einwandfrei, bis ich die neue Version installierte. Nach der Installation passierte dies.

spcqike, 2024-06-01:
Kann sein. 

Je nachdem, welche Version du hattest, gab es die Einstellung der Einheit bereits oder auch nicht 

aktuell steht halt „mW“ drin auf dem Bild. Damit werden die Leistungen eben als solche interpretiert. 333mW wären dann eben nur 0.33W, gerundet wahrscheinlich 0. 

bzw. was genau meinst du eigentlich mit deinem Problem? „Die Messung ergibt immer das gleiche“? 

OpenDTU misst die Leistung nicht. Sie wird von Shelly abgerufen. 

Was sagt der Shelly denn, wenn du die URL  im Browser aufrufst?

Omermann, 2024-06-01:
„Die Messung ergibt immer das gleiche“?
Success! Power: 333946.81W. Immer gleich kw mW oder Watt.
Shelly es funktioniert ohne Problem.
Nach dem Update gestern hat es nicht funktioniert
[image]

schlimmchen, 2024-06-01:
Aus deinen Screenshots:
[image]

> Success! Power: 333946.81W. Immer gleich kw mW oder Watt.

Bei dieser Ausgabe hattest du wohl kW eingestelt?

Also ich hab auch noch nicht begriffen, was hier das Problem ist...

Bitte kopiere die im PowerMeter eingestellt URL in deinen Browser und zeig uns, was dann zurückkommt.

Omermann, 2024-06-01:
> Aus deinen Screenshots: [...]

[image]
[image]
[image]

Bitte kopiere die im PowerMeter eingestellt URL in deinen Browser und zeig uns.... Wo ist die URL.

Omermann, 2024-06-01:
Wenn es jemanden gibt, der aus der Ferne helfen kann. Ich kann die Passwörter geben. (Teamviever)

spcqike, 2024-06-01:
> Wo ist die URL.

auf deinem 7. Screenshot. 

http://192.168.1.69/emeter/0

Omermann, 2024-06-01:
{"power":-167.06,"pf":-0.96,"current":0.74,"voltage":234.19,"is_valid":true,"total":334168.0,"total_returned":355814.1}

schlimmchen, 2024-06-01:
"total" ist die bezogene Energie (kWh), nicht die aktuelle Wirkleistung. Stelle "power" als JSON Pfad ein, dann wird es passen.

spcqike, 2024-06-01:
Jap. 

Dann ist dein json Pfad falsch. Du fragst „total“ ab, das ist der „Zählerstand“ 

du willst aber „power“

das wird aber vor dem openDTU Update ebenfalls falsch gewesen sein. 

Und immer noch keine Frage:
Warum willst du nur Phase 1 Abfragen?

Omermann, 2024-06-01:
> "total" ist die bezogene Energie (kWh), nicht die aktuelle Wirkleistung. Stelle "power" als JSON Pfad ein, dann wird es passen.

Ich hab kW gemacht 
[image]

Omermann, 2024-06-01:
> Jap.
> 
> Dann ist dein json Pfad falsch. Du fragst „total“ ab, das ist der „Zählerstand“
> 
> du willst aber „power“
> 
> das wird aber vor dem openDTU Update ebenfalls falsch gewesen sein.
> 
> Und immer noch keine Frage: Warum willst du nur Phase 1 Abfragen?

Ich habe für alle die gleichen Einstellungen vorgenommen.

spcqike, 2024-06-01:
> Ich hab kW gemacht

nur dass der Shelly keine kW, sondern W liefert 

> Ich habe für alle die gleichen Einstellungen vorgenommen.

also hast du drei openDTUs und drei wechselrichter? Oder hast du nun alle drei Phasen mit einzelnen URLs hinzugefügt?

Omermann, 2024-06-01:
das wird aber vor dem openDTU Update ebenfalls falsch gewesen sein.
total_power es war Schreibe.
Alles hat ordnungsgemäß funktioniert.
Ich habe schon viele neue Versionen installiert und hatte keine Probleme.

Omermann, 2024-06-01:
also hast du drei openDTUs und drei wechselrichter? Ein openDTU on battery. Noch ein openDTU aber Shelly nicht verbind.Oder hast du nun alle drei Phasen mit einzelnen URLs hinzugefügt? Ja.
[image]
[image]
[image]

Omermann, 2024-06-01:
[image]
Sind diese Einstellungen korrekt? für nulleinspeisung

Omermann, 2024-06-01:
[image]

spcqike, 2024-06-01:
> Sind diese Einstellungen korrekt? für nulleinspeisung

nein. Du fragst drei mal die gleiche Phase ab (emeter/0) 

warum hältst du dich nicht einfach ans Wiki und fragst den gesamten Leistungswert ab? 

http://ip/status 

Und als json Pfad

total_power

?

Omermann, 2024-06-01:
[image]
[image]

alles scheint in Ordnung zu sein

schlimmchen, 2024-06-01:
Danke @spcqike.

closed 2024-06-01 by schlimmchen