hoylabs/OpenDTU-OnBattery #1019: [Request] Using DPL Time based

fellpower, 2024-06-02:
### Is your feature request related to a problem? Please describe.

N/A

### Describe the solution you'd like

Im using a second Solarset (150W) to charge a battery over the day, to have Power for the night. (seperate HM-400)

My Main Solarset is untouched ( 2 x 420W Solar and HM-800)

Im using the unused power of my Mainset to load the battery too, with an AC Charger - automatic, if im not using the power of my mainset in the house. 

So it is possible, to make the DPL timebased? So i can say: "Only start at 10pm - whatever the SoC is - but stop at the SoC Minimum"? If im using the "normal" DPL, the second set will start, if my mainset has no sun - but its to early - i want, that the battery based second solarset only start to generate power at my desired time.

Solarteur, 2024-06-02:
Have you seen this discussion?

https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/586

fellpower, 2024-06-02:
yeah, i have. but i dont want to have more eqippment around, if ODTUoB this feature have