hoylabs/OpenDTU-OnBattery #1020: Support hoymiles hm-600

sissiwup, 2024-06-02:
### Is your feature request related to a problem? Please describe.

My hoymiles hm-600 is reported as unknown
Hardware: 
2024.05.07
132449357

### Describe the solution you'd like

Support the device

Manos1966, 2024-06-02:
Hardware: 2024.05.07

When did you buy this HM-600?
Are you sure it is a HM-600 that uses the NRF24+ communication protocol?

Or does it use standard WiFi  communication?
If it is the later, it will have a WiFi Logo on the lower right front. Or it would be mentioned in the documentation.
The bad news is: IF you have a WiFi HM-600, it is NOT SUPPORTED

schlimmchen, 2024-06-03:
> 132449357

Is that supposed to be the serial number?

sissiwup, 2024-06-03:
I buy the HM-600 in 2022. 
It has no WIFI. All Values are displayed correct in the WEB-Interface.
I can set the LIMIT manually.
But the Dynamic-Power-Limiter did not work. 
It did not reduce the Limit when I use less power.
It did not stop when the voltage of the batterie goes under the 48 Volt, I set in the webinterface.
Info:
Seriennummer | 1141xxxxxxx
-- | --
Produktionsjahr | 2022
Produktionswoche | 29
Modell | Unbekanntes Modell! Bitte melden Sie die "Hardware Teilenummer" und das Modell (z.B. HM-350) hier als Problem.
Ermittelte max. Leistung | 0 W
Bootloader-Version | 0.1.4
Firmware-Version | 1.0.10
Firmware-Erstellungsdatum | 2021-11-01 08:42:00
Hardware-Teilenummer | 132449357
Hardware-Version | 03.74

sissiwup, 2024-06-03:
Power-Limiter-Settings:
Allgemein
Aktiviert

Ausführliche Protokollierung

Angestrebter Netzbezug 
50
W
Hysterese 
30
W
Wechselrichter
Zu regelnder Wechselrichter

Batterie (HM-600/700/800-2T)
Wechselrichter wird von Solarmodulen gespeist

Eingang für Spannungsmessungen

1
Minmales Leistungslimit 
10
W
Grundlast 
300
W
Maximales Leistungslimit 
600
W
Stromzählermessung beinhaltet Wechselrichterleistung

Uhrzeit für geplanten Neustart 

06:00
Batterie Spannungs-Schwellwerte
Batterienutzung Start-Schwellwert
50
V
Batterienutzung Stop-Schwellwert
48
V
Lastkorrekturfaktor
0,001
1/A

sissiwup, 2024-06-03:
> > 132449357
> 
> Is that supposed to be the serial number?

No, the Hardwarenumber

Manos1966, 2024-06-03:
OK, thanks for the explanations.
Now, let's take everything Step-by-step:

> I buy the HM-600 in 2022. It has no WIFI. 

Very good so it is NRF24+

> All Values are displayed correct in the WEB-Interface.

So, you have OpenDTUonBattery and you can see all Values are displayed correct in the WEB-Interface.
The HM-600 **is suported** by OpenDTUonBattery

> I can set the LIMIT manually. But the Dynamic-Power-Limiter did not work.

Have you ever heard of the joke "The problem is between the keyboard and the chair" ?
In short, the problem is more related to your setup of the Dynamic Power Limiter, than on the "support of HM-600".

Have you read the Wiki? You know how to change the Inverter Limit manually so, you know your way arounnd the system 👍 
You can start here: https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Dynamic-Power-Limiter

sissiwup, 2024-06-03:
Ok, lets try it:
[image]
[image]

The batterie is connect via a y-cable to both inputs of the HM-600
At the moment the Power-AC is off.
The voltage of the battery should be taken from the HM-600.

I use a second openDTU-Bat. without a battery for zero feed throttle. This one works without a problem.

Manos1966, 2024-06-03:
Super, thanks for the pictures.
Since you speak German, why did you post in English? (...not that I complain 🤣 )

I admit, I do not see anything wrong in the settings.
I see the Inverter is YELLOW = the openDTU has put it in Standby.
I also see the entry "Angestrebter Netzbezug" (above Hysterese) Where does this come from???
I have the same version 2024.05.07 and I do not see this option! 
Maybe you need to load the Firmware again?

[image]

I can only suggest testing different options to see if the Inverter reacts correctly:

- Re-load the Firmware
See if the problem is solved
 
- Neustarten the Inverter to get it out of the Standby condition
- Minimales Leistungslimit set to 50W
- Grundlast set to 100W
See if the problem is solved

schlimmchen, 2024-06-03:
We need logs from the console.

Manos1966, 2024-06-03:
@sissiwup You can copy-paste the logs to PASTEBIN

https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/How-to-Search-for-Solutions%2C-Ask-Questions%2C-Report-Problems#41-create-a-logfile-datei-with-console

https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Wie-man-nach-L%C3%B6sungen-sucht%2C-Fragen-stellt%2C-Probleme-meldet#41-logdatei-mit-hilfe-der-web-konsole-erstellen

sissiwup, 2024-06-03:
Ok, most of the time the log says:
> Interrupt received
[image]

sissiwup, 2024-06-03:
https://pastebin.com/THZuE75t

sissiwup, 2024-06-03:
[image]

Ist es normal das max. ermittelte Leistung 0 W ist?

sissiwup, 2024-06-03:
[image]
![Bildschirmfoto 2024-06-03 um 20 55 38 PM](https://github.com/helgeerbe/OpenDTU
[image]
[image]
-OnBattery/assets/16230171/0eae4159-e2f6-4ece-b40e-36c037a563cd)

sissiwup, 2024-06-03:
> I can only suggest testing different options to see if the Inverter reacts correctly:
> 
> * Re-load the Firmware
>   See if the problem is solved
> * Neustarten the Inverter to get it out of the Standby condition
> * Minimales Leistungslimit set to 50W
> * Grundlast set to 100W
>   See if the problem is solved
[image]

Hi, i reload the same firmware.

Know I have new Hardware number.
Limit works.

closed 2024-06-03 by sissiwup

Manos1966, 2024-06-03:
👍 💯 🍾 

Good, that you posted the pictures, otherwise I would not have spotted the entry "Angestrebter Netzbezug" (above Hysterese)