hoylabs/OpenDTU-OnBattery #1023: Regelmäßiger Ausfall Verbindung zu Hoymiles HM1500 - nach Neustart OpenDTU on Batterie funktioniert es wieder

theMPstig, 2024-06-03:
### What happened?

Hallo,
seit ca. 2 Monaten (ich mache regelmäßig die Updates und fahre aktuell die Version ) fällt mir manchmal mehrmals täglich, manchmal alle paar Tage die Verbindung vom OpenDTU on Battery zu meinem Hoymiles HM1500 aus. Die OpenDTU steht direkt neben dem Wechselrichter und die Verbindung müsste stabil sein. Ich habe es auch schon einmal ein paar Meter weggestellt ohne Änderung des Problems.
[image]
[image]
Ereignisanzeige zeigt nichts. Konsole ebenfalls nichts:
[image]

Display vom OpenDTU ist ganz schwarz. 

### To Reproduce Bug

Passiert immer irgendwann nach einigen Stunden bis Tagen.

### Expected Behavior

Sollte sich nicht aufhängen ;-)

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.05.07

### Relevant log/trace output

```Shell
Wo finde ich das Log?
```

Solarteur, 2024-06-03:
Ich bin nicht sicher, ob es sich bei deinem Problem um ein Firmware-Problem handelt. Bei mir - und ich denke bei vielen anderen auch - läuft OpenDTU-onBattery (bei mir 2 HM-600, 1 HM-800, Pylontech US2000C, Huawei R4850G2) absolut stabil. Sonst würden hier auch mehr Fehlermeldungen aufschlagen. 

Welche Hardware hast du im Einsatz? 
Wie ist sie aufgebaut (gesteckt, gelötet, Platine)?
Wie ist die Spannungsversorgung aufgebaut (Leistung Netzteil, Kabel)?
Schon einmal ein anderes ESP32- und/oder nRF24-Modul getestet?

Zu "Wo finde ich das Log?": Unter "Infos" --> "Konsole"

theMPstig, 2024-06-04:
Danke dir für deine schnelle Antwort. Die meisten Fragen kann ich dir leider nicht beantworten, da ich die Hardware auf Ebay fertig gebaut gekauft habe. Einzig das Netzteil kann ich dir mit einem Standard 0,6A-USB-Netzteil benennen. Daran hängt ein kurzes USB Kabel.
Aktuell hängt es grad wieder. Das Log zeigt folgendes:
All missing
06:50:02.473 > Nothing received, resend count exeeded
06:50:03.364 > All missing
06:50:03.375 > Nothing received, resend whole request
06:50:04.193 > All missing
06:50:04.201 > Nothing received, resend whole request
06:50:05.013 > All missing
06:50:05.021 > Nothing received, resend whole request
06:50:05.751 > All missing
06:50:05.772 > Nothing received, resend whole request
06:50:06.538 > All missing
06:50:06.548 > Nothing received, resend count exeeded
06:50:06.855 > All missing
06:50:06.864 > Nothing received, resend whole request
06:50:07.050 > All missing
06:50:07.059 > Nothing received, resend whole request
06:50:07.296 > All missing
06:50:07.312 > Nothing received, resend whole request
06:50:07.574 > All missing
06:50:07.584 > Nothing received, resend whole request
06:50:07.761 > All missing
06:50:07.777 > Nothing received, resend count exeeded
06:50:07.791 > Fetch inverter: 116191208685
06:50:07.803 > Request SystemConfigPara
06:50:08.496 > All missing
06:50:08.506 > Nothing received, resend whole request
06:50:09.007 > All missing
06:50:09.022 > Nothing received, resend whole request
06:50:09.542 > All missing
06:50:09.553 > Nothing received, resend whole request
06:50:10.133 > All missing
06:50:11.260 > Nothing received, resend whole request
06:50:11.314 > All missing
06:50:11.326 > Nothing received, resend count exeeded
06:50:12.181 > PowerMeterClass: TotalPower: 1620.60
06:50:12.229 > All missing
06:50:12.239 > Nothing received, resend whole request
06:50:12.897 > All missing
06:50:12.910 > Nothing received, resend whole request
06:50:13.719 > All missing
06:50:13.728 > Nothing received, resend whole request
06:50:14.535 > All missing
06:50:14.543 > Nothing received, resend whole request
06:50:15.252 > All missing
06:50:15.264 > Nothing received, resend count exeeded
06:50:15.661 > All missing
06:50:15.690 > Nothing received, resend whole request
06:50:15.870 > All missing
06:50:15.901 > Nothing received, resend whole request
06:50:16.174 > All missing
06:50:16.250 > Nothing received, resend whole request
06:50:16.378 > All missing
06:50:16.389 > Nothing received, resend whole request
06:50:16.579 > All missing
06:50:16.591 > Nothing received, resend count exeeded
06:50:16.603 > Fetch inverter: 116191208685
06:50:16.614 > Request SystemConfigPara
06:50:17.175 > All missing
06:50:17.240 > Nothing received, resend whole request
06:50:17.739 > All missing
06:50:17.750 > Nothing received, resend whole request
06:50:18.324 > All missing
06:50:18.406 > Nothing received, resend whole request
06:50:18.846 > All missing
06:50:18.856 > Nothing received, resend whole request
06:50:19.324 > All missing
06:50:19.337 > Nothing received, resend count exeeded

theMPstig, 2024-06-04:
Hab in den Systeminfos nachgeschaut: ESP32-D0WD-V3 ist als Chip angegeben. Welche Firmware-Version wäre da eigentlich die richtige?

schlimmchen, 2024-06-04:
> Welche Firmware-Version wäre da eigentlich die richtige?

`generic_esp32`.

> Einzig das Netzteil kann ich dir mit einem Standard 0,6A-USB-Netzteil benennen.

Klingt nicht sonderlich potent, aber pauschal als die Fehlerursache zu bennenen macht auch keinen Sinn. Kannst du es trotzdem gegen ein stärkeres austauschen? Das darf gerne 2A liefern, sodass die Spannung bei 500mA auch noch stabil ist. Die Netzteile sacken in Richtung ihrer maximalen Leistung mit der Spannung stark ab.

> da ich die Hardware auf Ebay fertig gebaut gekauft habe.

Was hast du dafür bezahlt? Ist das günstiger als ein [OpenDTU Fusion](https://shop.allianceapps.io/products/allianceapps-opendtu-fusion)?

Ich hab leider keine gute Idee, warum das bei dir passiert. Sieht aber danach aus, als würde sich das RF Modul verklemmen, oder der SPI Bus dorthin. Das RF Modul gegen ein anders zu tauschen ist sinnvoll, um schrittweise auszuschließen, dass es daran liegt. Jedenfalls bleibt bei solchen sporadischen Fehlern eigentlich nur nacheinander Teile zu tauschen und zu beobachten, ob das Problem verschwindet. Mit dem Netzteil und dem USB-Kabel würde ich anfangen.

theMPstig, 2024-06-04:
Danke dir für deine Rückmeldung. Ich probiere es mal mit einem stärkeren Netzteil. 
Weitere Beobachtung: ich wollte jetzt mal eine ältere FIrmware flashen (da es längere Zeit funktioniert hat), das läuft auch durch über das Webtool, aber es bleibt trotzdem bei der aktuellen Version...irgendwas schein da gröber falsch zu laufen bei mir. 
OpenDTU Fusion ist ungefähr genauso teuer wie das, was ich damals bezahlt habe. Hatte ehrlicherweise nicht wirklich recherchiert und da ich nicht löten kann war ich froh über eine fertige Lösung. Module tauschen kommt daher eher nicht in Frage für mich. Teste jetzt erstmal mit einem stärkeren Netzteil und sehe dann weiter. Ggf. hilft es ja schon.

theMPstig, 2024-06-04:
Noch etwas: das Display wechselt in regelmäßigen Abständen von der Anzeige der Wattzahl auf "Offline". Mal öfter, mal seltener. Kann das zusammenhängen?

schlimmchen, 2024-06-04:
Im Display steht "offline", wenn ein (oder alle?) Inverter nicht erreichbar sind. So als würde der Balken der Karte in der WebUI rot. Das ist also vermutlich ein Symptom deines Problems. Also offenbar passiert das häufiger, dass der WR nicht erreichbar ist, aber meistens erholt sich das System von selbst.

theMPstig, 2024-06-04:
Mit der Sendeleistung habe ich schon rumgespielt mit den verschiedenen Stärken und keine richtige Änderung erzielt. Also z.B. mit dem Problem mit Offline ist das einigermaßen unabhängig von der Meldung. Aber ich habe jetzt mal ein stärkeres Netzteil dran und beobachte mal weiter.

uli-bs, 2024-06-04:
Ich hatte ähnliche Probleme, als meine DTU mit 20cm Abstand zwischen den WRs stand.
Geholfen hat dann temporär, das klingt jetzt schräg, das Einwickeln der Antennen an den WRs mit Alufolie (Abschirmung).

Mittlerweile steht die DTU 2 Etagen tiefer, läuft mit einer "Range-Extender NRF24" und betreut auch noch den WR im Gartenhaus und die Alufolie ist auch weg... ;)

theMPstig, 2024-06-05:
Ok, werde auch mal mit verschiedenen Standorten spielen. Mit dem stärkeren Netzteil jetzt immerhin schonmal seit gestern nachmittag ohne Ausfall. Heute mal abwarten, wie es läuft...

Generell habe ich aber viel von dem "Middle Missing" oder "All Missing" in den Logs. Ist das normal?

schlimmchen, 2024-06-05:
> Generell habe ich aber viel von dem "Middle Missing" oder "All Missing" in den Logs. Ist das normal?

Das ist nicht optimal, aber durchaus normal, dass Pakete wiederholt werden müssen. Ideal ist, wenn keine Meldung "request retransmit" in der Konsole auftaucht. Aber das ist nicht machbar, es wird immer mal wieder Fehlübertragungen geben.

theMPstig, 2024-06-05:
Ok, am Netzteil liegt es nicht. Hat sich heute gegen 16 Uhr wieder aufgehangen leider...

sissiwup, 2024-06-06:
Hallo, habe das gleiche Problem.
Ohne Neustart kann ich es lösen, indem ich den Wechselrichter in der Oberfläche lösche und anschließend wieder hinzufüge.
Ein Neustart über die Web-Oberfläche von OpenDTU hat allerdings nichts gebracht.

theMPstig, 2024-06-09:
Ich komme da irgendwie nicht hin. Ggf. muss ich halt doch mal auf den OpenDTU Fusion umsteigen. Bis dahin: Gibt es eine einfache Möglichkeit, die OpenDTU regelmäßig neu zu starten. Habe ioBroker, aber wenn ich da den Restart auf True setze, passiert leider nichts...

theMPstig, 2024-06-09:
> Hallo, habe das gleiche Problem. Ohne Neustart kann ich es lösen, indem ich den Wechselrichter in der Oberfläche lösche und anschließend wieder hinzufüge. Ein Neustart über die Web-Oberfläche von OpenDTU hat allerdings nichts gebracht.

Wenn ein Neustart bei dir nicht hilft, wird es nicht das gleiche Problem sein, denke ich. Bei mir funktioniert das ja...

sissiwup, 2024-06-09:
Hi, habe mir das Fusion Modell gekauft. Läuft jetzt ohne Probleme.

theMPstig, 2024-06-14:
Bin jetzt auch auf OpenDTU Fusion umgestiegen und beobachte mal...

Nick-411, 2024-07-12:
Hallo, ich habe auch das gleiche Problem dass mein Display offline anzeigt und dann ein roter Balken auf der Oberfläche der DTU angezeigt wird und in diesem Moment keine dynamische Regelung stattfindet und der WR bei dem letzten Limit verharrt.

Ich benutze einen HMS-1800 mit dynamischer Regelung, eim CMT2300A Funkmodul, welches den Status konfiguriert und verbunden anzeigt. Das tauschen des Funkmodules war auch nicht erfolgreich.

Ich habe auch schon versucht die Frequenz zu ändern und dann ist der WR nach nem Moment wieder erreichbar. Leider hilft das aber auch nur für einen oder mehrere Tage. Ich habe es auch schon mit verschiedenen Frequenzen und Sendeleistungen versucht aber das "Offline" Problem kam doch wieder.
Wenn man nichts macht, funktioniert es auch irgendwann wieder, spätestens am nächsten Morgen.
Hier ein Log-File vom letzten Ausfall:
[image]

Hat jemand eine Idee was diesen Fehler auslösen könnte oder was ich noch versuchen könnte ?

spcqike, 2024-07-12:
Ich beobachte das gleiche. Zuerst mit einem selbst gelöteten Modul, jetzt mit dem Fusion Board. Ich habe einen HMS-2000. 

Mit aktiven DPL bricht die Verbindung irgendwie aller paar Tage früh am Morgen ab. Bei mir reicht es, wenn ich die DTU Einstellungen erneut speichere. Dann läuft es sofort weiter für die nächsten 1-2 Tage. 

Bisher hab ich noch keine dauerhafte Lösung. 

Meine Frequenz steht aktuell auf 869MHz, da ich bei der Standard Frequenz viele Aussetzer hatte.

theMPstig, 2024-07-13:
Seitdem ich auf die Fusion umgestiegen bin, habe ich überhaupt keine Probleme mehr - denke also auch eher ein Hardwareproblem gewesen...

DerHavey1994, 2024-08-22:
@uli-bs Das mit der Alufolie habe ich heute mal getestet, funktioniert wunderbar! Bis jetzt im Log kein "Inverter ist offline" mehr gehabt. (vorher jede Minute mindestens 1x)

Warum aber auch der HM-1500 so schlecht erreichbar ist wundert mich ein wenig. Mit einem HM-1200 habe ich überhaupt keine Probleme.

schlimmchen, 2024-12-02:
Themenstarter schreibt:

> Seitdem ich auf die Fusion umgestiegen bin, habe ich überhaupt keine Probleme mehr - denke also auch eher ein Hardwareproblem gewesen...

Daher schließe ich das Issue.

closed 2024-12-02 by schlimmchen