hoylabs/OpenDTU-OnBattery #1025: Announcement: EOL for 4MB flash memory

schlimmchen, 2024-06-03:
### Describe the solution you'd like

This text will be replaced.

tobi0171, 2024-06-06:
wow, thank you @schlimmchen for this information. 

Would it be conceivable to add also an example of the "pinmap" for the "ESP32-S3 (44pin)" and an adapted "frizzing" for DIY-hobbyists add to the wiki? That would probably be very helpful for many who are starting out on the topic and are not sure yet. Thanks to you, you're all really doing a great job.

swingstate, 2024-06-06:
That should be possible.

schlimmchen, 2024-06-13:
For the record: The limit is hit again after merging the upstream changes from release v24.6.10:

`Exception: firmware binary too large: 1969872 > 1966080`

Next step: I am going to remove `sortables.js` from the web application. This removes the ability to sort inverters in the list of inverters (settings and live view). I already know that this is going to carve out around 20k of code (surprisingly much). This gives us headroom once more to merge the upstream changes and leaving around 16k of headroom, sacrificing a feature that hopefully only a very few will miss. My power meter refactoring efforts (ongoing for weeks now) will then use around 11k out of those 16k.

Once those 5k (possibly less) remaining are used up, I don't know how to postpone the inevitable any longer, as I (currently) have no more ideas what to carve out. As I said elsewhere, I am not going to put effort in looking for 500 Bytes here or there so save with hours worth of refactoring.

spcqike, 2024-06-13:
> As I said elsewhere, I am not going to put effort in looking for 500 Bytes here or there so save with hours worth of refactoring.

that’s totally understandable. Thank you for your effort so far! 

One idea: is it possible to store the webviews outside the firmware? Like in AI-on-the-edge, where you put the html into SD card. 

I know we don’t have an SD card, but is it possible to put the views in like littleFS filesystem? So that you don’t need to have it twice (in binary firmware and OTA firmware space)?

schlimmchen, 2024-06-13:
> One idea: is it possible to store the webviews outside the firmware?

That beast is ~200kB. It does not fit the littlefs even if it was empty. There is no other part of the flash we could use. There is no other non-volatile storage.

We could re-partition and designate a new shared partition to hold the web app. That would be a breaking change and require significant effort (different than changing the partition layout for 8MB flash but otherwise keeping things like they are). And it would not last very long. It also creates a new problem: Not only the firmware code must stay within its own bound, but the web app must as well, governed by a different limitation.

I am not convinced we gain anything...

Also: The webapp is tailed to a specific firmware version. The API must be compatible. It makes sense that it is integral part of the firmware.

spcqike, 2024-06-13:
Valid points. As I said, just an idea :) 

changing to 8MB modules will be easier I guess. That’s why I already have some N16R8 modules on hand. So I’m ready to switch, whenever needed :) (I just need to re-solder everything 🫣)

meisterlampe, 2024-06-14:
Well I guess, I have only 4MB Flash, but wanted to check anyway and cannot see it on the page 'System Information'. :)
Any ideas, why the information is not shown?
I just upgraded to 2024.06.03, deleted the cache, refreshed and tried different browsers.

[image]

schlimmchen, 2024-06-14:
@meisterlampe I am confused... It should be right under the line with the CPU frequency, like this:

[image]

You wrote that you cleared the browser cache and refreshed (also with Ctrl + F5?) and tried different browsers. Then I have only one explanation left: you built the firmware yourself and forgot to build the web app. Could that be the case? Otherwise I am puzzled...

schlimmchen, 2024-06-14:
> I already know that this is going to carve out around 20k of code (surprisingly much).

Well, that was a lie (by the vue vizualizer or whatever the name of that tool was). The actual savings are merely 13k. I tried, and merging the upstream changes as well as adding the power meter refactoring will already fail. That means we can only have one, and to get the other, we need to switch anyways.

Trying not to be a crybaby, but this is becoming a frustrating issue. I don't want to deal with it any more. We had a discussion, we have a solution, I did the announcement. Unfortunately, the implementation will happen much sooner than anticipated:

### The next release will only run on ESP32 with at least 8 MB of flash memory.

I will update this issue's description (the announcement) accordingly.

meisterlampe, 2024-06-15:
@schlimmchen you've got me, now it is working. Sorry.  (sadly 4MB, as expected. Will wait for the OnBattery Pro PCB. But right now everything is working good, so that's ok :) )
Honestly I've never built the webapp before and had no problems. It seems I was always lucky that the webapp_dist was up to date. O:-D

Just to encourage you: Requiring 8MB is the right decision. You said, that the low hanging fruits have been picked. If saving some bytes means way too much effort, just to postpone the inevitable, it makes no sense to ride the dead horse (and it also makes no fun :) )

schlimmchen, 2024-06-15:
> Honestly I've never built the webapp before and had no problems.

Have you switched to OpenDTU-OnBattery only recently? OpenDTU has up-to-date blobs of the web app in its repo. As I am ignorant and stubborn, I won't do that for OpenDTU-OnBattery. Binary blobs shall not be part of a source code repo, unless they are not an (intermediate) result of building something (manuals, external proprietary firmware, etc.).

> Just to encourage you: Requiring 8MB is the right decision. You said, that the low hanging fruits have been picked. If saving some bytes means way too much effort, just to postpone the inevitable, it makes no sense to ride the dead horse (and it also makes no fun :) )

Thanks for understanding!

Gumbagubanga, 2024-06-16:
Going 8 MB is a logical choice.

Nonetheless it saddens me as I'm running a WT32-ETH01 for the ethernet port even thou I've pretty much ran out of GPIO pins immediately. Are there any known (affordable) boards with supported ethernet and at least 8 MB flash?

schlimmchen, 2024-06-19:
I had yet another idea about how to continue support for devices with 4MB of flash memory. AFAIK we have not discussed it yet. Just now I completed some tests and it seems that this would fly:

We can introduce a new platformio environment, probably named `generic_esp32_4mb_no_ota`. The second app partition will be removed from the partition table. The freed space is allocated to the remaining single app partition, doubling it in size just like we want to. OTA updates will not be possible when using the firmware build for this environment. The firmware will detect this and the web UI will display a respective message when navigating to the "Firmware Upgrade" view. These devices can only be upgraded by wire. For this reason, there will only be a "factory" binary for this environment (at least as part of the release).

This change is still a breaking change, as the partition layout must be updated. However, once switched to this new type of release, users with ESP32 with only 4MB can continue using their device, with the major caveat that updates are only possible by wire going forward. Users who find this unacceptable need to upgrade to an ESP32(-S3) with 8MB+ flash.

Opinions?

swingstate, 2024-06-19:
Settings would still be preserved?

schlimmchen, 2024-06-20:
> Settings would still be preserved?

Yes, the filesystem partition is still left as it is, so settings are preserved when writing the new factory bin using esptool.

CKone01, 2024-06-20:
Excellent workaround!

helgeerbe, 2024-06-20:
Hi @schlimmchen, I would be OK to flash esp32 in future by wire. But not everybody has easy access to his openDTU-onBattery. 

If we go they forward to have a new plattformio environment. I still think, it could make sense to have a small ota firmware. Just thinking while writing:
- klick update firmware in the ui
- reboot to small ota image
- trigger ota update, or return to openDTU-onBattery image
- do ota update
- reboot to openDTU-onBattery firmware

Don't know how big this image would be, but I guess this brings us a lot of free memory.

Actually we don't make really use of the cold standby image (no fallback strategy on errors to the previous image). So in my point of view, its just a waste of resources.

We can discuss this within the upstream project. Thomas is facing the same situation more sooner than later.

schlimmchen, 2024-06-20:
> But not everybody has easy access to his openDTU-onBattery.

I can see that. Those people hopefully are easily convinced to upgrade to an ESP32 with 8 MB of flash to keep the convenience of OTA updates, at least eventually. My idea to provide the non-OTA firmware is merely an attempt to ease the transition for people with 4 MB devices. In the long run I assume that every OpenDTU-OnBattery user switches to an ESP32 with at least 8 MB of flash: Those who start fresh will know to obtain such a device in the first place, and many existing users will miss the OTA functionality, so they'll upgrade, or they are fine updating by wire from time to time, so no OTA update functionality is need for them either.

The process you describe using a minimal firmware makes sense. However, let me iterate disadvantages and problems, partially copied from https://github.com/helgeerbe/OpenDTU-OnBattery/issues/935#issuecomment-2085149009:
* Such a minimal firmware must be created in the first place. There are two options: (1) Add a whole bunch of `#ifdef` "everywhere" in the code. These will not only guard whole files, but blocks of code, and also use alternatives, like MessageOutput will only be available in the main firmware, while we must use Serial in the minimal firmware to save space. (2) Maintain a whole different branch and build the minimal firmware from there. (1) is bad because it clutters the code and because it does not work for the web app -- we would still need a standalone second web app. (2) is bad because changes in the main firmware need to be backported into the minimal firmware.
* The last sentence brings another issue: The main firmware must stay (forever) compatible with the minimal firmware (settings and such). This will bring serious headaches in the future -- at least that's what I am foreshadowing.
* The minimal firmware will never be updated: Let's be real, nobody ever updates the minimal firmware. Even if we make the effort to allow to update the minimal firmware using an OTA update (yet another special feature), nobody will do so unless we force the users to (which we could -- another special feature).
* For tests I built a simple serial SML sender in an Arduino sketch for ESP32. This firmware is as simple as it gets. It already uses 141k of memory. I just added WiFi capabililties (connecting to an Access Point or being one) and the code size jumped to 505k. The current web app is ~210k. Let's say we can cut that in half (which is optimistic, given that the uncompressed app is currently 940k and compression will be less efficient with less code), so the total is now 610k. Oh, wait, let's add a web server: 647k. And we need to read the config.json, so let's add ArduinoJson and support for littleFS: 680k. Missing is the update library to perform the actual update of the main firmware and the respective implementation of the SPI flash API. Let's allocate 20k of code for that, so we are at 700k in total. Was that it or is something else missing?
* Lets allow 25% of growth for the minimal firmware, to at least be able to update libraries: 875k is the minimum partition size for the minimal firmware.

It is doable, I guess. Is it worth the headaches: I say that it isn't. It's also not worth the risk that the whole approach simply does not work as expected, after investing dozens of hours into this.

There are inconveniences for the users, too: OTA updates will be a multi-step procedure using this approach, rather than a one-click operation.

How do you @helgeerbe feel about this approach after reading my arguments?

I was unsure about whether or not I should bring this up in the upstream project. If you think so as well, then I will start a discussion (it will be linked below this post).

schlimmchen, 2024-06-21:
I spent another hour exploring. This time I went the other way around: I created a new branch based on the current development branch and stripped everything that I could. I managed to build and run this minimal firmware. However, it is 1160kB. This is much larger than my previous estimate. The web app is not yet stripped down, so let's assume the firmware will be 1060kB. Still more than a Megabyte.

Things that I realized cannot be removed and were not accounted for in yesterdays estimate: (1) Ethernet Support: If we provide such a minimal firmware, it also needs to support existing ESP32 that are connected via Ethernet. (2) PinMapping support: required for Ethernet.

My best guess for the larger size compared to yesterdays estimate is that this time, functions that are actually needed to provide the intended functionality, are now included, whereas yesterday I merely tried to tickle the respective lib by using a single function of it.

To implement this breaking change, users still need to update their partition layout using a wired connection. We need to maintain the minimal firmware, invest more time to make it at least a little smaller, and we don't gain very much for the main firmware partition, maybe ~500kB. I still think this is not worth the effort. @helgeerbe Let us know what you think.

spcqike, 2024-06-21:
> (2) PinMapping support: required for Ethernet.

is this really true? I mean, someone who wants to upgrade an existing firmware, already has a working pinmapping. So there is no need for the full code that belongs to pinmapping itself. As we don’t need to use anything out of it expect the pins for Ethernet. 

Would it be possible to like store the Ethernet pins in a different file if and only if Ethernet is used? In the main firmware. 

So if minimal firmware boots it only checks whether or not this Ethernet file exists and if so, it loads it. If it doesn’t exist, it will use WiFi. So the minimal firmware wouldn’t need to handle or read everything related to pinmapping. 

Idk but tasmotas minimal firmware is just about 259Kb. It doesn’t come with a fancy web interface where you can setup something. Everything you can do is upload another firmware. So it kind of only uses wifi with a very simple and limited html server file and OTA handling. 

If we consider dynamic Ethernet ports (meaning they can be connected to whatever pin the user defines), we would need to check for them and their pins, too. So we would need wifi, littlefs, Ethernet and OTA, wouldn’t we?

schlimmchen, 2024-06-22:
Not sure that I understand. The code to be able to read a file from the littlefs partition and to interpret it (JSON) has to be there, regardless of how much data is in it (all pins used by the main firmware or just the ones for Ethernet). I was not talking about support for editing the pin mapping, neither to edit the config, in the minimal firmware. Those would be read-only in the minimal firmware that I explored. The config *must* be read, however, to be able to connect to the configured Wi-Fi network, or to configure for Ethernet, etc.

Have a look in the [respective branch](https://github.com/helgeerbe/OpenDTU-OnBattery/tree/minimal-poc).

I have been thinking about this for months now, as I anticipated this issue a long time ago, and I was so happy to have found support to justify the switch to 8 MB devices, and even happier to find that 4 MB devices could be supported when sacrificing OTA updates -- with very little effort. Now the discussion about the minimal firmware approach is starting anew, after I published an announcement and people started buying new hardware and upgrading their setups... I invested a couple of hours into exploration of something that I don't even support, in order to come up with a good estimate of how much flash the "minimal" firmware would need. Now I am spending time to argue with this:

> Idk but tasmotas minimal firmware is just about 259Kb

I don't know where you found that number and I did not check how large the Tasmota minimal firmware for "legacy" ESP (ESP8266) is, but the Tasmota safeboot firmware for ESP32 (non-S3) is 813kB:

[image]

The safeboot partition is 832kB, see [the docs](https://tasmota.github.io/docs/Safeboot/). This is *very* close to my estimate of 875kB above.

Moreover, I think that a comparison with Tasmota, at least regarding firmware update strategies, makes no sense. Tasmota devices are buried behind some physical switch or inside lamps or something. And most of them don't even have a port to connect to for firmware updates. Also, those devices cannot just be upgraded to an ESP32-S3 with 8MB. The requirement for supporting OTA updates on these devices is obvious. I would never argue about that. I wouldn't use Tasmota on my many ESP devices if I had to dig them up and solder wires to them (Shelly mini series) every time I wanted to perform an update. It makes perfect sense to me that people invest time in having and maintaining the Tasmota safeboot firmware.

The situation for OpenDTU-OnBattery, however, is very different. So different, that I think OTA updates are a convenience, not a requirement. And users may keep this convenience, they merely have to install a different MCU. It's a one-time effort. And as other people stated: 30 bugs for a Fusion board is justifiable given that the cost for all the other parts of a small solar plant are so much higher.

Given that, I still want to have Helge on board and feeling good about how we move forward. And he was right to suggest that we ask upstream for an opinion. Even though I was about to move forward, we will wait for their input on this matter.

schlimmchen, 2024-06-23:
@spcqike is right, we can just serve static (or mostly static) HTML pages in the minimal firmware. The minimal firmware will have a completely different look, it will be ugly, and there will be no translations (at least no dynamic ones). Should be okay for something one only sees when performing an OTA update. That should bring us down to ~960kB for the minmal firmware.

I think we should add some sort of API between the main firmware and the minimal firmware, i.e., the minimal firmware uses its own (very limited) set of settings, and each time an OTA update is triggered, the main firmware constructs this from the active config. This allows to change the main config without the need to keep compatible with the minimal firmware. The only thing that might need an update in such a case is the function writing/updating the minimal firmware config. Also, getting rid of ArduinoJson should save code (roughly 33k as estimated earlier).

spcqike, 2024-06-23:
> Also, getting rid of ArduinoJson should save code (roughly 33k as estimated earlier).

That’s what I thought when writing 

> Would it be possible to like store the Ethernet pins in a different file if and only if Ethernet is used? In the main firmware.

So to have a file like „Ethernet.conf“ wich is just one like of comma separated PIN numbers (like „12,4,15,16“ or whatever pins are used for Ethernet) this file could be created by pinmapping loading process and holds the Ethernet pins in a certain order. Of course only if Ethernet is used 😄 

we could also use some EEPROM bytes to hold these pins. If eeprom handling uses less Programm size than LittleFS, idk. (Even tho at the end eeprom is also just emulated in flash)

But you wrote we need the config file and therefore JSON anyway, as it holds the wifi credentials.

Snoopy-HSS, 2024-06-24:
I try to compile the current development and it says:
    raise Exception("firmware binary too large: %d > %d" % (fw_size, max_size))
Exception: firmware binary too large: 1970624 > 1966080

is it already over with ESP32_generic 4MB ????

Snoopy-HSS, 2024-06-24:
It's weird.
When I compile for the S3 size is smaller !?
why is there a 2MB limit when i use the generic_S3 ???

Checking size .pio\build\generic_esp32s3\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  24.0% (used 78680 bytes from 327680 bytes)
Flash: [==========]  95.9% (used 1886133 bytes from 1966080 bytes)
Building .pio\build\generic_esp32s3\firmware.bin

In Platform.ini default ENV is still generic_ESP32

Changing partition worked, can also be uploaded Serial via, visual Studio code

schlimmchen, 2024-06-24:
> It's weird. When I compile for the S3 size is smaller !?

Yes, as the S3 is a slightly different processor with a different instruction set, so the code is can be smaller (more efficient instructions, etc.).

> why is there a 2MB limit when i use the generic_S3 ???

I don't know what you mean. The partition table is the same for ESP32-S3 and non-S3.

Snoopy-HSS, 2024-06-24:
Flash: [==========] 95.9% (used 1886133 bytes from 1966080 bytes)

Only a little under 2MB availabe.
How can i change, that it uses 16MB filesystem when compiling generic_esp32s3

is this part for all boards? 
board_build.partitions = partitions_custom_4mb.csv

it's set in platform.ini to 4MB and ESP32 per default...

greymda, 2024-07-15:
> I had yet another idea about how to continue support for devices with 4MB of flash memory. AFAIK we have not discussed it yet. Just now I completed some tests and it seems that this would fly:
>
> We [...]

is this option still on the table? @schlimmchen

schlimmchen, 2024-07-15:
@greymda This is already implemented and I don't see any reason why this variant of the firmware should go away. There is no release with this variant, yet, but build runs already build this kind of firmware, named as suggested.

futzicomp, 2024-07-28:
Hallo zusammen,
Nochmals vielen Dank für euren Einsatz! Die Aussicht sinnvolle Funktionserweiterungen zukünftig erhalten zu können ohne an die Grenze des ESP mehr zu stoßen lassen mich den Umstieg auf ein ESP32S3 locker verkraften!
Ich habe einen ES32S3 N16R8 bei mir rumliegen. Spaßeshalber habe ich 2024.06.03 aufgespielt und bin etwas verwirrt was die PSRAM Größe angeht. 
In der Info Anzeige wird 16MB Flash aber 0MB PSRAM angezeigt, obwohl dieser Chip 16/8MB haben sollte. Kann ich mich auf die Anzeige N16R8 auf dem Chip oder auf die Anzeige von Flash und PSRAM Speicher in der Info verlassen? (habe ich ein Fake Exemplar erwischt?)
Danke Uli
[image]

schlimmchen, 2024-07-28:
Die Unterstützung für PSRAM muss ausdrücklich eingeschaltet werden. Dafür braucht es dann eine zusätzliche Firmware-Variante. Eine solche könnten wir demnächst mal einführen. Beispiel:

* generic_esp32_4mb_no_ota
* generic_esp32_8mb
* generic_esp32s3
* generic_esp32s3_psram
* generic_esp32s3_usb
* generic_esp32s3_psram_usb

Allerdings bin ich mir der Risiken nicht bewusst. Vielleicht gibt es einen Grund, warum tbnobody noch keine solchen Images anbietet. Außerdem ist ja "die Wahrheit", dass bisher kein PSRAM gebraucht wird, wir kommen ja bisher gut mit dem integrierten RAM aus.

futzicomp, 2024-07-28:
Thanks for your answer and again:thanks for your good work! My system is now working flawlessly since one year(!) 
I am looking forward for cool new developments which are possible with the upgraded Hardware 🙂

schwatter, 2024-08-02:
@schlimmchen 
Tag, meine 8mb Flashsize Esp's sind im Zulauf. Nu dachte ich, bis dahin spiele ich mit meinen 4m Flashsize Esp's. Aber da bootet OpenDTU on Battery nicht. Bzw nur bis zum Release 2023.08.29. Das ist 1.66mb groß. Alles ab 1.70mb macht Probleme.
EspHomeFlasher schreibt aber, mein Flash wäre 4mb groß. Kann mich wer erhellen. Falsche angaben? fake Chip?

`Using 'COM5' as serial port.
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.......
Detecting chip type... ESP32
Connecting.....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WD-V3 (revision 3)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: A0:B7:65:4E:F9:DC
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...`

Gruß schwatter

spcqike, 2024-08-02:
@schwatter das „Problem“ ist die partitionierung. OpenDTU hält eine zweite 2MB Partition für das OTA Update vor. Daher kannst du nur die Hälfte benutzen und auch das Image ist <2MB.

schwatter, 2024-08-02:
Ah, danke!
Vielen Dank für die Erklärung und den Link. Die hatte ich nicht gesehen.
Sind die Non-Ota auch irgendwo verlinkt? Oder wo sind die beschrieben?

Gruß schwatter

schwatter, 2024-08-02:
Ok, 
unter Linux mit esptool hat es geklappt.

`root@pi3Pihole:/tmp# esptool.py -p /dev/ttyUSB0 -b 230400 write_flash --flash_size detect --flash_freq 40m 0x0 opendtu-onbattery-generic_esp32_4mb_no_ota.factory.bin
esptool.py v4.7.0
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a0:b7:65:4e:f9:dc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 230400
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00000000 to 0x001fafff...
Compressed 2073312 bytes to 1292183...
Wrote 2073312 bytes (1292183 compressed) at 0x00000000 in 59.5 seconds (effective 278.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
root@pi3Pihole:/tmp#`

[image]

Danke und Gruß

schwatter

Gumbagubanga, 2024-08-04:
> Going 8 MB is a logical choice.
> 
> Nonetheless it saddens me as I'm running a WT32-ETH01 for the ethernet port even thou I've pretty much ran out of GPIO pins immediately. Are there any known (affordable) boards with supported ethernet and at least 8 MB flash?

Update:
I bought an Lilygo T-ETH-Lite S3 board (16 MB Flash, 8 MB PSRAM) with an currently unsupported W5500 ethernet controller. Luckily in an upstream PR https://github.com/tbnobody/OpenDTU/pull/2068 an initial support has been implemented.
As @marvincarstensen mentioned this project directly, I'm positive that he tested it against the latest OpenDTU-OnBattery too and we'll finally get ethernet support back on ESP32-S3 boards 👍

Dani1802, 2024-08-17:
If anyone needs a case, I remixed an existing one to fit the WROOM-1-N16R8 ESP32-S3-DevKitC-1.
It can be found here: https://www.thingiverse.com/thing:6734089

schlimmchen, 2024-08-18:
@Dani1802 Thanks, linked [in the docs](https://opendtu-onbattery.net/3rd_party/cases/).

schlimmchen, 2024-10-07:
We published two releases which implement the changes outlined here. This issue/announcement has been stale for almost two months now, so I will close this issue now. Please open a new issue in case of problems regarding the changed partition layout.

closed 2024-10-07 by schlimmchen