hoylabs/OpenDTU-OnBattery #1029: Keine Verbindung zur Pylontech Batterie

schlimmchen, 2024-06-04:
Fortsetzung aus #1018:

Hab jetzt nochmal alles neu gemacht. Die openDTU-onBattery hängt sich jetzt nicht mehr auf, nachdem ich die pinmapping geändert hab.

allerdings kriege ich den Ladezustand der beiden US3000C nicht übermittelt!

Hab geändert von GPIO 17+18 auf 3+8 und 44+43.

Keine Anzeige der Batterie in der openDTU-onBattery!🤷🏻‍♂️

[image]
[image]
[image]
[image]

_Originally posted by @Snusme83 in https://github.com/helgeerbe/OpenDTU-OnBattery/issues/1018#issuecomment-2147041616_

schlimmchen, 2024-06-04:
@Snusme83 Wie sehen denn deine Batterie-Einstellungen aus? Was wir auf der Konsole ausgegeben (zusätzliches Browserfenster offen halten) wenn du die Batterieeinstellungen speicherst? Was hängt denn an den Pins 43 und 44 für ein Chip? Wie kriegt der Strom? Hat der in einem vorherigen Setup schon mal funktioniert? Du hast doch auf einen ESP32-S3 hochgerüstet und vorher lief das, oder nicht?

Snusme83, 2024-06-04:
Servus schlimmchen,

sorry, dass ich dir solche Umstände mache. Ich glaube das hat sich jetzt aber erledigt!:-) Ich Trotte habe beim SN65HVD230 CAN-Board  (für den Pylontech) die 3,3 V nicht mit dem ESP32 S3 verbunden.

Es funktioniert jetzt und ich werde dir künftig nicht mehr auf den ... gehen! ;-)

Für dich zum Verständnis: Ich habe meine ursprünglich funktionierende openDTU-onBattery (ESP32) unangetastet gelassen!

Parallel dazu habe ich mir die zweite openDTU-onBattery (ESP32S3) aufgebaut. So wie es ausschaut funktioniert die jetzt und ich kann die ursprüngliche vom Netz nehmen.

Genauso wollte ich die openDTU-onBattery haben. Jetzt ist sie nicht nur nahezu perfekt, sonder PERFEKT!!!
Vielen Dank für deine Unterstützung - ganz dickes Lob!!!

Eine Frage hätte ich jetzt noch. Von Firmwareaktualisierungen für die openDTU-onBattery (ESP32S3) lasse ich die Finger, oder?

Einen Dank auch an  die anderen, die mitgewirkt und mir geholfen haben.

[image]

schlimmchen, 2024-06-04:
> So wie es ausschaut funktioniert die jetzt

Super!

> Eine Frage hätte ich jetzt noch. Von Firmwareaktualisierungen für die openDTU-onBattery (ESP32S3) lasse ich die Finger, oder?

Wenn du mit deinem Setup zufrieden bist und zufrieden bleibst, dann musst du auch keine neue Firmware installieren. Welche PowerMeter-Anbindung hast du denn? Wenn es nicht MQTT ist, dann behaupte ich, dass sich das nächste Update für dich lohnen wird.

Warum denkst du denn, du müsstest die Finger von Updates lassen? Hast du Angst vor Regressionen? Wenn irgendetwas absichtlich inkompatibel wird, dann wirst du es erfahren im Changelog des Release. Andernfalls darfst du damit rechnen, dass neue Firmware besser und nicht schlechter wird.

Hast du einen ESP32-S3 mit mind. 8MB Flash-Speicher gekauft? Siehe Infos -> System in der Web UI.

closed 2024-06-04 by schlimmchen

Snusme83, 2024-06-04:
Also grundsätzlich bin ich jetzt zu 100 % zufrieden.

Grundsätzlich möchte ich gerne Up-To-Date sein, bin aber auch schon ein paar Mal auf die Schnauze geflogen,
als ich das "Running System" getouched habe. Die Firmware die du mir da bereitgestellt hast ist ja eine Spezielle und wenn ich eine Firmwareaktualisierung mache in der OpenDTU-OnBattery, dann habe ich wieder eine Allgemeine drauf und dann funktioniert wahrscheinlich die Hälfte nicht mehr bzw. gar nicht mehr! Ich kenne mich da zu wenig aus und es ist mühsam für den Laien sich da ständig alles zusammen suchen zu müssen und in 6 Wochen z. B.: weiß ich dann nicht mehr was ich da genau gemacht hab, damit alles wieder so läuft wie es das aktuell tut.

Jetzt bin ich aber neugierig, wieso sich ein Update für mich lohnen soll - ich bin zu 100 % zufrieden:-)

PowerMeter-Anbindung: Volkszähler am Stromzähler --> HTTP(S) + JSON

Ich habe mir einen ESP32-S3 gekauft, so wie du es mir geraten hast! ;-)
[image]

schlimmchen, 2024-06-04:
> dann habe ich wieder eine Allgemeine drauf und dann funktioniert wahrscheinlich die Hälfte nicht mehr bzw. gar nicht mehr!

Achso... Also es ist so: Die Änderungen für die Unterstützung von 3 MPPTs sind im Pull-Request #1018 in den development branch gekommen. Gestern sind die dann Teil des master Branches geworden und Teil des Releases 2024.06.03. Daher steht das auch in den Changelogs zu diesem Release, dass diese Funktion jetzt drin ist. Siehe https://github.com/helgeerbe/OpenDTU-OnBattery/releases/tag/2024.06.03 Daher kannst du bedenkenlos zukünftige offizielle Releases installieren. Ich kann verstehen, wenn dir das nicht klar war. Die Firmware war aber nicht für dich gebastelt, sondern die war das Ergebnis des Build-Runs des Pull-Request. Und dieser Code ist unverändert in den freigegebenen Teil der Firmware gekommen.

Ein Update auf die 2024.06.03, als auf das Release, wäre durchaus sinnvoll. Du solltest grundsätzlich keine Pull-Request-Firmware dauerhaft laufen lassen müssen. In deinem Fall kannst du schon auf das letzte Release. Manchmal dauert das natürlich länger.

> Jetzt bin ich aber neugierig, wieso sich ein Update für mich lohnen soll - ich bin zu 100 % zufrieden:-)

Also ich bin mir einigermaßen sicher, erst recht bei nunmehr drei MPPT, dass die teilweise länger erscheinen als würden keine validen Daten empfangen. Der HTTPS+JSON Power Meter hat bisher das Problem, dass der HTTP-Request synchron ausgeführt wird. Dadurch kommt die loop() Funktion des MPPT Controllers (Teil der OpenDTU-OnBattery Firmware) regelmäßig nicht häufig genug dran. Das wird mit dem PowerMeter Refactoring behoben sein.

> Volkszähler am Stromzähler

Geht das nicht auch mit MQTT? Welche Variante hast du denn? Ist das Ding eigenständig? Wo kommt der Webserver her? Ein Hichi ist das ja nicht?!

> Ich habe mir einen ESP32-S3 gekauft, so wie du es mir geraten hast! ;-)

Ich hätte nicht mit Bestimmheit sagen können, dass ich dir auch schon dringend geraten habe, einen mit 8MB zu kaufen. Aber das hat ja geklappt.

Snusme83, 2024-06-04:
Achso, ich könnte es quasi wagen das Release 2024.06.03 einzuspielen, weil da alles integriert ist und normal laufen muss!

> Also ich bin mir einigermaßen sicher, erst recht bei nunmehr drei MPPT, dass die teilweise länger erscheinen als würden keine validen Daten empfangen. Der HTTPS+JSON Power Meter hat bisher das [...]

VERSTEHE DA LEIDER NUR BAHNHOF!
> 
> > Volkszähler am Stromzähler
> 
> Geht das nicht auch mit MQTT? Welche Variante hast du denn? Ist das Ding eigenständig? Wo kommt der Webserver her? Ein Hichi ist das ja nicht?!

MQTT?
Habe nen bitshake SmartMeterReader.  Dieser läuft mit einem ESP32 über Tasmota! Wie genau ich das gemacht habe 🤷🏻‍♂️. Schon wieder zu lange her.

wäre also MQTT besser als meine Art und Weise die Werte in die OpenDTU-on Battery zu bringen?🤔

schlimmchen, 2024-06-04:
> wäre also MQTT besser als meine Art und Weise die Werte in die OpenDTU-on Battery zu bringen?

Ja, aber mach dir keinen Kopf drum, lass dich nicht nervös machen nur weil ich glaube, dass es noch besser geht. Wenn es für dich funktioniert, ist das super.

> VERSTEHE DA LEIDER NUR BAHNHOF!

Beobachte mal den Live View der MPPTs. Steht da bei "letzte aktualisierung" regelmäßig mehr als 1 sekunde? Die Daten sollten jeweils immer nur höchstens 1 Sekunde alt sein. Ich vermute, dass das bei dir nicht der Fall ist, weil der PowerMeter zwischendrin mit seinem HTTP request zu tun hat. Das sollte dir alle zehn sekunden auffallen, dass die "letzte Aktualisierung" länger als 1 Sekunde zurückliegt.

Es sind aber nur Details. Lass dich wie gesagt nicht von mir wuschig machen :wink: