hoylabs/OpenDTU-OnBattery #1032: [Request] Activate in dynamic power limiter a second inverter, when the first is at 100%

dragricola, 2024-06-06:
### Is your feature request related to a problem? Please describe.

(Text in German see below)
When the power of the inverter, which is controled by dynamic power limiter, is not sufficient, a second inverter can be added with constant power. The first inverter keeps on controling the first inverter with reduced relative magnitude as long as the total power of both inverters is sufficient. Until now the start of a second inverter can only be done by a script which is excecuted on a home automatisation like ioBroker. It may become difficult, because messages for limitaion of the inverter must be sent without retained flag and some MQTT-Servers (like ioBroker MQTT-Adapter) sent MQTT-Messages by default with retained flag. See also the discussion (in German) unter https://discord.com/channels/984173303147155506/1230866028414767168/1247928761345642647. 
It would be fine if the dynamic power limiter would directly support a second inverter.

Wenn die Leistung des durch den dynamic power limiter gesteuerten Inverters nicht ausreicht, kann man einen zweiten Inverter mit konstanter Leistung dazuschalten. Der erste kann dann weiterhin durch den power limiter geregelt mit geringerer Aussteuerung arbeiten, sofern die Gesamtleistung beider Inverter ausreicht. Derzeit kann das Zuschalten des zweiten Wechselrichters nur über ein Script erfolgen, dass auf einem Home-Automatisierungssystem wie z.B ioBroker ausgeführt wird. Erschwerend kommt hinzu, dass die MQTT-Meldungen zum Limitieren des Inverters ignoriert werden, wenn eine Meldung mit Retained-Flag gesendet wird und einige MQTT-Server solche Meldungen defaultmäßig mit Retained-Flag versenden (z.B. MQTT-Adapter des ioBrokers). Siehe hierzu auch die Diskussion unter https://discord.com/channels/984173303147155506/1230866028414767168/1247928761345642647. Um diese Probleme zu umgehen, wäre es schön, wenn der dynamic power limiter direkt einen zweiten Inverter unterstützen würde. 

### Describe the solution you'd like

The second inverter is activated, when the first inverter reaches level 100% and deactivated, when the first inverter can perform the requested power at his own. The limit of the second inverter could be increased stepwise until the first inverter is below 100% level or the secend inverter has reached 100%. In configuration menu input for a second inverter and the maximum power  has to be added.

Der zweite Inverter wird eingeschaltet, wenn der erste Inverter 100% Aussteuerung erreicht hat. Er wird wieder ausgeschaltet, wenn der erste Inverter die geforderte Leistung alleine aufbringen kann. Das Limit des zweiten Inverters könnte man solange stufenweise heraufsetzen, bis der erste Inverter unter 100% Aussteuerung angelangt ist oder auch der zweite Inverter bei 100% ist. Im Konfigurationsmenu müsste man einen zweiten Inverter wählen und dessen Maximalleistung eingeben können. Eventuell ist es auch sinnvoll anzugeben, mit welcher Aussteuerung der zweite Inverter eingeschaltet werden soll.

### Describe alternatives you've considered

Man könnte auch die angeforderte Leistung auch gleichmäßig auf beide Inverter verteilen. Das hätte aber vermutlich einen schlechteren Gesamtwirkungsgrad zur Folge.

spcqike, 2024-06-06:
> Das hätte aber vermutlich einen schlechteren Gesamtwirkungsgrad zur Folge.

nicht nur das. Es würde auch die Zyklus Zeit erhöhen da immer nur ein wechselrichter zu einer Zeit angesprochen werden kann. 

Regelt man zwei wechselrichter gleichzeitig müssen auch beide angesprochen und abgefragt werden. 

Das Thema „mehrere wechselrichter im DPL“ wurde schon öfter gewünscht und diskutiert. 

Eigentlich braucht es keinen zusätzlichen issue 
https://github.com/helgeerbe/OpenDTU-OnBattery/issues/230

gitisgreat2023, 2024-06-06:
@genmad has a solution based on a shelly, capable of API 2 or higher, creating virtual power meters that multiple DTUs (on Battery) can use.

See here for the code https://github.com/genmad/shellyAPIGen2AndAboveScripts

closed 2024-11-17 by schlimmchen