hoylabs/OpenDTU-OnBattery #1033: prometheus api runs out of resources.

rudolfp1979, 2024-06-06:
### What happened?

when calling the prometheus api from browser after some time it runs out of ressources and tells Too Many Requests even if no requests where made active

### To Reproduce Bug

no information

### Expected Behavior

no information

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.06.03

### Relevant log/trace output

```Shell
13:53:16.440 > Calling /api/prometheus/metrics has temporarily run out of resources. Reason: "std::bad_alloc".
```

rudolfp1979, 2024-06-07:
Found out that this happens when to much tabs in browser are open showing the main site. So this can be closed

schlimmchen, 2024-08-18:
> Found out that this happens when to much tabs in browser are open showing the main site. So this can be closed

Thanks for your feedback. Your observation makes sense, as every connection to the OpenDTU websocket(s) use resources.

closed 2024-08-18 by schlimmchen