hoylabs/OpenDTU-OnBattery #1035: [Request] Limit / regulate MPPTs differently or calculate limit-percentage by missing total power

Dozer-hh, 2024-06-07:
### Is your feature request related to a problem? Please describe.

I have an HMS-2000-4T with four PV modules of 435 W each. Two modules have a south-east orientation and two modules have a south-west orientation, so all four modules never have the same power applied to the power inverter.
I currently have no storage, so I feed the full power up to the permitted 800 W limit to my house.

If I set the upper power limit to 800 W as a fixed value, the power inverter is set to a limit of 40% and therefore each MPPT is limited to 200 W of the possible 500 W.

If, for example, the 4 modules can produce a total of 1000 W at a time, of which modules 1 and 2 (south-east) each produce (or better could produce) 400 W and modules 3 and 4 (south-west) each can only produce 100 W at that time, the real feed-in is limited to a total of 600 W by the limit of 40% per MPPT (200 W each from M1 and M2 and 100 W each from M3 and M4). This means that 200 W is lost.

### Describe the solution you'd like

As a solution I wold like the OpenDTU-OnBattery firmware to be adapted, so that the individual MPPTs of the inverter can be limited / regulated differently in order to cumulatively reach the upper power limit?

Or (perhaps easier) to let the DTU recognize / calculate, how many % of the individual MPPTs cumulatively lead to the defined upper power limit and from this to the needed total percentage limit at that time?

If OpenDTU-OnBattery would be able to recognize, that in order to reach the desired upper power limit of 800 W, it should have to compensate the “missing” 200 W by adjusting the percentage limit, this would be a great gain to the system.

schlimmchen, 2024-06-07:
> so that the individual MPPTs of the inverter can be limited / regulated differently

That's not possible. The inverter knows only one limit, which is the total limit, and the DTU has no control over the individual MPPTs.

In #956 a feature to overscale the limit to compensate shading (or different power inputs) is already in the making.

Since I don't see what this request is asking other than what is already asked in #917 and in the making in #956, I am closing this issue.

Dozer-hh, 2024-06-07:
Ok, thanks for the link to the existing issues, I'll give it a closer look. On first sight this seems to match with my request.