hoylabs/OpenDTU-OnBattery #1037: [Request] VE.Direct MPPT: Process "RELAY" message

schlimmchen, 2024-06-07:
### Describe the solution you'd like

Process this info received over VE.Direct from a charge controller with relay output:

```
22:21:22.976 > [VE.Direct MPPT 10/9] Unknown text data 'RELAY' (value 'OFF')
```

ingrimsch, 2024-07-06:
My MPPT 250/85 sends an additional "FWE" message, that is also not processed yet.
Example: [VE.Direct MPPT 34/-1] Unknown text data 'FWE' (value '315FF')

According to Victrons Protocol description this is the Firmware Version as 24Bit Value.

Citing https://www.victronenergy.com/upload/documents/VE.Direct-Protocol-3.33.pdf:
"The firmware version of the device. The version contains up to 6 digits, 0 padding on the left side is 
not mandatory. Examples: 0208FF or 208FF for firmware version 2.08 (last digit FF indicates an 
official release), 020801 for firmware version 2.08-beta-01."

SW-Niko, 2024-11-04:
Hello @ingrimsch ,
I have a question regarding the "FWE" message. 
According the documentation (VE.Direct-Protocol-3.33.pdf) from Victron: "FWE" is for "Phoenix Charger" and "Smart BuckBoost" only.

[image]

Can you verify again?

SW-Niko, 2024-11-04:
Hello @schlimmchen ,
regarding the "RELAY" message. I can implement the missing functions, (web view, MTTQ and HASS) in the same way like the "IR" / "LOAD" messages, if you like.

[image]

schlimmchen, 2024-11-06:
Yes, please, that ist much appreciated.

ingrimsch, 2024-11-18:
> Hello @ingrimsch , I have a question regarding the "FWE" message. According the documentation (VE.Direct-Protocol-3.33.pdf) from Victron: "FWE" is for "Phoenix Charger" and "Smart BuckBoost" only.

> Can you verify again?

Hi Niko,

sorry for my late response. Yes, I can confirm the FWE message represents the current FW version for my Victron MPPT 250|85 (identified as "SmartSolar MPPT VE.Can 250|85 rev2"). Firmware version is currently shown as "Firmware-Version: ." on the Live-View  page.

This is the message I see in the "Konsole" log:
11:14:04.324 > [VE.Direct MPPT 34/-1] Unknown text data 'FWE' (value '315FF')

SW-Niko, 2024-11-18:
Thanks  @ingrimsch,
it seems the Victron documentation is outdated or wrong.

>Firmware version is currently shown as "Firmware-Version: ." 

On your device Victron do not send "FE" Data. Now I understand why you want to process FWE.
How do do connect the CAN? Converter to UART?

@schlimmchen 
How about to process FW and FWE. If FE is not available we use FWE and ignore the 2 last digit.

>(value '315FF')

Means firmware 3.15

closed 2024-11-20 by schlimmchen