hoylabs/OpenDTU-OnBattery #1038: Hoymiles HM800 Blinkt abwechselnd rot und grün im ungefähr 5 sec takt

Solar-Newbie, 2024-06-07:
### What happened?

heute 07.06.2024 konnte ich ein mir unbekanntes Phänomen beobachten. Seit ca 19:00 blinkt der WR abwechselnd Rot und grün im 5-10 Sek-Takt( seitdem habe ich es bemerkt). Alle Komponenten ausgeschaltet, neu gestartet, brachte keine Abhilfe. Erst als ich die Version 07.05.2024 wieder eingespielt habe, lief der WR wieder durchgehend im grünen Modus.
Leider habe ich mir nicht die Fehlerprotolle der Konsole mitgeschnitten. 
Aufbau: 2x 430 Wp in Reihe an Victron 100/20, Pylontech us2000c , Opendtu mit ESP32 (kein S3 od. wertiger) .
Sollte unter dieser Konstellation gewünscht sein , den Fehler zu reproduzieren, werde ich es zeitnah erledigen.
mfg

### To Reproduce Bug

Keine Ahnung

### Expected Behavior

Kenntnisnahme, Beobachtung

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

03.06.2024, generic_esp32, kONFIGURATIONSVERSION 0.1.28

### Relevant log/trace output

```Shell
nicht mitgeschnitten
```

Solar-Newbie, 2024-06-07:
ESP32-D0WD-V3
wird als Chip-Model angezeigt

closed 2024-06-07 by Solar-Newbie

spcqike, 2024-06-07:
Ich denke nicht, dass es ein openDTU Problem ist. Was sollte die DTU machen? 

Laut Handbuch ist das abwechselnd grüne und rote blinken ein „Firmware Problem“

[image]

Vielleicht hat sich der wechselrichter irgendwie aufgegangen und einen Zustand erreicht, aus dem er von allein nicht heraus kam. Und dein kompletter Neustart mit allem drum und dran hat es halt gelöst.

Solar-Newbie, 2024-06-07:
**Problem taucht nun auch unter der älteren Version auf. Hier ein Auszug der Meldungen von der Konsole.**
Websocket: [/livedata][4] disconnect
20:06:29.180 > Websocket: [/vedirectlivedata][4] disconnect
20:06:29.246 > Websocket: [/batterylivedata][4] disconnect
20:06:30.154 > [Pylontech] Alarms: 0 0 0 0 0 0 020:06:30.249 > [Pylontech] Warnings: 0 0 0 0 0 0 020:06:30.351 > [Pylontech] chargeVoltage: 53.200001 chargeCurrentLimitation: 25.000000 dischargeCurrentLimitation: 25.00000020:06:30.401 > [Pylontech] soc: 72 soh: 10020:06:30.547 > [Pylontech] voltage: 49.500000 current: -6.100000 temperature: 24.00000020:06:30.610 > [Pylontech] chargeStatusBits: 1 1 020:06:30.674 > [Pylontech] Manufacturer: PYLON   20:06:32.121 > [Pylontech] Alarms: 0 0 0 0 0 0 020:06:32.506 > [Pylontech] Warnings: 0 0 0 0 0 0 020:06:32.569 > [Pylontech] chargeVoltage: 53.200001 chargeCurrentLimitation: 25.000000 dischargeCurrentLimitation: 25.00000020:06:32.641 > [Pylontech] soc: 72 soh: 10020:06:32.694 > [Pylontech] voltage: 49.450001 current: -7.400000 temperature: 24.00000020:06:32.754 > [Pylontech] chargeStatusBits: 1 1 020:06:32.861 > Fetch inverter: 114190531163
20:06:33.146 > TX RealTimeRunData Channel: 75 --> 15 90 53 11 63 90 53 11 63 80 0B 00 66 63 4C 17 00 00 00 00 00 00 00 00 1C F1 2D 
20:06:33.351 > Interrupt received
20:06:33.439 > RX Channel: 23 --> 95 90 53 11 63 90 53 11 63 01 00 01 01 EE 01 83 07 76 01 EF 01 78 07 41 00 02 5A | -80 dBm
20:06:33.517 > Interrupt received
20:06:33.620 > RX Channel: 23 --> 95 90 53 11 63 90 53 11 63 02 E6 D8 00 02 7B DA 00 79 00 75 09 2D 13 89 0E 03 B5 | -80 dBm
20:06:33.722 > Interrupt received
20:06:33.830 > RX Channel: 40 --> 95 90 53 11 63 90 53 11 63 83 00 03 00 99 03 E8 01 23 00 55 49 6F 36 | -80 dBm
20:06:33.916 > PowerMeterClass: TotalPower: -8.24
20:06:34.033 > [DPL::loop] ******************* ENTER **********************
20:06:34.093 > [DPL::loop] battery interface enabled, SoC: 72 %, StartTH: 45 %, StopTH: 15 %, SoC age: 1 s, ignore: no
20:06:34.155 > [DPL::getBatteryVoltage] BMS: 49.45 V, MPPT: 49.46 V, inverter: 49.70 V, returning: 49.45V
20:06:34.210 > [DPL::loop] dcVoltage: 49.45 V, loadCorrectedVoltage: 49.46 V, StartTH: 52.00 V, StopTH: 47.00 V
20:06:34.277 > [DPL::loop] StartTH reached: yes, StopTH reached: no, SolarPT enabled, use at night: no
20:06:34.351 > [DPL::calcPowerLimit] battery use allowed, solar power (DC): 0 W
20:06:34.463 > [DPL::calcPowerLimit] target consumption: -5 W, base load: 12 W, power meter does NOT include inverter output
20:06:34.518 > [DPL::calcPowerLimit] power meter value: -8 W, power meter valid: yes, inverter output: 9 W, solar power (AC): 0 W
20:06:34.584 > [DPL::calcPowerLimit] match household consumption with limit of -3 W
20:06:34.887 > [DPL::setNewPowerLimit] input limit: -3 W, min limit: 10 W, max limit: 500 W, hysteresis: 5 W
20:06:35.398 > [DPL::announceStatus] calculated limit is less than minimum power limit
20:06:35.577 > [DPL::updateInverter] Stopping inverter...
20:06:35.678 > RX Period End
20:06:35.786 > Success
20:06:35.842 > [DPL::announceStatus] waiting for a start/stop/restart command to complete
20:06:35.903 > TX AlarmData Channel: 3 --> 15 90 53 11 63 90 53 11 63 80 11 00 66 63 4C 17 00 00 00 00 00 00 00 00 C6 EA F6 
20:06:35.966 > Interrupt received
20:06:36.079 > RX Channel: 23 --> 95 90 53 11 63 90 53 11 63 01 00 01 B0 01 00 01 4A 67 4A 67 00 00 00 00 70 7C 29 | -80 dBm
20:06:36.187 > Interrupt received
20:06:36.634 > RX Channel: 23 --> 95 90 53 11 63 90 53 11 63 02 00 39 53 FA 54 04 00 00 00 00 70 7C 00 3B 54 0F 3B | -80 dBm
20:06:36.839 > [Pylontech] Alarms: 0 0 0 0 0 0 020:06:37.042 > [Pylontech] Warnings: 0 0 0 0 0 0 020:06:37.242 > [Pylontech] chargeVoltage: 53.200001 chargeCurrentLimitation: 25.000000 dischargeCurrentLimitation: 25.00000020:06:37.822 > [Pylontech] soc: 72 soh: 10020:06:37.887 > [Pylontech] voltage: 49.430000 current: -7.300000 temperature: 24.00000020:06:37.950 > [Pylontech] chargeStatusBits: 1 1 020:06:38.018 > Interrupt received
20:06:38.076 > RX Channel: 40 --> 95 90 53 11 63 90 53 11 63 03 54 1A 00 00 00 00 70 7C 00 3D 54 24 54 2F 00 00 E2 | -80 dBm
20:06:38.221 > Interrupt received
20:06:38.289 > RX Channel: 40 --> 95 90 53 11 63 90 53 11 63 04 00 00 70 7C 00 3F 54 3A 54 44 00 00 00 00 70 7C D0 | -80 dBm
20:06:38.342 > Interrupt received
20:06:38.395 > RX Channel: 40 --> 95 90 53 11 63 90 53 11 63 05 00 41 54 51 54 65 00 00 00 00 70 7C 00 43 54 6F 91 | -80 dBm
20:06:38.475 > Interrupt received
20:06:38.535 > RX Channel: 40 --> 95 90 53 11 63 90 53 11 63 06 54 7C 00 00 00 00 70 7C 00 45 54 86 54 99 00 00 ED | -80 dBm
20:06:38.627 > Interrupt received
20:06:38.718 > RX Channel: 23 --> 95 90 53 11 63 90 53 11 63 07 00 00 70 7C 00 47 55 15 55 26 00 00 00 00 70 7C E6 | -80 dBm
20:06:38.805 > Interrupt received
20:06:38.896 > RX Channel: 40 --> 95 90 53 11 63 90 53 11 63 08 00 49 55 38 55 43 00 00 00 00 70 7C 00 4B 55 4D F0 | -80 dBm
20:06:38.984 > Interrupt received
20:06:39.080 > RX Channel: 40 --> 95 90 53 11 63 90 53 11 63 09 55 58 00 00 00 00 70 7C 00 4D 55 63 55 6D 00 00 DE | -80 dBm
20:06:39.173 > Interrupt received
20:06:39.264 > RX Channel: 23 --> 95 90 53 11 63 90 53 11 63 0A 00 00 70 7C 00 4F 55 78 55 83 00 00 00 00 70 7C 2B | -80 dBm
20:06:39.358 > Interrupt received
20:06:39.420 > RX Channel: 40 --> 95 90 53 11 63 90 53 11 63 0B 00 51 55 8F 55 A7 00 00 00 00 70 7C 00 53 55 AD 40 | -80 dBm
20:06:39.485 > Interrupt received
[... cut ...]
20:10:35.848 > RX Period End
20:10:35.990 > Middle missing
20:10:36.048 > Request retransmit: 3
20:10:36.114 > TX RequestFrame Channel: 61 --> 15 90 53 11 63 90 53 11 63 83 96 
20:10:36.179 > RX Period End
20:10:36.236 > Middle missing
20:10:36.307 > Request retransmit: 3
20:10:36.366 > TX RequestFrame Channel: 75 --> 15 90 53 11 63 90 53 11 63 83 96 
20:10:36.413 > Interrupt received
20:10:36.463 > RX Channel: 75 --> 95 90 53 11 63 90 53 11 63 03 55 43 00 00 00 00 70 7C 00 4B 55 4D 55 58 00 00 D2 | -80 dBm