hoylabs/OpenDTU-OnBattery #1045: MQTT subscriptions not updated when base topic changes

schlimmchen, 2024-06-13:
### What happened?

Saw 918c3449da4dec639b9bfbb9846153b272a8ec4f (upstream).

### To Reproduce Bug

Change the base topic in the MQTT setting. Notice that publishing to DPL topics using the new base does nothing as the respective software is still subscribed to the topic with the old base. True for other pieces of software as well (which?).

### Expected Behavior

When changing MQTT settings, the new settings should work as expected, i.e., OpenDTU-OnBattery shall be subscribed using the new settings.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

123456

TomBrett72, 2024-06-20:
Hi @schlimmchen,
thank you for reporting this. I see also a cycletime problem. I set up the global MQQT cycle time to 20s. But */powermeter/# values are sent in 10s intervall. I already made a restart of the DTU and try setup other values, f.i. 15s. No change. For monitoring I use a simple android MQQT-dashboard and therefor I need the same cycletime for all values. No big bug, but it would ne nice to have other then 10s intervall. Thank´s!
[image]

schlimmchen, 2024-06-20:
Please, don't hesitate to open a *new* issue. You comment has nothing to do with this issue.

AndreasBoehm, 2024-07-31:
Could this commit fix the issue?
https://github.com/skippermeister/OpenDTU-OnBattery/commit/f7fc22d64dd8fbf1e7f1511c95b5eb40ea752105

schlimmchen, 2024-07-31:
> Or did you mean to say that that we need to add subscribe and unsubscribe to all MQTT handlers to make it work?

Yes.

Did I link the upstream commit? Thomas did a fix to address this, but only for the upstream pieces of software.

I dont understand why the linked commit in skippermeister repo does. If it is important, it obviously needs to go into the upstream.

commit 184c6591 (2023-06-22): Added new found HMS-400 to DevInfoParser

Fixes #1045
closed 2024-08-20 by schlimmchen