hoylabs/OpenDTU-OnBattery #1046: Set temporary Limit to zero, when the Inverter is switched off by dynamic power limiter

dragricola, 2024-06-14:
### What happened?

(in English see below)
Wenn der Inverter durch den dynamic power limiter abgeschaltet wird, bleibt das letzte Limitwert bestehen. Das führt dazu, dass beim Wiedereinschalten der Inverter mit einem unpassenden Wert gestartet wird. Insbesondere, wenn zuvor ein Verbraucher mit hoher Leistung ausgeschaltet wird, steht das letzte Limit auf einem  sehr hohen Wert. Wenn nun der Inverter wieder gestartet wird, weil z.B. sich der PV-Ertrag leicht verschlechtert hat, speist der Inverter aus der Batterie zunächst eine hohe Leistung ein, die zu negativer Netzleistung führen kann, was wiederum gegebenenfalls den AC-Lader startet. Dann arbeitet der Inverter gegen den Lader und die Leistung fließt im Kreis. 

When the inverter is switched off by dynamic power limiter, the last actual limit consists. That leads to an unsuitable power value, when the inverter is restarted. Especially when before a power consumer was switched off, actual limit is high. On restart the inverter produces possibly to much power, which may lead to negative line power that may start the AC-charger. Then inverter works against AC-charger and the power flows in circle.   

### To Reproduce Bug

[image]
Der Screenshot zeigt einen Zustand mit ausgeschaltetem HM1200. Das aktuelle Limit steht jedoch noch auf 1022W, weil zuvor ein Verbraucher ausgeschaltet wurde. Der Huawei AC-Charger lädt den PV-Überschuss in die Batterie, sodass nur 33 W ins öffentliche Netz eingespeist werden. Verschwindet der PV-Überschuss nun wieder und die Netzleistung wird langsam positiv und überschreitet den Schwellwert, dann schaltet der dynamic power limiter den Inverter mit 1022 W ein und erzeugt dann wieder eine sehr hohe Netzeinspeisung, die durch den dynamic power limiter erst weggeregelt werden muss, sofern ein wieder startender AC-Lader das nicht stört.

The screenshot shows a state with HM1200 switched off. The actual imit is 1022 W. Huawei AC Charger works so that only 33 W is supplied to publich line.  When the line power becomes slowly positive and crosses the threshold inverter is switched on with 1022 W and produces high infeed to public line. 

### Expected Behavior

Der dynamic power power limiter sollte zum Ausschalten der Inverters nicht nur das Power-Kommando auf Null setzen sondern auch das temporäre absolute und/oder relative Limit auf Null setzen. Dann gibt es beim Wiedereinschalten keinen zu hohen Leistungswert.

When dynamic power limiter switches off the inverter not only the power command but also the temporary absolute and/or relative power limit should be set to zero. 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.06.03

dragricola, 2024-06-14:
> Angenommen, der HM-1200 steht auf voller Leistung und nach dem Starten durch den DPL braucht es 30 Sekunden, bis herunter geregelt wurde. in der Zeit speist er also 10Wh ein (0.01kWh oder 0.3Cent). Das ist ein Problem?

Nein, das ist nicht das Problem. Ursprünglich wollte ich in meiner Home-Automatisierung den MQTT-Wert status.limit_absolute/ verwenden und habe gesehen, dass er bei ausgeschaltetem Inverter nicht null ist (siehe Screenshot). Da kam mir der Verdacht, dass die von mir beobachteten störenden Schwingungen zwischen Inverter und Charger hierauf zurückzuführen sind. Jetzt bin ich mir aber nicht mehr sicher, ob das stimmt oder ob das von mir beobachtete Ereignis vielleicht auf einen anderen Störeinfluss zurückzuführen ist. 
 
> Ich sehe es eher problematisch, dass der AC Charger lädt, obwohl aus der Batterie unterstützt werden muss. Wäre es nicht sinnvoller, dort anzusetzen und den AC Charger zu stoppen, wenn aus der Batterie eingespeist wird?
 
Mein Problem ist eher umgekehrt, dass der Inverter noch arbeitet, wenn der Charger schon lädt.
 
> Wie oft kommt das denn vor? wie viel Energie wird damit wirklich verschwendet? 

Wie gesagt, Energieverschwendung ist vernachlässigbar und ich muss zugeben, dass ich nun, nachdem ich gezielt danach gesucht habe, nicht viele Beispiele in meiner Datenbank gefunden habe.

> Warum startet der AC Charger, wenn aus der Batterie eingespeist wird?

Der Charger startet, wenn mehr als 50W ins Netz eingespeist wird. Ein Hochsetzen des Limits ist im Sommer sicher kein Problem, weil über den Tag auch bei schlechtem Wetter die Batterie meist voll wird. Im Winter würde allerdings mehr Energie ungenutzt ins Netz gehen. Probehalber werde ich das Limit mal hochsetzen.
Da ich mir nicht mehr sicher bin, ob die Schwingungen auf den eigentlich veralteten "aktuellen" Wert  "limit_absolute" zurückzuführen sind, käme es auf einen Versuch an. Es müsste ja nur dieser eine Wert auf Null zurückgesetzt werden, immer wenn der Inverter ausgeschaltet wird. Wenn mir jemand sagen kann, an welcher Stelle des Codes und mit welchem Befehl das geschehen kann, würde ich die Änderungen mal selbst programmieren und testen. 
Ich wäre einverstanden, das Label von bug auf enhancement zurückzusetzen, da es ja zumindest schön wäre, wenn status.limit_absolute den Status korrekt widerspiegeln würde.

schlimmchen, 2024-06-15:
> Wenn der Inverter durch den dynamic power limiter abgeschaltet wird, bleibt das letzte Limitwert bestehen. Das führt dazu, dass beim Wiedereinschalten der Inverter mit einem unpassenden Wert [...]

Den Teil habe ich gelesen und verstanden. Alles danach habe ich nur überflogen. Es scheint nämlich so, als sei das in diesem Absatz einleitend beschriebene Problem die Ursache für alles andere hier diskutierte. Ist das richtig?

Dieses einleitend beschriebene Problem gibt es meines Wissens nach nicht.

Es stimmt zwar, dass der Inverter "neuerdings" (ich hab das vor wenigen Monaten umgebügelt) auf dem alten Limit "stehen bleibt" und nur ausgeschaltet wird, wenn er nicht mehr gebraucht wird. Das große Aber ist folgendes: Beim Wiedereinschalten schickt der DPL immer erst das neue Limit, wartet bis es angekommen ist, und startet erst danach den Inverter.

Bitte überprüfe das nochmal genau. Ist das Verhalten so, wie ich es glaube programmiert zu haben, oder produziert der Inverter tatsächlich hohe Leistung beim Wiedereinschalten wenn er mit hohem Limit ausgeschaltet wurde?

dragricola, 2024-06-16:
> Bitte überprüfe das nochmal genau. Ist das Verhalten so, wie ich es glaube programmiert zu haben, oder produziert der Inverter tatsächlich hohe Leistung beim Wiedereinschalten wenn er mit hohem Limit ausgeschaltet wurde?

@schlimmchen  Da habe ich mich offensichtlich geirrt indem ich das Verhalten beim Einschalten fälschlicherweise auf das noch nicht zurückgesetzete Limit zurückgeführt habe. Das hatte ich ja auch schon bei meiner Antwort an @spcqike angedeutet. Ich habe mir nun mal genauer die aufgezeichneten Werte in meiner Datenbank angeschaut: Die geschilderten Probleme treten erst auf, seitdem ich zusätzlich per ioBroker-Script über MQTT einen zweiten Wechselrichter bedarfsweise ein- und ausschalte. Das ist der Ersatz für die Regelung, die ich als enhancement in https://github.com/helgeerbe/OpenDTU-OnBattery/issues/1032 zur Realisierung direkt in OpenDTU-OnBattery vorgeschlagen hatte.  Offensichtlich führt der Umweg über die MQTT-Kommunikation zu Verzögerungen und damit zu Störungen in der Regelung des ersten Wechselrichters. Wenn das direkt in OpenDTU-OnBattery gemacht würde, könnte diese zusätzliche Totzeit in der Regelung entfallen und das Regelverhalten würde sicher auch mit dem Zuschalten eines zweiten Wechselrichters wieder stabil.

Unabhängig davon: Was spricht dagegen, das Limit wieder wie vorher auf Null zu setzen, wenn der Inverter ausgeschaltet wird? Ich verwende diesen Wert auch in meiner Visualisierung, die dann nicht mehr den Inverterzustand immer korrekt anzeigt.

schlimmchen, 2024-06-16:
Der Grund ist, dass dann der RF Kanal dann belegt ist damit ein Limit zu senden, das keine Effekt hat. Wenn der WR nur kurz aus sein soll, muss man dann erstmal warten ein obsoletes Limit einzustellen, das man sofort danach wieder ersetzen muss.

dragricola, 2024-06-16:
OK, das leuchtet ein. Aber ist die nötige Wartezeit wirklich so groß, dass es stört und muss der WR wirklich so oft sofort wieder eingeschaltet werden?

schlimmchen, 2024-06-20:
> Ich verwende diesen Wert auch in meiner Visualisierung, die dann nicht immer den Inverterzustand korrekt anzeigt.

Dann musst du bei der Visualisierung bitte berücksichtigen, ob der Inverter überhaupt produziert bzw. ob er eingeschaltet ist. Auch das vorherige Verhalten war so, dass zumindest das "lower power limit" eingestellt wurde. Ein Limit von 0% einstellen werden wir nicht, weil wir schon wissen, das ein zu kleines Limit zu Problemen führen kann, insb. zum Aufschwingen und zur Selbstabschaltung.

Die Anforderung "wir müssen bitte die Realität ändern, damit meine Visualisierung passt" gefällt mir ehrlich gesagt überhaupt nicht. Die Visualisierung folgt der Realität, nicht andersherum.

schlimmchen, 2024-12-02:
Das eigentliche Begehren des Issues will ich nicht umsetzen. Die Begründung bzgl. der Visualisierung finde ich unzureichend. Die Nachteile habe ich erläutert.

> Die geschilderten Probleme [...] treten erst auf, seitdem ich zusätzlich per ioBroker-Script über MQTT einen zweiten Wechselrichter bedarfsweise ein- und ausschalte. Das ist der Ersatz für die [...]

Das klingt so, als würde der Support für mehrere Inverter im DPL, der im letzten Release hinzugekommen ist, das Problem lösen.

Daher schließe ich das Issue als "not planned". Bitte nochmal melden, falls ich einen Aspekt übersehen habe.

closed 2024-12-02 by schlimmchen