hoylabs/OpenDTU-OnBattery #1052: Add RS485 Pylontech support

Manos1966, 2024-06-16:
### Is your feature request related to a problem? Please describe.

Currently threre is CANbus Support for Pylontech & other Batteries
The information CAN provides is good but limited compared to the information available from the RS485 port:
**CANbus Information shown currently:**

[image]

**Information available when using the RS485 port:**

[image]

It is obvious that the RS485 port is giving more information compared to existing CANbus:
- It handles each battery separately
- gives information on temperature from every sensor inside the battery
- shows the Voltage of each cell inside the battery
- shows the Balancing situation (mV)
- can be used combined with the CANbus connection

The software module for OpenDTU-OnBattery has been already developped #622 
by skippermeister https://github.com/skippermeister/OpenDTU-OnBattery

### Describe the solution you'd like

Add RS485 support to the OpenDTUonBattery in order to make the additional information available.

Additional Information: Both CANbus and RS485 are working at the same time = **There is no requirement to change the existing programming which uses the SOC% data from the CANbus.**

A solution exists already, monitoring multiple Pylontech Batteries:
#622  adding RS485 Pylontech support

https://github.com/skippermeister/OpenDTU-OnBattery

### Describe alternatives you've considered

Use a separate ESP32 with the Fork created by skippermeister
https://github.com/skippermeister/OpenDTU-OnBattery
Both OpenDTUs can be online, collecting information from the Pylontech Battery/ies, using the CANbus and RS485 connection at the same time.

Matti07, 2024-06-23:
Hi @Manos1966, I think that's a very, very good suggestion /reminder.
The display of the cell voltages alone is a breakthrough for me.
But who is able and has the time to implement this?
At first glance, the only people I can think of are the usual suspects, in this case @schlimmchen and @skippermeister.
But how do we convince them to install it?
IIRC the Fusion Board 2.X even has RS485.

schlimmchen, 2024-06-23:
This is indeed a good idea. I am not available to implement this for the foreseeable future. I don't own such hardware and these boxes are too expensive to buy for fun. If I could get my hands on a broken one that I can repair or where I can hook up replacement cells, that would at least enable me to test the feature. Doing stuff like this blind is not rewarding. Moreover, I simply have other goals right now, and I am already spending all the time I can possibly spare with this project.

Tiese, 2024-06-24:
Hm &#129300;

Was nützen all diese Informationen, wenn sie dann doch nicht stimmen...
[image]

Oder liege ich da falsch?

Manos1966, 2024-06-24:
@Tiese 
Berechnungsfehler schon korrigiert...

[image]

[image]

Tiese, 2024-06-24:
@Manos1966 
Ich dachte schon, ich steh irgendwie auf dem Schlauch.

Aber dennoch: "Hat die US5000 wirklich nur 100Ah Kapazität?"

Manos1966, 2024-06-24:
Ich verstehe die Frage nicht @Tiese was sollte sie den haben??? es ist eine 4,8kWh Batterie

Tiese, 2024-06-24:
@Manos1966 
Ich Dussel, habe die 4,8kWh mit den 100Ah verwechselt.
Sowas passiert mir leider öfter. &#129322;

Matti07, 2024-06-25:
Moin @Manos1966, sag mal, könntest Du das eventuell in den Hauptzweig integrieren?
_C++ ist leider überhaupt nicht meins und als Rentner will ich mir nicht noch eine Sprache ans Bein binden._

Manos1966, 2024-06-25:
Wenn ich das machen könnte @Matti07 dann hätte ich nicht selber die Anfrage gestellt 😢

skippermeister, 2024-06-26:
siehe meinen letzten Kommentar von heute in #622