hoylabs/OpenDTU-OnBattery #1070: DTU web interface hängt sich auf

RealNBB, 2024-06-24:
### What happened?

Sobald ich eine eigene pin_mapping.json einspiele, verabschiedet sich der Netzwerkzugriff (WLAN?) der OpenDTU-onBattery. Es hilft nur ein neues flashen des ESP32-S3.
Danach: Gleiches Spiel von vorne. 

Meine pin_mapping.json:

```
[
    {
        "name": "Victron&Battery", 
        "battery": {
            "rx": 27,
            "rxen": -1,
            "tx": 14,
            "txen": -1
        },
        "victron": {
            "rx": 22,
            "rx2": -1,
            "rx3": -1,
            "tx": -1,
            "tx2": -1,
            "tx3": -1
        }
    }
]
```

Ich wei0ß wirklich nicht, was ich falsch mache.

### To Reproduce Bug

bei jedem Versuch, eine pin_mapping.json einzuspielen

### Expected Behavior

...

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

5ab6fe9 

### Relevant log/trace output

```Shell
...
```

### Anything else?

...

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

schlimmchen, 2024-06-24:
Bist du sicher nicht die `config.json` zu überschreiben beim Upload? Passiert mir ab und zu immer noch...

RealNBB, 2024-06-24:
Danke erstmal für die schnelle Antwort.

Ich wähle definitiv die pin_mapping.json aus.

ABER: Jetzt habe ich zuerst nach einem clean install die pin_mapping.json überspielt --> ESP startet danach. NAch dem obligatorischen rebbot dann den Schalter von VE.Direct auf aktiv gestellt...und zack...nichts geht mehr. Das gleich, wenn ich stattdessen Batterie wähle. Auch dann geht nichts mehr. Muss da im Vorfeld aktiv was angeschlossen sein, damit er die Settings schluckt.

Ectl. versuche ich es mal mit dem ESP32 ohne "S3". :)

schlimmchen, 2024-06-24:
> Muss da im Vorfeld aktiv was angeschlossen sein, damit er die Settings schluckt?

Nein.

Hast du das Board denn noch am PC angeschlossen? Kannst du mal mit der seriellen Konsole draufschauen, was da steht wenn das Problem auftritt?

Du lädst die pin mapping JSON hoch, das Board startet neu, dann wählst du das Profil im Device Manager und es passiert erstmal nichts? Erst, wenn du entweder die Batterieschnittstelle oder VE.Direct aktivierst?

Was für ein Board ist das denn? Sind die GPIOs 14, 22, und 27 tatsächlich frei verwendbar?

RealNBB, 2024-06-24:
Ich glaub, du hast mir gerade den entscheidenden Hinweis gegeben:

> Was für ein Board ist das denn? Sind die GPIOs 14, 22, und 27 tatsächlich frei verwendbar?

Board ist dieses hier:

https://de.aliexpress.com/item/1005006866671541.html?spm=a2g0o.order_list.order_list_main.4.2cf95c5fs0i6I2&gatewayAdapt=glo2deu

Kein 22 und auch kein 27! 

Oh je...wie peinlich! 

4 und 6 kann ich anstelle nehmen oder?

Das wär's jetzt! Ich versuche es in ein paar Minuten nochmal...muss aber kurz mal was essen.

Vielen vielen lieben Dank aber bis hierher! :)

schlimmchen, 2024-06-24:
22 und 27 können in der Tat nicht auf den Pinleisten sein, denn 22 gibt es gar nicht laut Datenblatt und 27 ist ein internet SPI pin. 4 und 6 sind definitiv verfügbar. 14 auch.

RealNBB, 2024-06-24:
Das war's! Hab jetzt für die Batterie 12 und 14 genommen, der Victron bekam die 6. Damit startet der ESP ohne Probleme!

Nochmal: Vielen vielen DANK Ohne deinen Hinweis wäre ich wohl verzweifelt! Habe ich ehrlich gesagt überhaupt nicht drauf geschaut! 

Jippiee! Jetzt kann es weiter gehen!

schlimmchen, 2024-06-24:
> Jippiee! Jetzt kann es weiter gehen!

Viel Spaß :rocket:

closed 2024-06-24 by schlimmchen