hoylabs/OpenDTU-OnBattery #1081: Wechselrichter startet oft nicht

Cartman586, 2024-07-03:
### What happened?

Hallo, wenn Strom eingespeist werden soll, dann speist der Wechselrichter (Hoymiles HM-1200) oft nicht ein, sondern beginnt irgendwann mal. Das Gerät hängt an einen LifePo4 Akku 25,6V 100Ah. Oft steht auch die Meldung durch Fernsteuerung abgeschaltet. Ich komme hier nicht weiter, danke 

### To Reproduce Bug

Einspeisen

### Expected Behavior

Speist bei Netzbezug ein 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.06.03

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

spcqike, 2024-07-03:
Sry, keine Logs, keine sinnvolle Hilfe. 

Die Meldung „durch Fernsteuerung abgeschaltet“ sagt, dass die openDTU den wechselrichter stoppt. Warum auch immer sie das tut. Ich vermute eine falsche Konfiguration. 

Was hast du für Einstellungen im DPL? Und wie gesagt, Logs sind zwingend nötig.

Manos1966, 2024-07-03:
https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Wie-man-nach-L%C3%B6sungen-sucht%2C-Fragen-stellt%2C-Probleme-meldet#wie-man-nach-l%C3%B6sungen-sucht-fragen-stellt-probleme-meldet

Solar320, 2024-07-13:
Ich habe heute die letzte Änderung in die DTU eingespielt, konnte aber gestern auch wider beobachten das mein Hauptwechselrichter in den StandyBy geschickt wird, was er ausdrücklich nicht machen soll!!!
Ich vermute hier das es wegen dem minimalen limit von 10% gemacht wird, welches beim HM800 80W entsprechen würde.
Allerdings sehe ich auch das die Regelung den HM800 scheinbar trotzdem auf <10% herunter regeln kann.
Sollte es möglich sein den HM800 auf zB 1% herunter zu regeln dann wäre die Standbyschaltung in diesem Fall sogar nutzlos.
Ich habe es manuell bis 3% herunter laufen lassen, ich kann aber nicht sagen ob der Wechselrichter bei 3% dauerhaft eingeschaltet bleibt oder ob er sich irgendwann verabschieded. Bis jetzt läuft er jedenfalls stabil auf 3% ca 24w.

Das eigentliche Problem wurde leider, entgegen einiger Meinungen im Forum, immer noch nicht verstanden bzw. umgesetzt.  Wenn parallel zur DTU ein weiterer ungeregelter Wechselrichter läuft und folgende Situation auftritt dann wird der regelbare Wechselrichter in den Standby geschickt und verbleibt dort mehrere Minuten lang obwohl erneut eine Last anliegt.

(Nulleinspeisung, derzeit ohne Batterie) 

Die Situation ist folgende: Die Wohnung braucht um die Mittagszeit weniger Strom als der **ungeregelte Wechselrichter** erzeugt, dadurch steht am Stromzähler eine Minus bei der Leistung. Der regelbare Wechselrichter wird nun bis auf 10% herunter geregelt und viel zu schnell in den Standby geschickt. Im nächsten Moment schaltet sich zb. der Motor der Waschmaschine ein und verbraucht ca. 250Watt was zur Folge hat das in Summe (mit der Erzeugung des ungeregelten Wechselrichters) meinetwegen 150Watt aus dem Netz gezogen werden. In dieser Zeit wird Strom aus dem Netz gezogen obwohl er erzeugt werden könnte.
Der Grund ist das der HM-800 ewig lang braucht um hochzufahren.

Demzufolge wäre es wünschenswert das die Standbyschaltung entweder ganz abgeschaltet werden kann oder die Abschaltung erst nach eine Zeitspanne von gut 1-15Minuten erfolgt und zwar so das die Zeit immer wider auf null zurück gesetzt wird solange 
auch nur kurzzeitig ein positiver Strombezug registriert wird. Erst wenn die Waschmaschine länger als die eingestellte Zeitspanne inaktiv bleibt, also kein positiver Strombezug mehr auftritt, dann kann darüber nachgedacht werden den Wechselrichter in standby zu schalten. Natürlich bleibt die Regelungenauigkeit durch die Trägheit der Regelschleife bestehen aber man kann zumindest einen Teil der auftretenden Last kompensieren.

Mir würde es reichen wenn ich die Standbyschaltung manuell deaktivieren könnte, da das mit der einstellbaren Zeitspanne sicher extrem kompliziert ist weil das je nach Gerät (Waschmaschine, Geschirrspüler usw.) einfach zu stark variiert.

Übrigens läuft der WR immer noch auf 3%, ohne Geräteseitige Abschaltung.
[image]

Manos1966, 2024-07-13:
Hallo @Solar320 

eine Option "die Standbyschaltung ganz abgeschaltet werden kann" wuerde mir auch gefallen.

Aber vielleicht hilft dir dieser Loesung weiter:
https://github.com/genmad/shellyAPIGen2AndAboveScripts/blob/master/src/openDTUonBattery/virtualPowerMeters_oDoB.js

spcqike, 2024-07-13:
> Nulleinspeisung, derzeit ohne Batterie

solange der wechselrichter „solar only“ betrieben wird, ist die Mindestleistung genau das. Dann bleibt er dort stehen und speist ein. 

Hast du das überhaupt mal probiert? 

> Mir würde es reichen wenn ich die Standbyschaltung manuell deaktivieren könnte, da das mit der einstellbaren Zeitspanne sicher extrem kompliziert ist weil das je nach Gerät (Waschmaschine, Geschirrspüler usw.) einfach zu stark variiert.

Forken und die Stelle im Code auskommentieren. Sollte nicht zu schwer sein. 

> Demzufolge wäre es wünschenswert das die Standbyschaltung entweder ganz abgeschaltet werden kann oder die Abschaltung erst nach eine Zeitspanne von gut 1-15Minuten erfolgt und zwar so das die Zeit immer …..

so etwas in der Art ist angedacht. Zumindest haben wir in einem issue letztens darüber gesprochen.

Solar320, 2024-07-13:
"[spcqike]

Nulleinspeisung, derzeit ohne Batterie

solange der wechselrichter „solar only“ betrieben wird, ist die Mindestleistung genau das. Dann bleibt er dort stehen und speist ein.

Hast du das überhaupt mal probiert?"

-Nein, ich behaupte das natürlich nur einfach so, um die Entwickler zu ärgern?!?!?! ;p

Heute war es besonders einfach das zu testen. Die Wohnung hat weniger verbraucht als der ungeregelte Wechselrichter produziert hat. Wenn ich den Computer einschalte braucht die Wohnung mehr.
Ergebnis war, nachdem ich den Computer ausgeschaltet habe hat es nicht sehr lange gedauert bis der Wechselrichter ausgeschaltet wurde, die Statuszeile färbt sich dann gelb.
Danach habe ich den Computer wider eingeschaltet, um mehr zu verbrauchen, dann hat es geschlagene 7 Minuten gedauert bis sich der Wechselrichter wieder online geschaltet hat. In der zeit hab ich locker mit 100 - 200Watt aus dem Netz gezogen.

-Für mich ist das nicht so einfach, ich bin Anfänger was Programmieren betrifft.
 Ich habe das untere Limit jetzt auf 1% gesetzt  (nicht über die HTML Oberfläche) und hoffe das der Wechselrichter morgen in einer ähnlichen Situation nicht mehr ausgeht.

Ich gehe davon aus das der Limiter in der Lage ist den Wechselrichter von 1 bis 100% zu regeln und das er nur prüft ob die eingestellte (ermittelte) Leistung <10% ist, wenn das längere Zeit der Fall ist schickt er den Wechselrichter in Standby. Meine Hoffnung ist nun das er jetzt prüft ob der eingestellte Wert <1% ist und da es <1% vermutlich nicht gibt bleibt die Prüfung auf StandBy immer negativ und der Wechselrichter bleibt hoffentlich an. Mangels Sonne kann ich das heute nicht mehr ausprobieren.

spcqike, 2024-07-13:
> Ergebnis war, nachdem ich den Computer ausgeschaltet habe hat es nicht sehr lange gedauert bis der Wechselrichter ausgeschaltet wurde, die Statuszeile färbt sich dann gelb.

kann ich nicht bestätigen. 

Bei mir läuft es, wie es soll. 

Wenn es bei dir also nicht geht, weil der Code es nicht hergibt, muss bei mir wohl was falsch sein, dass es durch einen glücklichen Zufall geht wie gedacht, aber eben anders als programmiert … ein glücklicher Bug?

hast du den Haken „wechselrichter wird durch solarmodule gespeist“ aktiv? Wo ist dein log Auszug der uns nachvollziehbar zeigt, warum deine DTU den WR deaktiviert? 

Nimms bitte nicht persönlich, aber wenn ich und viele andere das Verhalten nicht nachvollziehen können, weil die Firmware arbeitet wie sie soll, muss das Problem eine andere Ursache haben. Und ohne Logs kann man nichts nachvollziehen. 

Und welche Version verwendest du? Vielleicht ist es ja wirklich n Bug im letzten stable oder das Feature ist bisher nur im Development branch … ich verwende die #1089 und damit geht es definitiv. Rutscht mein Bedarf unter das untere Limit, bleibt der wechselrichter beim unteren Limit aktiv.

Solar320, 2024-07-14:
Hallo, leider gibt es oft leute hier die behaupten es funktioniert wie es soll oder hast du dieses oder jenes gemacht, man dreht sich im Kreis.

Also mit #1089 kann ich direkt nichts anfangen, ich habe in meinen Systeminformationen auf Firmwareversion geklickt, da steht bei mir "b90da21" und ich komme auf dem folgenden Link heraus: https://github.com/helgeerbe/OpenDTU-OnBattery/commits/b90da21 Das ist eine Entwicklerversion.

Ich verwende VSC um die sachen zu kompilieren und aufzuspielen, weil ich zu Anfang eine eigene 8Mb Partition.csv für mein FusionV2 board gemacht habe und ich habe den PSRAM aktiviert. Inzwischen ist das Projekt aktuell und ich brauche den ganzen Firlefanz nicht mehr.  Bedeutet das ich bis auf das Board "opendtufusionv2" und  -DBOARD_HAS_PSRAM=1 nichts geändert habe.
Es wäre möglich das man letztere Option nicht mehr braucht, allerdings dürfte das das Verhalten des limiters nicht beeinflussen.
Meine Limitereinstellungen sind die hier.

Beim unteren limit steht jetzt 1% was ich über Umwege per config.json eingestellt habe um den Standby Modus zu umegehen...
falls es funktioniert. Testen kann ich es erst später am Tag.

[image]

Ich habe jetzt nur Bilder von nach dem Ausfall. Der Inverter wird gelb angezeigt und danach steigen die Limits stufenweise an aber der Inverter liefert nichts.  Erst bei ca 25% limit ist er wider auf Blau umgesprungen und hat dann angefangen die Nulleinspeisung auszuregeln. Leider habe ich davon kein Bild gemacht. Das letzte Bild zeigt mir nur den Moment wo der inverter wider auf Blau umspringt. Die ganze Prozedur hat 7 minuten gedauert, vom Ausfall bis zu dem Zeitpunkt wo der inverter wider blau wurde.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Die angezeigte Inverterleistung kam nur durch die zweite Balkonsolar zustande aber nicht durch den eigentlich zu regelnden Inverter. Wie gesagt leider habe ich vergessen einen screenshot zu machen als der Inverter dann wider eingespeist hat. Mal sehen ob ich es heute oder morgen schaffe zu testen ob ich die Standby schaltung erfolgreich austricksen konnte.
Leider gehen wir nachher noch Wandern und momentan bringt meine Frau das Stromnetz mit Mittagessen kochen durcheinander, da wird es schwer das zu testen.

Solar320, 2024-07-14:
Das ist jetzt bischen viel. man müsste vieleicht mal das log kopieren kur bevor der inverter offline geht, leider muss ich jetzt erstmal weg.

Hier ist der Inverter schon aus:

Websocket: [/livedata][5] disconnect
13:12:40.616 > Websocket: [/batterylivedata][5] disconnect
13:12:43.859 > Fetch inverter: 114173305377
13:12:43.912 > TX RealTimeRunData Channel: 75 --> 15 73 30 53 77 80 18 23 20 80 0B 00 66 93 B2 B6 00 00 00 00 00 00 00 00 BE 63 4E 
13:12:43.959 > Interrupt received
13:12:44.004 > RX Channel: 61 --> 95 73 30 53 77 73 30 53 77 01 00 01 01 75 00 03 00 0C 01 72 00 01 00 02 00 0A 94 | -80 dBm
13:12:44.050 > Interrupt received
13:12:44.098 > RX Channel: 23 --> 95 73 30 53 77 73 30 53 77 02 C1 A1 00 0A 71 B9 02 0E 01 EF 09 30 13 89 00 00 74 | -80 dBm
13:12:44.143 > Interrupt received
13:12:44.189 > RX Channel: 23 --> 95 73 30 53 77 73 30 53 77 83 00 00 00 00 00 00 01 B8 00 03 1F 32 81 | -80 dBm
13:12:44.405 > RX Period End
13:12:44.452 > Success
13:12:44.497 > [DPL::updateInverter] Starting inverter...
13:12:44.543 > TX PowerControl Channel: 3 --> 51 73 30 53 77 80 18 23 20 81 00 00 B0 01 9D 
13:12:44.589 > [DPL::announceStatus] waiting for a start/stop/restart command to complete
13:12:44.638 > Interrupt received
13:12:44.682 > RX Channel: 40 --> D1 73 30 53 77 73 30 53 77 81 00 00 00 00 24 00 74 | -80 dBm
13:12:46.443 > RX Period End
13:12:46.557 > Success
13:12:46.718 > [PowerMeterHttpJson] New total: 132.44
13:12:46.814 > [DPL::updateInverter] timeout (7 in succession), state transition pending: yes, limit pending: no
13:12:46.863 > [DPL::announceStatus] waiting for sufficiently recent inverter data
13:12:48.860 > Fetch inverter: 114184855236
13:12:48.910 > TX RealTimeRunData Channel: 23 --> 15 84 85 52 36 80 18 23 20 80 0B 00 66 93 B2 BB 00 00 00 00 00 00 00 00 2E 3B 89 
13:12:48.953 > Interrupt received
13:12:49.000 > RX Channel: 40 --> 95 84 85 52 36 84 85 52 36 01 00 01 01 1E 01 D1 05 30 00 08 00 01 00 00 00 01 67 | -80 dBm
13:12:49.045 > Interrupt received
13:12:49.091 > RX Channel: 40 --> 95 84 85 52 36 84 85 52 36 02 E3 7B 00 00 1A BA 01 DE 00 00 09 36 13 8A 04 F0 22 | -80 dBm
13:12:49.137 > Interrupt received
13:12:49.182 > RX Channel: 75 --> 95 84 85 52 36 84 85 52 36 83 00 01 00 36 03 E8 02 18 00 05 8D 40 18 | -80 dBm
13:12:49.375 > RX Period End
13:12:49.525 > Success
13:12:53.878 > Fetch inverter: 114173305377
13:12:54.088 > TX RealTimeRunData Channel: 40 --> 15 73 30 53 77 80 18 23 20 80 0B 00 66 93 B2 C0 00 00 00 00 00 00 00 00 DC 2D 14 
13:12:54.136 > Interrupt received
13:12:54.182 > RX Channel: 75 --> 95 73 30 53 77 73 30 53 77 01 00 01 01 75 00 03 00 0C 01 72 00 01 00 02 00 0A 94 | -80 dBm
13:12:54.228 > Interrupt received
13:12:54.274 > RX Channel: 3 --> 95 73 30 53 77 73 30 53 77 02 C1 A1 00 0A 71 B9 02 0E 01 EF 09 30 13 89 00 00 74 | -80 dBm
13:12:54.320 > Interrupt received
13:12:54.366 > RX Channel: 61 --> 95 73 30 53 77 73 30 53 77 83 00 00 00 00 00 00 01 B8 00 03 1F 32 81 | -80 dBm
13:12:54.415 > RX Period End
13:12:54.460 > Success
13:12:54.506 > [DPL::announceStatus] waiting for sufficiently recent power meter reading
13:12:56.854 > [PowerMeterHttpJson] New total: 133.63
13:12:56.907 > [DPL::updateInverter] sending limit of 17.2 % (138 W respectively), max output is 800 W
13:12:56.962 > TX ActivePowerControl Channel: 61 --> 51 73 30 53 77 80 18 23 20 81 0B 00 00 AC 00 01 81 00 0B 
13:12:57.008 > [DPL::announceStatus] waiting for a power limit command to complete
13:12:58.929 > RX Period End
13:12:59.049 > All missing
13:12:59.151 > Nothing received, resend whole request
13:12:59.197 > TX ActivePowerControl Channel: 75 --> 51 73 30 53 77 80 18 23 20 81 0B 00 00 AC 00 01 81 00 0B 
13:13:00.965 > RX Period End
13:13:01.011 > All missing
13:13:01.057 > Nothing received, resend whole request
13:13:01.102 > TX ActivePowerControl Channel: 3 --> 51 73 30 53 77 80 18 23 20 81 0B 00 00 AC 00 01 81 00 0B 
13:13:01.148 > Interrupt received
13:13:01.198 > RX Channel: 40 --> D1 73 30 53 77 73 30 53 77 81 00 00 0B 00 14 07 48 | -80 dBm
13:13:02.968 > RX Period End
13:13:03.043 > Success
13:13:03.088 > Fetch inverter: 114184855236
13:13:03.134 > [DPL::updateInverter] actual limit is 17.0 % (136 W respectively), effective 6111 ms after update started, requested were 17.2 %
13:13:03.180 > [DPL::updateInverter] Starting inverter...
13:13:03.228 > TX RealTimeRunData Channel: 23 --> 15 84 85 52 36 80 18 23 20 80 0B 00 66 93 B2 C9 00 00 00 00 00 00 00 00 8C 47 25 
13:13:03.274 > [DPL::announceStatus] waiting for a start/stop/restart command to complete
13:13:03.320 > Interrupt received
13:13:03.367 > RX Channel: 61 --> 95 84 85 52 36 84 85 52 36 01 00 01 01 1E 01 CF 05 29 00 08 00 01 00 00 00 01 60 | -80 dBm
[... 1084 lines cut ...]
13:15:02.720 > Interrupt received
13:15:02.767 > RX Channel: 40 --> 95 73 30 53 77 73 30 53 77 01 00 01 01 6C 00 CE 02 EE 01 69 00 D0 02 EE 00 0A 84 | -80 dBm
13:15:02.812 > RX Period End
13:15:02.858 > Success
13:15:02.905 > [DPL::announceStatus] waiting for sufficiently recent power meter reading
13:15:06.640 > Fetch inverter: 114184855236
13:15:06.692 > TX RealTimeRunData Channel: 23 --> 15 84 85 52 36 80 18 23 20 80 0B 00 66 93 B3 45 00 00 00 00 00 00 00 00 8F 22 CE 
13:15:06.738 > Interrupt received
13:15:06.783 > [PowerMeterHttpJson] New total: -5.09
13:15:06.829 > RX Channel: 3 --> 95 84 85 52 36 84 85 52 36 01 00 01 01 1E 01 DE 05 53 00 08 00 01 00 00 00 01 0B | -80 dBm
[... cut ...]
13:15:07.260 > TX ActivePowerControl Channel: 40 --> 51 73 30 53 77 80 18 23 20 81 0B 00 00 B1 00 01 87 90 80 
13:15:07.304 > Interrupt received
13:15:07.351 > RX Channel: 75 --> D1 73 30 53 77 73 30 53 77 81 00 00 0B 00 14 07 48 | -80 dBm
13:15:09.202 > RX Period End
13:15:09.298 > Success
13:15:09.349 > [DPL::updateInverter] actual limit is 17.0 % (136 W respectively), effective 2480 ms after update started, requested were 17.8 %
13:15:09.396 > [DPL::announceStatus] waiting for sufficiently recent inverter data

spcqike, 2024-07-14:
Die Firmware macht, was sie soll. 

Ohne es beweisen zu können (wobei dein Log genau das beschreibt) vermute ich, dass dein wechselrichter einfach „festhängt“. Du betreibst ihn mit zu kleinen Leistungen (du hast nicht 1% eingestellt, was 8W entsprechen würde und bereits viel zu wenig wäre, sondern 1W!) die Erfahrung hier in der Community zeigt, dass die wechselrichter mit ~10-12W pro Eingang gefahren werden sollten, da sie sonst Probleme machen, sich aufhängen und nicht mehr ordentlich reagieren. 

Weil die Probleme oft vorgekommen sind, wurde ein workaround einprogrammiert, der den wechselrichter nach mehreren erfolglosen startaufforderungen Neustartet. Genau das passiert bei dir 

> 13:14:46.976 > [DPL::loop] issuing inverter restart command after update timed out repeatedly

du sagst selber dass er nach 13:14 Uhr wieder blau wurde. Das ist nicht wegen der Leistung also dem Limit, sondern wegen dem Neustart Befehl. 

Wenn du deine Probleme ernsthaft lösen willst, tu uns (oder mir) den Gefallen und setze deine Mindestleistung auf, sagen wir, 600W. Deine Grundlast auf 650 (sie muss halt höher sein). Geh in die Konsole und gucke, ob und warum der WR deaktiviert wird. 

Wir brauchen den erzwungenen Moment, in dem der Wechselrichter auf ein Limit < Mindestleistung gesetzt werden würde in der Konsole. (600W waren nur, weil das schnell erreicht wird. Wenn du mit normalen Werten solch ein Verhalten erzwingen kannst, bspw mit 200W und nem aktiven Wasserkocher, den du aus schaltest, auch gut) 

dazu: pass deine Einstellungen der „Module“ an. Ausführliches logging im Reiter „DTU“ erstmal aus, dafür im Reiter „Power limiter“ an.

Solar320, 2024-07-14:
Du hast recht, ich habe ausversehen Prozent mit Watt verwechselt. 

Das mit den vermeintlichen  1 %  (1Watt) war nur ein Test um zu sehen ob ich damit das Standby Problem umgehen kann. 
Ich hätte also eigentlich 8Watt einstellen müssen, aber ich lass es jetzt auf 10W bzw. ich werde es eher mit >15 Watt versuchen.

Allerdings komme ich innerhalb der Woche erst so spät von der Arbeit das es sein kann das ich zu wenig Sonne habe um den test zu machen. 

Eventuell liegt es einfach daran das selbst die originalen min. 10Watt zu instabilitäten führen und das vermeintliche Standby Problem vom Wechselrichter selbst ausgelöst wird. Ich habe das selbe Verhalten mit beiden HM800 Wechselrichtern mit unterschiedlichen Firmwareständen beobachten können. Ich habe damals für den Test extra die Solarmodule umgehängt.

Allerdings muss ich dir noch etwas zeigen:

[image]

Das könnte eine Altlast sein die noch nicht aus dem Quellcode entfernt wurde allerdings verhält sich meine Anlage genau so wie es dort steht. Ob das von der DTU oder durch eine zu geringe WR Leistung ausgelöst wird sei jetzt mal dahingestellt.

Es bleibt also nur, mit einer höheren Minimalleistung nochmals zu testen und das Log abzuspeichern, um die DTU als Ursache weitestgehend auszuschließen.

So bald es möglich ist mache ich den Test, es kann aber sein das ich erst am Freitag dazu komme. (da.12Uhr Feierabend)

spcqike, 2024-07-14:
> Eventuell liegt es einfach daran das selbst die originalen min. 10Watt zu instabilitäten führen 

ja, der Erfahrung nach führen 10W zu Problemen. 10-12W pro MPPT, bei dir also 20-24W, sollten für einen stabilen Betrieb genutzt werden. Das mag ein Richtwert sein und durch Bauteil fertigungstoleranzen hier und da anders aussehen. Aber 10W insgesamt sind wohl zu wenig. 

> und das vermeintliche Standby Problem vom Wechselrichter selbst ausgelöst wird.

So ist es. Wird der wechselrichter mit einer zu geringen Leistung betrieben, bekommt er Probleme und reagiert nicht mehr. Bis auf den Neustart. 

> Das könnte eine Altlast sein

Ich bin mir nicht ganz sicher. Aber ja kann sein. Früher wurde der wechselrichter beim unterschreiten auch immer deaktiviert. Heute passiert das nur noch, wenn der wechselrichter von einer Batterie gespeist wird, der Haken drüber also nicht gesetzt ist. 

> Es bleibt also nur, mit einer höheren Minimalleistung nochmals zu testen und das Log abzuspeichern, um die DTU als Ursache weitestgehend auszuschließen.

Das wäre super. Wie geschildert, bräuchte man einen log Auszug von dem Moment, wo die DTU den wechselrichter in den standby schickt. Was hier eigentlich nicht passieren sollte. 

Ich hab es bei mir wie gesagt ebenfalls getestet, das min Limit auf 800 gestellt und, da das weit über meinem Bedarf lag, wurde der WR auf 800W limitiert. Also so, wie es sein sollte. 

> So bald es möglich ist mache ich den Test, es kann aber sein das ich erst am Freitag dazu komme. (da.12Uhr Feierabend)

probiere doch einfach ein Limit von … sagen wir 30W und beobachte in der Aufzeichnung, ob er läuft oder doch stoppt. Fürs erste. Home Assistant oder dergleichen vorausgesetzt. 
sollte er mit 30W begrenzt laufen, brauchst du auch Freitag keinen log Auszug ziehen 😄

Solar320, 2024-07-16:
Morgen wird es zu knapp, da bin ich nach der abreit noch unterwegs,vieleicht ab Donnerstag. :)

Solar320, 2024-07-19:
Der Versuch hat nichts gebracht, der Wechselrichter schaltet sich nicht ab, bei zb 600W pmin. 
Im Protokoll stand irgendwas das er auf das untere limit gehen muss.
 Er hat aber nicht abgeschaltet weil auch nicht genug Sonne da war um das Wattmeter dauerhaft ins Minus zu drücken.

Das Problem liegt wo anders, ich habe jetzt eben wider 15W als minimum eingetragen und dann den Geschirrspüler abgeschaltet, dann hatte ich ca. -36Watt auf der Uhr und was macht der Limiter? Er senkt die Wechselrichterleistung auf 8W 1% !!!! obwohl 15Watt 1,875% eingestellt sind. Kann es sein das er nur ganze Zahlen einstellen kann also 2% 16 Watt als minimum?
Danach ist der Wechselrichter wider abgestürzt und verschwunden.

Da liegt das Problem und vermutlich tritt das nur dann auf wenn der gemessene Wattmeter Wert negativ ist.

Ich werde mich jetzt mal langsam hocharbeiten und mal 16Watt und danach 20 einstellen. Vieleicht hab ich mal Glück mit der Sonne das genug Leistung da ist. 

Das Protokoll kann ich jetzt nicht liefern, dazu muss ich meinen laptop verwenden weil aufgrund der vorhandenen Grundlast und der stark schwankenden Sonne die Wohnung ständig zu viel braucht. 
Ich muss das Protokoll  mit dem laptop beobachten während ich den Computer abschalte, sobald die Last vom Computer wegfällt komme ich mit der Leistung ins minus und dann tritt der Fehler auf und ich kann ihm am Laptop beobachten. Leider hat meine Frau den besch... Geschirrspüler eingeschaltet und deswegen kann ich jetzt nicht messen ohne ständig den GS zu unterbrechen. 

Am Donnerstag ist es mir leider nicht gelungen durch P min. 600W, mehr zu erzeugen als die wohnung braucht. Dadurch war die Leistungsanzeige die meiste Zeit positiv und nur kurz negativ. 

Normalerweise dürfte der wechselrichter 1,9% abregelung nicht unterschreiten, er tut es aber trotzdem.

[image]

Solar320, 2024-07-19:
So, mein letzter Stand ist derzeit das ich ein 30W limit eingestellt habe, damit regelt er auf 3% und "24Watt" herunter.
16Watt und 2 % sind schon zu wenig!

Bei 600W limit hat der wechserichter gestern immer um die 100-140Watt produziert, daher konnte er nicht abstürzen.

Scheinbar schaffen die HM800 nicht oder nicht viel weniger als 24Watt als minimum......

spcqike, 2024-07-19:
> Der Versuch hat nichts gebracht, der Wechselrichter schaltet sich nicht ab, bei zb 600W pmin.

ja, genau das sollte ja auch passieren. Wenn der WR von Solarpaneelen gespeist wird, sollte er nie abschalten, sondern auf dem untersten Limit stehen bleiben. 

> Scheinbar schaffen die HM800 nicht oder nicht viel weniger als 24Watt als minimum......

wie ich bereits schrieb
> ja, der Erfahrung nach führen 10W zu Problemen. 10-12W pro MPPT, bei dir also 20-24W, sollten für einen stabilen Betrieb genutzt werden.

> Kann es sein das er nur ganze Zahlen einstellen kann also 2% 16 Watt als minimum?

Da bin ich mir grad nicht ganz sicher, aber ja, irgendwie sowas.... wir berechnen das absolute Limit, übertragen es glaube ich als Gleitkommazahl in % (also bspw. 2.3%), aber der WR scheint intern abzurunden. zumindest sieht es in meinem Log so aus

``14:20:45.592 > [DPL::updateInverter] actual limit is 20.0 % (400 W respectively), effective 2392 ms after update started, requested were 20.4 %``

was genau war jetzt dein Grundproblem? Mit einem passenden unteren Limit (bspw. 24W für 3% und für 10-12W pro MPPT), sollte der WR bei dieser Leistung stehen bleiben und nicht abschalten. Genau das willst du doch, oder?

schlimmchen, 2024-07-19:
> wir berechnen das absolute Limit, übertragen es glaube ich als Gleitkommazahl in % (also bspw. 2.3%), aber der WR scheint intern abzurunden

Genau so ist es.

spcqike, 2024-07-19:
@schlimmchen macht es Sinn, das untere Leistungslimit dahingehend zu begrenzen? die Weboberfläche sieht Werte ab 10W vor.
[image]

kann/sollte man das ggf. dynamisch machen und auf nen ganzen Prozentpunkt vorgeben?

also 12W = 4% beim HM-300, 24W = 4% beim HM-600, 24W = 3% beim HM-800, 30W = 2% beim HM-1500 (bei dem sind es doch ebenfalls 12W pro MPPT, also 24W, was nur 1.6% entsprächen, vom WR abgerundet 1%, also 15W, was wieder zu wenig wär, also aufgerundet auf 2% eben 30W) bis hin zu 60W beim HMS-2000 (auch hier: 12W pro MPPT sind 48W = 2.4%, da wir aufrunden auf 3% haben wir 60W)

eigentlich sollte es reichen, wenn es im Wiki beschrieben ist, aber scheinbar ja nicht :) (zumal man es per config direkt so oder so überschreiben kann)

Solar320, 2024-07-20:
Vieleicht wäre ein deutlicherer Warnhinweis über dem Eingabefeld ausreichend. Plus ein entsprechender Link auf eine Liste von Wechselrichtern auf der dann die durch Erfahrungswerte ermittelte Untergrenze verzeichnet ist. 

Sowas wie " !!Ein zu geringer Wert kann bei bestimmten Modellen zu unvorhergesehenen Ausfällen führen, eine Liste mit
                      bereits ermittelten Minimumwerten finden sie hier "link" !!!

Ich bin beim derzeitigen Eingabefeld auch ursprünglich davon ausgegangen das die im Eingabefeld nach unten hin begrenzten 10W schon ausreichend für den HM800 sind, was sich dann als Fehler herausgestellt hat.

Ich gehe auch davon aus, das irgendwann noch mehr Wechselrichter unterstützt werden könnten, dann müsste man ja für jedes Modell jedemenge zusätzlich programmieren. (Ob man sich das antun will?)

Mit den jetzt eingestellten 30W also quasi 24W scheint der HM800 ohne unvorhergesehene Neustarts zu laufen. 

Wenn ich von der Benutzeroberfläche auf das Problem hingewiesen werde und irgendwo eine Liste finde was ich hier minimal einstellen darf, dürfte das für viele Nutzer schon ausreichend sein vermute ich mal.
Die Liste könnte alternativ auch irgendwie nach jedem Update mit im ESP integriert werden, so das man nicht zwingend auf eine ständige Internetanbindung angewiesen ist, muss aber nicht sein.

Solar320, 2024-07-20:
> was genau war jetzt dein Grundproblem? Mit einem passenden unteren Limit (bspw. 24W für 3% und für 10-12W pro MPPT), sollte der WR bei dieser Leistung stehen bleiben und nicht abschalten. Genau das willst du doch, oder?

Eigentlich genau das, es muss durch umprogrammieren oder einen deutlicheren Hinweis und eventuell einer Liste mit Minimumwerten verhindert werden das andere Nutzer ebenfalls in die Falle tappen.   Welche Variante jetzt weniger Aufwand produziert müsst ihr selber wissen, ich vermute das es mit der Liste einfacher ist. 

Im derzeitigen Hinweis steht ja nur das ein stabiler Wert eingetragen werden muss aber es ist weder ein Hinweis zu finden was eigentlich passiert wenn der Wechselrichter instabil wird noch ein Link auf den Wikieintrag.

Manos1966, 2024-07-20:
> es ist weder ein Hinweis zu finden was eigentlich passiert wenn der Wechselrichter instabil wird noch ein Link auf den Wikieintrag.

Meinst du das?

[image]

spcqike, 2024-07-20:
@Manos1966 
Man sollte es wohl doch konkretisieren. Wie wir hier gesehen haben, reichen 10W pro MPPT nicht. Vor allem dann, wenn der WR das übertragene Limit (in Prozent) abrundet. 

20W beim HM-800 sind 2.5%. Der WR rundet ab, stellt 2% ein. Also 16W, was zu wenig ist. 

Vielleicht eher „10W pro MPPT, aufgerundet auf volle Prozent.“ mit 2 oder 3 Beispielen, wie ich sie weiter oben beschrieben habe?

Manos1966, 2024-07-20:
@spcqike OK, ich stelle etwas zusammen 👍

Manos1966, 2024-07-20:
@spcqike werfe bitte ein Auge: https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Dynamic-Power-Limiter#minimum-power-limit---minimales-leistungslimit

spcqike, 2024-07-21:
Ich kann es nachvollziehen :) aber ich stecke da auch tiefer drin als viele andere, vor allem als neue Benutzer. 

@Solar320 wie findest du die Beschreibung im Wiki? Hilft sie weiter?

Sorbat, 2024-07-21:
Was ist der Punkt dieser Beschreibung?
Man soll mindestens 2% einstellen? Warum wird dann erstmal mit 1,6% gerechnet? 🤔 ich versteh es nicht

Manos1966, 2024-07-21:
> Man soll mindestens 2% einstellen? Warum wird dann erstmal mit 1,6% gerechnet? 🤔 ich versteh es nicht

Ich habe den Eindruck @Sorbat du weiss nicht was "z.B." bedeutet....

Wie wuerdest du denn es selber beschreiben?

Sorbat, 2024-07-21:
Ja es kann natürlich sein, dass ich dumm bin. Aber das ist ja dann nur für euch schlimm, nicht für mich 😄

Manos1966, 2024-07-21:
> Ja es kann natürlich sein, dass ich dumm bin. Aber das ist ja dann nur für euch schlimm, nicht für mich 😄

Du hast Recht.

@spcqike Ich glaube wir geben uns viel Muehe, komplexe Themen begreiflich fuer Normale-Anwender zu machen,  die sowieso kein Interesse haben die interne Komplexitaet der OpenDTU-OnBattery zu verstehen (ich meine NICHT dich, @Sorbat ).

Machen wir also "xxxx-sicher":

Ich habe eine Tabelle der MPPT Anzahl / Wechselrichter Leistung zusammengestellt und das Minimum berechnet.
Die Rechte Spalte (Round-up W) zeigt die Minimum Power Limit laut deiner Berechnung @spcqike 
[image]

Man erkennt, mit Ausnahme der HMS-1600/1800/2000 (die 48W, 54W , bzw. 60W brauchen),
**alle Wechselrichter kommen mit Minimum Power Limit von 45W klar!**

Dann werde ich das Wiki demensprechend anpassen:
> Minimum Power Limit sollte 45W (fuer die HMS-1600/1800 55W, HMS-2000  60W)
Das Einstellen niedrigere Werte machen Sie auf eigene Gefahr.

und damit haben wir ein komplexes Thema begreiflich fuer alle gemacht 😉 

Denn die ganze Anleitung wie Sorbat bemerkt hat, ist zu "heavy" fuer Durchschnitts-Anwender:

***
### Minimum Power Limit - Minimales Leistungslimit 
**Minimum Power Limit:** Only if it has to produce at least this value (10W in this example) the inverter will be enabled. 
Otherwise, the inverter will remain disabled (yellow color https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Farbe-des-Wechselrichterfensters ). <br/>
`CAUTION!: Too low values (eg 10W in this example) risk of "blocking" your inverter, thus requiring a restart.`<br/>
As a rule of thumb, calculate 12W for every MPPT your inverter has e.g. for HM-1500 (2 MPPT) the minimum should be 2x12 = 24W<br/>
`Important Update: Minimum Power Limit is transfered as percentage % to the inverter. For the HM-1500, 24W would mean 1,6% percent.`
`UNFORTUNATELY the Hoymiles accepts only the integer part of the relayed command: 1,6% percent becomes 1% = 15W! This is much less than the 24W we mentionned - there is risk your inverter may freeze and require restart!!!`<br/>
**SOLUTION:** Calculate the minimum value (number of MPPTS multiplied by 12W) your inverter must have, e.g. for the HM-1500, 2 MPPT x 12W = 24W.<br/>
Calculate what this value represent in percent for this inverter, e.g. for the HM-1500 24W divided by 1500W = 1,6% percent.<br/>
Round this percent to the next highest integer value, e.g. 1,6% should be 2% percent.<br/>
Calculate the Watt value based on the new percentage, e.g. 2% percent of the HM-1500 is 2% x 1500 = 30W.<br/>
Subsequently, enter 30W as Minimum Power Limit (instead of the 24W calculated initially).

**Minimales Leistungslimit:** Nur wenn er mindestens diesen Wert (in diesem Beispiel 10 W) erzeugen muss, wird der Wechselrichter aktiviert. Andernfalls bleibt der Wechselrichter deaktiviert (gelbe Farbe https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Farbe-des-Wechselrichterfensters )<br/>
`ACHTUNG! Bei zu niedrigen Werten (z.B. 10W in diesem Beispiel) besteht die Gefahr, dass Ihr Wechselrichter „blockiert“ und somit ein Neustart erforderlich wird.`<br/>
Als Faustregel gilt: Berechnen Sie 12W für jeden MPPT, den Ihr Wechselrichter hat, z.B. für HM-1500 (2 MPPT) sollte das Minimum 2x12 = 24W sein.<br/>
`Wichtiges Update: Die Mindestleistungsgrenze wird als Prozentsatz an den Wechselrichter weitergegeben. Für den HM-1500 würden 24W also 1,6% Prozent bedeuten.`
`LEIDER akzeptiert der Hoymiles nur den ganzzahligen Teil des übermittelten Befehls: Aus 1,6% Prozent wird 1% = 15W! Das ist viel weniger als die erwähnten 24W - es besteht die Gefahr, dass der Wechselrichter einfriert und neu gestartet werden muss!!!`<br/>
**LÖSUNG:** Berechnen Sie den Mindestwert (Anzahl der MPPTS multipliziert mit 12W), den Ihr Wechselrichter haben muss, z.B. für den HM-1500, 2 MPPT x 12W = 24W.<br/>
Berechnen Sie, was dieser Wert in Prozent für diesen Wechselrichter bedeutet, z.B. für den HM-1500 24W geteilt durch 1500W = 1,6% Prozent.<br/>
Runden Sie diesen Prozentsatz auf den nächsthöheren ganzzahligen Wert, z.B. 1,6 % sind 2 %.<br/>
Berechnen Sie den Wattwert auf der Grundlage des neuen Prozentsatzes, z.B. 2% Prozent des HM-1500 ist 2% x 1500 = 30W.<br/>
Geben Sie anschließend 30 W als Mindestleistungsgrenze ein (anstelle der ursprünglich berechneten 24 W).

***

Sorbat, 2024-07-21:
@Manos1966 

habe ich die Anleitung so richtig verstanden?

```
Die Leistung sollte pro MPPT immer Mindestens 12w betragen. Da vom Wechselrichter Dezimalstellen nicht berücksichtigt werden, muss hier auf die nächste ganze Zahl aufgerundet werden.

Beispiele:

Hm-800 2x12w = 24w = 3% -> 3 % (24w) 
Hm-1000 2x12w = 24w = 2,4% -> 3% (30w)
Hm-1200 4x12w = 48w = 4% -> 4% (48w) 
Hm-1500 2x12w = 24w = 1,6% -> 2% (30w)
Hms-1800 4x12w = 48w = 2,7% -> 3% (54w)
```

Manos1966, 2024-07-21:
OK @Sorbat ich muss (mindestens!) zugeben, deine Beschreibung ist viel besser als meine 😄😄😄

Du hast einen guten Punkt gemacht. Komplexe Beschreibungen im Wiki helfen Einsteiger nicht. "Profis" werden sowieso Issue/Diskussions-Themen eröffnen.

Solar320, 2024-07-21:
> > es ist weder ein Hinweis zu finden was eigentlich passiert wenn der Wechselrichter instabil wird noch ein Link auf den Wikieintrag.
> 
> Meinst du das?
> 
> ![Capture](https://private-user-images.githubusercontent.com/129660360/350696839-81e75c7e-9500-4166-9047-4aae5e9995bf.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE1NTc1NDksIm5iZiI6MTcyMTU1NzI0OSwicGF0aCI6Ii8xMjk2NjAzNjAvMzUwNjk2ODM5LTgxZTc1YzdlLTk1MDAtNDE2Ni05MDQ3LTRhYWU1ZTk5OTViZi5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcyMVQxMDIwNDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02ZmJmOTI0MjhjY2M1ODBhZDQyNzU0MDIzOGUxNmU2YjA1YzQwNDg2ZjZiMzdlNmVjYzBlMjE1NTRiZjIzODAzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.RfTy0s42Y-t9saNnxTRESQrBDKdX-SUXBqtyvBsjdxI)

Im Grunde ja, ein Link in die Richtung wäre wahrscheinlich ausreichend, Wobei wie gesagt bei mir mindestens 24W anliegen müssen. Ich könnte mal 20W versuchen aber ich vermute das das wider zu wenig ist bzw. durch die Abrunderei im Wechselrichter er in einen Bereich kommt der instabil wird. Das wären bei mir 2,5% und der würde das auf 2% abrunden (16 Watt) was dann instabil wäre. Der Hinweis "...zwischen 30-50W" wäre daher eher passend.

------------------------------------------------
Die oben genannte Liste wäre schonmal passend, wenn wirklich mindestens 12Watt an einem MPPT anliegen müssen für einen stabilen Betrieb dann wird es nicht anders gehen. Ich kann hier nur für den HM800 sprechen mit dem es so funktioniert.

Hm-800 2x12w = 24w = 3% -> 3 % (24w)
Hm-1000 2x12w = 24w = 2,4% -> 3% (30w)
Hm-1200 2x12w = 24w = 2% -> 2% (24w)
Hm-1500 2x12w = 24w = 1,6% -> 2% (30w)
Hms-1800 4x12w = 48w = 2,7% -> 3% (54w)

PhilJaro, 2024-07-21:
Ich hätte an der Stelle mal eine Zwischenfrage, warum wird hier immer aufgerundet auf die ganze Zahl? 

Ich hatte bisher immer den Eindruck, die WR können auch genauer Regeln als in 1 Prozent schritten, allerdings liefern die immer nur ganze Zahlen als aktuelles Limit zurück, oder bin ich da komplett falsch mit der Annahme?

spcqike, 2024-07-21:
Das mag sein. Der DPL übergibt den Wert aber als prozentualen Wert als gleichkommazahl, rechnet also aus 30W beim HM-800 3.75% und überträgt die. Der WR scheint dann aber nur 3% zu verarbeiten. 

Im manuellen Betrieb, ohne DPL, kann man zwischen Prozent und absolutem Limit wählen. Da nimmt er auch krumme Werte als absolut. 

Der DPL hingegen sendet es als %. 

So zumindest was ich sehe.

schlimmchen, 2024-07-21:
> Der DPL hingegen sendet es als %.

Korrekt.

Es besteht (der berechtige) Wunsch, dass man mit diesem Wissen, dass der WR abrundet, vorher nochmal 0.5% Punkte draufrechnet, damit man näher am gewünschten Limit ist, allerdings dafür ggf. auch zu viel produziert.

> macht es Sinn, das untere Leistungslimit dahingehend zu begrenzen?

@spcqike Ne, finde ich nicht. Aber es gibt ein Issue, mit einem verwandten Thema, das noch offen ist, weil wir irgendwo eine fetter Warnung platzieren wollen, wenn das untere Leistungslimit zu gering ist. Mein Vorschlag: In den DPL Settings werden Metadaten über die Wechselrichter verarbeitet. Da könnte man auch die Anzahl Kanäle nehmen (schon bekannt im Web UI) und daraus ein sinnvolles unteres Leistungslimit berechnen (eigentlich schätzen, weil Kanäle != MPPTs). Wenn das unterschritten wird, erscheint eine gelbe/orangene Warn-Box mit einem Text "Wir haben festgestellt, dass die Wechselrichter bei kleinen Limits zum Schwingen neigen, weshalb die minimale Leistung dieses WR 123W nicht unterschreiten sollte". Wie wäre das?

spcqike, 2024-07-21:
> vorher nochmal 0.5% Punkte draufrechnet

Ich glaube er rundet auch 1.9% ab auf 1%. Bin ich mir grad aber nicht 100%ig sicher und würd das die Tage nochmal validieren. 

Wenn er das macht, macht er das ja immer. 
sollte er beim absoluten Limit weniger stark abrunden, spricht etwas dagegen das absolute Limit zu übertragen? 

Wenn die Sonne die Tage scheint werde ich es mal testen, ohne DPL per webUI. Kann man ja beide Limits übertragen. 

> Wie wäre das?

klingt gut.

schlimmchen, 2024-07-21:
> Ich glaube er rundet auch 1.9% ab auf 1%.

Ja, eben, deshalb macht es Sinn, auf das relative Limit nochmal 0.5% Punkte draufzurechnen. Wenn man dann 1.9% haben will, schickt man 2.4% und erhält 2%, was näher an 1.9% ist als 1%. Das gleiche Spiel gilt für 1.6%.

> sollte er beim absoluten Limit weniger stark abrunden, spricht etwas dagegen das absolute Limit zu übertragen?

Soll ich dir ein Geheimnis verraten? Der WR kennt nur relative Limits. Ein absolutes Limit wird vorher per bekannter Maximalleistung in ein relatives umgewandelt.

spcqike, 2024-07-21:
> Der WR kennt nur relative Limits.

bisher dachte ich, dass er auch absolute Limits kennt. Dass der DPL nur relative schickt, weiß ich. Aber im UI kann man auch absolute schicken. Und, wenn ich es richtig in Erinnerung habe, werden die anders übernommen als relative. Aber das kann auch an der Rückmeldung / Auswertung / Darstellung liegen. Wie gesagt, ich werd es mal testen (: 

aber auch wegen der lib dachte ich, dass es da Unterschiede gibt. 

https://github.com/helgeerbe/OpenDTU-OnBattery/blob/415c767d1d376ca1feb2b35e10cecf2ed256efbb/lib/Hoymiles/src/commands/ActivePowerControlCommand.cpp#L14

> Wenn man dann 1.9% haben will, schickt man 2.4% und erhält 2%, was näher an 1.9% ist als 1%. Das gleiche Spiel gilt für 1.6%.

naja. 2 MPPT am HM-700 brauchen wahrscheinlich auch ihre 24W. Das sind 3.4%. plus 0.5% immerhin 3.9%, was wohl aber ebenfalls auf 3% abgerundet würde. 

Wenn man hart auf addieren will, dann 0.99%. Oder man rundet an der Stelle einfach fest auf. Oder?

Manos1966, 2024-07-21:
Hi @schlimmchen 
das ganze ist in C++ programmiert, richtig? Dann die Funktion ceil(MinimumLimit) wird das Problem loesen.

spcqike, 2024-07-22:
@schlimmchen 

ich hab jetzt einen kurzen Test gemacht. In Ermangelung der Sonne leider nur bei 120-140W. 

Mein Fazit:
Der hoymiles kann durchaus ein absolutes Limit empfangen und es einstellen. 

Kobkret habe ich es mir 120, 130 und 140W probiert was bei meiben HMS-2000 eben 6%, 6.5% und 7% sind. 

6% und 7% werden klar genommen. 6.5% nicht. Da kommen nur 6% an. Egal ob „6.5“ oder „6,5“ eingetragen wird. 

Beim absoluten Wert hingegen werden alle 3 Leistungen übertragen und eingestellt. 

130W werden im UI als 7% angezeigt. Ich denke das ist ein anzeige oder rundungsfehler. 

[image]

würde der hoymiles intern immer mit prozentualen Limits rechnen und abrunden, könnte der HMS-2000 keine 130W regeln. Oder?

Manos1966, 2024-07-22:
HMS-2000 (schönes Stück 😍 )
Wir sollten nicht vergessen, er braucht (laut der "12W-Formel") 60W um überhaupt zu funktionieren...

"Wissenschaftliches-Interesse" bei Seite gestellt, ich halte eine 4-5W, oder 0,5%-0,9% Abweichung bei einer Leistung von 500W und mehr (dafür haben wir den Wechselrichter, oder? 🤔 ) als nicht "Kriegsentscheidend".

Auf der anderen Seite, die Erkenntnisse bezueglich des "Minimales Leistungslimit" waren extrem wichtig!
Ich hatte einige "Probleme" mit meinem Test-System (teste zwei HM parallel laufen) und jetzt weiss ich woher es kommt 👍

spcqike, 2024-07-22:
Dass eine 1W Regelung beim HMS-2000 utopisch wäre, ist mir klar :) selbst 1%, also 20W, ist für mich durchaus ok. 

Mir geht es grad eher darum: kann er es prinzipiell. 

Beim HM-700 wäre es schon interessant zu wissen, ob er mit 4% (28W) gefahren werden muss, oder ob er nicht doch 24W oder gar 20W als absolutes Limit nehmen könnte. 

Auch hinten raus:
Vielleicht kann man den DPL ja nochmal angehen und gucken, ob man nicht auch dort das absolute Limit übertragen kann. Bisher übertragen wir es nur relativ was scheinbar immer in 1% Schritten seitens des wechselrichters endet. Zu wissen, dass vielleicht 0.5% in Form eines absoluten Limits gehen …. Dann wäre ja die Abweichung vom gewünschten Soll zum eingeregelten Ist kleiner. Und die hysterese würde besser laufen.

Manos1966, 2024-07-22:
>  Und die hysterese würde besser laufen.

Das war auch mein Gedanke 👍 

Sollten wir vielleicht in das Wiki auch eine Empfehlung zu Hysterese geben?

schlimmchen, 2024-07-22:
> was scheinbar immer in 1% Schritten seitens des wechselrichters endet.

Das ist ein Rundungsfehler in der Hoymiles lib. Siehe #1121. Dieser Fehler hat mich auch dazu verleitet anzunehmen, dass der WR ganzzahlig abrundet beim Verarbeiten des Limits. Du hast aber recht: Der kann auf wenige Watt genau ein Limit annehmen. Leider erreicht er es so schlecht^^

KlimekDE, 2024-08-11:
> Hallo @Solar320
> 
> eine Option "die Standbyschaltung ganz abgeschaltet werden kann" wuerde mir auch gefallen.
> 
> Aber vielleicht hilft dir dieser Loesung weiter: https://github.com/genmad/shellyAPIGen2AndAboveScripts/blob/master/src/openDTUonBattery/virtualPowerMeters_oDoB.js

Würde diese Optin auch sehr begrüßen. Selbiges Verhalten hier bei uns, dass wir noch ein unkontrollierbares BKW haben und es daher zum massiven On-Off geschalte des Inverters kommt. Ein manuelles deaktivieren des automatischen Abschalten des Inverters wäre wünschenswert, gerne auch via MQTT und WebUI.

Solar320, 2024-08-12:
Angeblich bzw. ziemlich sicher ist diese Standbyschaltung ja schon deaktiviert. Zumindest wenn man "Wechselrichter wird von Solarmodulen gespeist " aktiviert hat. Wie das im Batteriebetrieb gehandhabt wird weis ich nicht, da müsste ja nach einer ausreichend große Zeitspanne der Wechselrichter abgeschaltet werden, um die Batterie nicht sinnlos zu entladen. Vieleicht kann man dort die Zeitspanne einstellbar machen und so umsetzen das bei "0" die Schaltung deaktiviert ist und bei zB 15 minuten dann erst diese Zeit vergehen muss bis die Standby schaltung greift. Allerdings ergibt sich der Sinn erst aus Erfahrungswerten aus der Praxis die ich ohne Akku jetzt nicht habe.

Bei meinem HM800 muss man zudem mindestens 30Watt bei "minimales Leistungslimit" einstellen sonst hängt er sich auf. 
Das war bei mir der Fehler weswegen er immer aus gegangen ist. Die DTU hatte damit nicht direkt was zu tun.

schlimmchen, 2024-08-15:
> Wie das im Batteriebetrieb gehandhabt wird weis ich nicht, da müsste ja nach einer ausreichend große Zeitspanne der Wechselrichter abgeschaltet werden, um die Batterie nicht sinnlos zu entladen.

So wird es sein. Siehe #871.

> Sry, keine Logs, keine sinnvolle Hilfe.

So hat es @spcqike geschrieben und seitdem haben wir vom Themenstarter nichts mehr gehört. Ich sehe nicht, was in diesem Issue zu tun ist und schließe es daher. Wenn ich etwas übersehen habe, gerne Bescheid geben.

closed 2024-08-15 by schlimmchen

Solar320, 2024-08-17:
Hat sich auch aus meiner Sicht erledigt, close.