hoylabs/OpenDTU-OnBattery #1083: [Request]verzögertes, zeitlich einstellbares Herunterregeln des Wechselrichters bei Intervall-Last

TriBoone, 2024-07-04:
### Is your feature request related to a problem? Please describe.

Bei Intervall-Lasten, wie z. B. einer Mikrowelle, regelt der Wechselrichter gerade herunter oder hat heruntergeregelt, wenn die Last wieder steigt. Dadurch wird ggf. Energie aus dem öffentlichen Netz gezogen, obwohl es ausreichend vom Wechselrichter geben könnte, wenn dieser noch hochgeregelt wäre.

### Describe the solution you'd like

Eine gute Lösung für mich (Solar-only-Balkonkraftwerk) wäre, dass man eine Zeitdauer einstellen kann, in der der Wechselrichter nicht heruntergeregelt wird, wenn die Last weniger wird. Erst bei weniger Last und verstrichener Zeit x soll der Wechselrichter herunterregeln.

### Additional context

siehe Diskussion #1079

Das hierdurch mehr Energie ins öffentliche Netz eingespeist wird ist für mich klar und für mich nicht negativ.
Positiv ist für mich, dass ich die mögliche Energie des Balkonkraftwerks effektiver nutzen kann und dadurch Energie aus dem öffentlichen Netz spare.

oldman4U, 2024-07-04:
Wenn das Einspeisen kein Problem ist weshalb dann nicht immer den ganzen Ertrag ins Hausnetz einspeisen? Wozu den DPL Aufwand betreiben um es dann temporär zu deaktivieren?

schlimmchen, 2024-07-04:
Bedenke Batterie-Systeme, wo du natürlich trotz aller Energie, die vielleicht "zu viel" hast, dennoch Überschuss in deine Batterie bringen möchtest.

Die Frage, warum Benutzer ohne Batterie Nulleinspeisung machen möchten führt vermutlich sehr schnell (wieder) zu Glaubenskriegen und daher wäre es schön, wenn wir das einfach hinnehmen, dass manche das so haben möchten. Da es für Batteriesysteme absolut Sinn macht, hat das Feature meiner Meinung nach eine Berechtigung, und für Benutzer ohne Batterie fällt das dann also Option ab :wink:

TriBoone, 2024-07-04:
> Wenn das Einspeisen kein Problem ist weshalb dann nicht immer den ganzen Ertrag ins Hausnetz einspeisen? Wozu den DPL Aufwand betreiben um es dann temporär zu deaktivieren?

Ich möchte gern das maximum an Energie nutzen, dabei aber die "erlaubten" 800 W ins Netz nicht länger überschreiten. Unser BKW kann bis zu 1,6 kW liefern.

TriBoone, 2024-07-04:
> Bedenke Batterie-Systeme, wo du natürlich trotz aller Energie, die vielleicht "zu viel" hast, dennoch Überschuss in deine Batterie bringen möchtest.
>
> Die Frage, warum Benutzer ohne Batterie [...]

Mir geht es nicht um eine Nulleinspeisung, sondern um selbst mehr als 800 W nutzen zu können, sofern verfügbar.
Unlimitiert könnte unser BKW bis zu 1,6 kW ins Netz abgeben.

spcqike, 2024-07-04:
@TriBoone ich glaube so einfach ist das nicht, mit dem erlaubten "BKW" ;) 800VA Leistung. egal ob ins Haus oder raus. sonst könnte jeder eine 25kWp Anlage mit Nulleinspeisung als BKW anmelden :D

TriBoone, 2024-07-04:
> @TriBoone ich glaube so einfach ist das nicht, mit dem erlaubten "BKW" ;) 800VA Leistung. egal ob ins Haus oder raus. sonst könnte jeder eine 25kWp Anlage mit Nulleinspeisung als BKW anmelden :D

Ja, so einfach ist das nicht. Das Einspeise-Limit für BKW liegt aktuell bei 800 W. Mehr darf nicht aus dem Wechselrichter bei einem BKW zur Verfügung gestellt werden. Das ist für diesen Feature Request aber erstmal nebensächlich.

oldman4U, 2024-07-04:
Nicht ganz. Weil dieses Feature mit deinem Setup dazu führt, länger die nicht erlaubte Menge an Strom ins Hausnetz einzuspeisen.

TriBoone, 2024-07-05:
> Nicht ganz. Weil dieses Feature mit deinem Setup dazu führt, länger die nicht erlaubte Menge an Strom ins Hausnetz einzuspeisen.

Aber gerade, wenn ich die erlaubten 800 W als Limit für den Wechselrichter einstelle, wäre es doch schön, das Maximum an Energie dieser 800 W nutzen zu können und dann nicht noch bei Last-Intervallen mehr Energie aus dem öffentlichen Stromnetz zu beziehen als nötig.

oldman4U, 2024-07-05:
Dann limitiere den WR auf 800W und du hast das. Nebenbei bist du wieder legal unterwegs.

schlimmchen, 2024-07-05:
@oldman4U Du hast es nicht verstanden. @TriBoone stellt *nie* mehr als 800W Leistung am WR ein. Es gibt keinen Anhaltspunkt ihm (ihr?) zu unterstellen, dass irgendetwas nicht "legal" sei oder nicht sicher.

Können wir es bitte unterlassen @TriBoone bekehren zu wollen ein festes Limit einzustellen und den DPL auszuschalten, und uns stattdessen auf das Feature konzentrieren, dessen Sinnhaftigkeit meiner Meinung nach nicht in Frage steht.

oldman4U, 2024-07-05:
Er schreibt vor 18 Stunden selbst, er möchte mehr als 800W einspeisen. Da gibt es nichts zu unterstellen.

Ein Herd mit zB neun Stufen taktet in ebenso vielen unterschiedlichen Abständen. Welche Zeit definierst du da? Und eines ist klar. Wenn die Platte dann gerade nicht heizt wird mehr, und in diesem Fall eventuell auch über den erlaubten Wert eingespeist.

spcqike, 2024-07-05:
>Welche Zeit definierst du da? 

ich hoffe doch, dass der Wert vom Benutzer einstellbar wird :) immerhin hat (m)ein Waschmaschinenmotor vielleicht ein ganz anderes Taktverhalten als (s)eine Mikrowelle 😄 

ich glaub mir würden 5s schon fast reichen, höchstens 10 :) müsste man testen, was die Aufzeichnung am Ende ergibt.

> Und eines ist klar. Wenn die Platte dann gerade nicht heizt wird mehr, und in diesem Fall eventuell auch über den erlaubten Wert eingespeist.

man kann den DPL auf eine Obergrenze von 800W einstellen und möchte trotzdem  kein Takten / Schwingen. Auch wenn @TriBoone weiter oben was anderes geschrieben hat, vielleicht war es nur ein Missverständnis, ist dieser Request durchaus interessant und wahrscheinlich bei Batterie gestützten Systemen im so wichtiger. von daher hat es in "onBattery" durchaus seine Berechtigung wenigstens diskutiert zu werden.

oldman4U, 2024-07-05:
Abgesehen davon ist es mir persönlich vollkommen egal, ob jemand mehr als die erlaubten 800W einspeist oder nicht. Und wenn er dabei auch noch weiß was er tut (wie die meisten hier wahrscheinlich), kann das ja auch vollkommen ungefährlich sein. Es ist auch jedem freizustellen, ob er einen Überschuss ins öffentliche Netz einspeisen möchte oder nicht - und ich kenne die Diskussionen dazu;-)

_man kann den DPL auf eine Obergrenze von 800W_ ...das wusste ich nicht, finde ich aber eine ganz tolle Funktion und geht bei der Hoymiles DTU nicht.

Ich halte diese Funktion nicht dafür geeignet die Eigenverbauchsquote (und darum geht es TriBoone letztlich) zu steigern, da sie das Problem nur zeitlich verschiebt. Denn auch wenn du EINEN Wert eingeben kannst, funktioniert dieser nur für eine bestimmte Taktung, oder, du stellst einen sehr hohen Wert ein. Und wenn man das macht kommt man wieder auf die hohe, potentiell gefährliche Einspeisung ins Hausnetzt.

Das Problem ist ja eigentlich eines der Trägheit der DPL Funktion im Verhältnis zur Taktung der Verbrauchers. Wenn diese zB nur 2 Sekunden brauchen würde um sich auf einen neuen Wert einzustellen würde das Problem nur schlagend werden bei Taktungen kleiner 4 Sekunden. Darüber hinaus würdest du, wie es bei jeder Nulleinspeisung "passiert" beim Einschalten des Verbrauchers Strom vom Netzt beziehen, und beim Ausschalten des Verbrauchers Strom ins Netzt einspeisen. Sprich du kannst die Eingenverbrauchsquote nur beim Einschalten erhöhen. So richtig wird die Trägheit zum Problem, wenn diese sehr hoch ist und die Taktung des Verbrauchers sehr nahe an diesem Wert liegt. Dann kann es passieren, dass du einspeist wenn das Gerät Strom zieht und umgekehrt. Diesem Problem wird nun versucht zu begegnen, indem man das Hinunterregeln um einen Wert X verzögert. Das kann meiner Meinung nach für eine ganz bestimmte Taktung funktionieren, aber eben nur für diese, und in einem normalen Haushalt gibt es halt ganz unterschiedliche Taktungen. Letztlich kann es trotz dieser Regelung dazu kommen, dass sehr lange Strom ins Netz eingespeist wird und sich der Verbraucher gleich nachdem die Zeit X vorbei ist wieder einschaltet.

Ich selbst habe ein Nulleinspeisung basierend auf einer Hoymiles DTU Pro (welche natürlich das selbe Problem hat) und verfolge dieses Projekt, weil ich es toll finde. Hin und wieder melde ich mich, weil ich den Sinn eines Featurerequest nicht verstehe und dieses Mal, weil ich es für geeignet halte eine potential gefährliche Situation noch weiter zu verschärfen.

Bin schon gespannt, wie sich dieses Projekt weiter entwickelt und Hut ab vor allen beteiligten Personen!

oldman4U, 2024-07-05:
Kritik scheint hier unerwünscht, auch wenn sie noch so konstruktiv ist.

schlimmchen, 2024-07-05:
> Kritik scheint hier unerwünscht

Deinen Kommentar zu minimieren/verstecken ist vermutlich ein Missbrauch meiner Kompetenz, daher hab ich das rückgängig gemacht.

Nur weil es mir zu anstrengend wird mich mit jemanden auseinanderzusetzen, der noch nicht einmal das obere Leistungslimit des DPL kennt geschweige denn OpenDTU-OnBattery für die Nulleinspeisung nutzt, muss man ihn ja nicht gleich für alle zensieren.

oldman4U, 2024-07-05:
Danke

Keine meiner Anmerkungen war böse gemeint oder der Versuch etwas zu unterstellen. Wenn das so angekommen ist tut es mir leid, sorry.

Wenn ich die Funktion nicht für potentiell gefährlich halten würde, hätte ich mich auch nicht gemeldet.

Bin schon gespannt ob die Funktion den gewünschten Erfolg bringen wird. Ich wünsche dazu jedenfalls alles Gute!

spcqike, 2024-07-05:
> Wenn ich die Funktion nicht für potentiell gefährlich halten würde, hätte ich mich auch nicht gemeldet.

Ich verstehe immer noch nicht, was daran gefährlich sein soll. Die maximale Leistung legt der Benutzer ja bereits an anderer Stelle fest. Dieser PR soll doch lediglich das sofortige herunter regeln hinauszögern, so dass es weniger Schwingungen gibt. 

(Bzw. so wie @schlimmchen anderswo vorgeschlagen hat für Batterie Benutzer im Winter auch dazu führen, dass nicht sofort bei der kleinsten spitze hoch geregelt wird)

oldman4U, 2024-07-05:
Wenn ein BKW Benutzer mit mehr als 800W WR Leistung diese auf 800W limitiert, sehe ich auch keine Gefahr.

Wir sind uns aber darin einig, dass die hier vorgeschlagene Lösung zu einer höheren Einspeisung ins Netz führt, auch wenn der WR nicht auf die 800W limitiert ist, und darin sehe ich das Problem. Aber ich scheine alleine damit zu sein und daher gehe ich davon aus, dass ich mich irre.

Die Funktion selber finde ich deshalb nicht so prickelnd, weil sie das Problem nicht löst sondern nur verschiebt. Nur eine möglichst rasche und genaue Reaktion auf Veränderungen im Verbrauch führen zu einer möglichst hohen Eigenverbrauchsquote und damit zu einem minimalen Bezug vom Netz, was wahrscheinlich das Ziel von uns allen hier ist. Und das noch umso mehr, wenn man "on battery" ist.

Und wenn das wirklich das Ziel ist, dann ist auch eine Verzögerung beim Anpassen eines höheren Verbrauchs kontraproduktiv.

my 2 cents

😁

saschaabraham, 2024-07-20:
Ich hatte so ein ähnliches Problem bei der Waschmaschine, diese hat ja einen stark schwankenden Verbrauch, weil mal der Motor läuft (bis 200W Verbrauch), dann wieder nicht(nur 10 Watt). Das DPL kommt nicht mit und ist eigentlich immer daneben ;-)

Also habe ich einen Schaltsteckdose an der Waschmaschine, welche den Verbrauch misst. Wenn dieser über ein Zeit- Intervall über einem bestimmten Wert ist, dann wird der DPL- Mode auf 2 gesetzt, d.h. die Begrenzung wird auf das Maximum gesetzt, in meinem Fall 800W. So wird zwar etwas Überschuss eingespeist, aber der schwankende Bedarf der Waschmaschine wird abgedeckt.

Wenn der Verbrauch wieder unter einen bestimmten Wert fällt, wird der DPL wieder auf 0 gesetzt, d.h. normaler Modus.

Das geht auch mit Schaltsteckdosen, die keinen Verbrauch messen, dann einfach wenn Waschmaschine an, dann DPL auf 2.

Realisiert ist das mit ein paar Regeln im Home Assistant.

Bisher geht das mit Spülmaschine, Waschmaschine und Trockner recht gut.

spcqike, 2024-08-10:
@schlimmchen 

wollen wir (du? :) ) dieses Thema wieder aufgreifen? 
das „Tal der tränen“ rückt näher, die Tage werden kürzer und die Überschüsse für die Batterien kleiner. 

Mein solar only System läuft auf -300W Target, damit stört so kleines schwingen nicht auf der stromrechnung. Aber das ist ja kein erstrebenswerter zustand für eure Batterien :D

schlimmchen, 2024-08-10:
sobald nicht. ich hab andere (opendtuonbattery)themen vor der brust und natürlich "keine" zeit...

saschaabraham, 2024-08-10:
Verwende lieber das NullEinspeiseSkript von hier [https://github.com/reserve85/HoymilesZeroExport](https://github.com/reserve85/HoymilesZeroExport)

Da ist das mit den Schwankungen besser gelöst. Solange mehr Energie erzeugt wie gebraucht wird, erfolgt die Nachregelung in kurzen Abständen, wenn mehr verbraucht wird, wie erzeugt, dann wird der Inverter auf Max gestellt (800) und erst nach 20 Sekunden nach geregelt....

schlimmchen, 2024-08-10:
gute idee, der DPL in OpenDTU-OnBattery is eh schrott. lieber einen rechner oder raspi zusätzlich aufstellen 👍 der kann auch schneller rechnen.

saschaabraham, 2024-08-10:
WTH? I just wanted to point out the different strategy of the script, more suitable for the use without battery. Nothing more. And the way i have described above implements something similar...

schlimmchen, 2024-08-10:
warum schreiben wir jetzt englisch? ist WTH = why the hate? hass ist hier sicher ein zu starkes wort. das war schon mit einem augenzwinkern zu verstehen.

aber "nimm lieber" zu schreiben und damit ein externes skript zu bewerben ist nun einmal nicht produktiv für dieses projekt. daher finde ich ist ein schnippischer kommentar meinerseits als reaktion durchaus nachvollziehbar?

wir werden das implementieren. wenns an mir hängt, dauerts noch, leider. ich würde gerne mehr tun. geht halt nicht...

swingstate, 2024-08-10:
What the heck

saschaabraham, 2024-08-10:
Das war absolut nichts abwertend gemeint, dachte halt nur, wenn jemand Anforderungen hat, die hier nicht erfüllt werden (sollten), dann soll er halt ein anderes Projekt verwenden. So wie es jetzt ist, hat es mir gereicht und die Änderung wegen dem DPL aus https://github.com/helgeerbe/OpenDTU-OnBattery/issues/900 hattest Du ja umgesetzt. Damit geht aich eine "Waschmaschinenregelung".

Englisch, weil ich durcheinander gekommen bin, hatte vorher ein Issue auf Englisch erstellt...