hoylabs/OpenDTU-OnBattery #1087: Überskalierung klappt nicht, verschattung falsch erkannt

spcqike, 2024-07-06:
### What happened?

Die überskalierung klappt nicht wie erwartet, da die Eingänge als verschattet erkannt werden. 

### To Reproduce Bug

Laufen lassen mit Skalierung 

### Expected Behavior

Hoch regeln 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

g33e111f

### Relevant log/trace output

```Shell
19:04:01.102 > Success
19:04:01.414 > Success
19:04:01.513 > [DPL::loop] ******************* ENTER **********************
19:04:01.514 > [DPL::calcPowerLimit] battery use prevented, solar power (DC): 10000 W
19:04:01.514 > [DPL::calcPowerLimit] target consumption: -500 W, base load: 250 W, power meter does include inverter output
19:04:01.514 > [DPL::calcPowerLimit] power meter value: 35 W, power meter valid: yes, inverter output: 212 W, solar power (AC): 9217 W
19:04:01.518 > [DPL::calcPowerLimit] limited to solar power: 747 W
19:04:01.519 > [DPL::setNewPowerLimit] input limit: 747 W, min limit: 200 W, max limit: 2000 W, hysteresis: 25 W
19:04:01.519 > [DPL::scalePowerLimit] all channels are shaded, keeping the current limit of 250 W
19:04:01.520 > [DPL::setNewPowerLimit] inverter max: 2000 W, inverter is producing, requesting: 250 W, reported: 250 W, diff: 0 W
19:04:01.520 > Fetch inverter: 116492305207
19:04:01.625 > [PowerMeterHttpJson] New total: 33.22
19:04:02.126 > Success
19:04:02.541 > [DPL::loop] ******************* ENTER **********************
19:04:02.542 > [DPL::calcPowerLimit] battery use prevented, solar power (DC): 10000 W
19:04:02.542 > [DPL::calcPowerLimit] target consumption: -500 W, base load: 250 W, power meter does include inverter output
19:04:02.542 > [DPL::calcPowerLimit] power meter value: 33 W, power meter valid: yes, inverter output: 212 W, solar power (AC): 9217 W
19:04:02.542 > [DPL::calcPowerLimit] limited to solar power: 745 W
19:04:02.542 > [DPL::setNewPowerLimit] input limit: 745 W, min limit: 200 W, max limit: 2000 W, hysteresis: 25 W
19:04:02.542 > [DPL::scalePowerLimit] all channels are shaded, keeping the current limit of 250 W
19:04:02.546 > [DPL::setNewPowerLimit] inverter max: 2000 W, inverter is producing, requesting: 250 W, reported: 250 W, diff: 0 W
19:04:02.546 > Fetch inverter: 116492305207
```

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

spcqike, 2024-07-06:
Wir haben das letztens im PR des Features angesprochen, hier der issue. 

So recht erklären kann ich mir das grade noch nicht. Die AC Leistung ist kleiner als das gesetzte Limit, daher ist ein „alles verschattet“ nicht ganz abwegig. 

Auf der anderen Seite führt ein manuelles erhöhen das Limits prompt zu einem Anstieg der AC Leistung. Auch sieht man im Graphen, dass der WR scheinbar konstant mit einer Leistung und damit mit einer deosselung föhrt. 

[image]
[image]
[image]
[image]

AndreasBoehm, 2024-07-07:
Danke für deinen Detailierten Report.
Wie du sehen kannst liefern die Eingänge nur zwischen 55,1 W und  56,1 W, bei einem Limit von 250 Watt erwarten wir 62,5 W pro Eingang bzw 95% davon also 59,38 W um einen Eingang als nicht verschatett zu werten. 
56,1 W entsprechen nur 89,7% der erwarteten Leistung.

Es gibt jetzt zwei Optionen wie wir hier weiter optimieren können:
a) Wir reduzieren die erwartete Leistung pro Eingang auf 90% (Das würde dir in deinem Fall trotzdem nicht helfen) oder noch weniger?
b) Wir setzen das neue limit immer, wenn es höher als das aktuelle Limit ist (in deinem Beispiel 745 W). Aktuell behalten wir das aktuelle Limit bei wenn alle Eingänge verschattet sind. Durch die änderung könnten wir das Problem fehlender Leistung zumindest theoretisch etwas eindämmen.

AndreasBoehm, 2024-07-07:
Ich hab einen PR angelegt, dort solltest du in kürze auch builds zum ausprobieren finden: https://github.com/helgeerbe/OpenDTU-OnBattery/pull/1089

Die Änderungen konnte ich leider noch nicht testen.

spcqike, 2024-07-07:
Ich hatte auch bereits überlegt, was man tun kann. Wie gesagt, ich hab keine Ahnung warum der HMS-2000 bei 250W nicht liefert, was er soll, wo er denn kann. Bzw lieferte. Denn wenn ich das Limit so manuell setze, klappt es mit 250W 

ich denke option B wäre ab sinnvollsten. So muss man, im Falle der rauskommende Sonne, nicht erst auf einen regelzyklus des DPL und powermeter warten. 

> Ich hab einen PR angelegt

Danke für deine Mühe. Ich nehme an #1077 ist dort noch nicht inkludiert, oder? Da ich grad kein direkten Zugriff auf das System habe, würde ich da mit einem Update dann lieber warten, da ein downgrade aus 1077 wohl nicht ohne weiteres möglich ist? 😅

AndreasBoehm, 2024-07-07:
Hab meinen PR so eben aktualisiert, basis ist nun https://github.com/helgeerbe/OpenDTU-OnBattery/pull/1077. Gute Idee 👍

spcqike, 2024-07-07:
Super. Perfekt :) Werd es dann mal raussuchen und aktualisieren 

hab grad nochmal Remote auf das System geguckt. Der WR hängt schon wieder bei 13% / 260W fest. Hab jetzt mal stufenweise in 1%-Schritten erhöht. Bei 14% war es das gleiche Bild. Er lieferte zu wenig. Bei 15% schnellte es dann nach oben. 

Die genauen Zahlen bekomme ich nicht mehr zusammen, aber irgendwie so 

% / AC_Limit / DC_Leistung
13 / 250 / 220
14 / 280 / 270 
15 / 300 / 320

Ich werd es mal weiter beobachten und testen. Aber vielleicht schafft der HMS erst ab ~15% sauber zu regeln. Erst da hat er genug DC Leistung verarbeitet um dasAC Limit zu bedienen…

spcqike, 2024-07-07:
So. Update meinerseits 

der PR ist installiert und läuft wie erwartet. Wenn ich manuell 250W als Limit einstelle, erzeugt der WR kurz zu wenig Leistung, der DPL geht nun aber wieder nach oben. 

Ich denke fast, wir haben durch das falsche „verschatten“ erkennen zufällig ein ganz anderes Problem gefunden. Und zwar dass die WR im unteren leistungsbereich einfach schlecht das Limit halten. 

Ich werd die kommenden Tage mal ne Tabelle machen und gucken, welche Leistung der HMS-2000 bei welchem Limit tatsächlich liefert. 
hat noch jemand einen HMS-2000 und kann das eventuell ebenfalls testen? Ob es systematisch ist oder an meinem setup liegt 😄 

@AndreasBoehm Danke für deine super schnelle Anpassung. Ich würd das hier mit dem PR erstmal als erledigt abtun. Zumal das Problem scheinbar auch nicht aus dem Feature raus kam, sondern nur durch es sichtbar wurde (: aber der fix hilft auf jeden Fall, dass sich das System nicht an der Stelle festfährt. (Was heute wieder 3h lang der Fall war 😂)

closed 2024-07-07 by spcqike

AndreasBoehm, 2024-07-07:
Ich kann gerne mit meinem HMS-2000 testen ob er sich genau so verhält, lass mich einfach wissen wenn du eine Tabelle erstellt hast, dann versuche ich die Werte nachzustellen.

Die Frage ist jetzt nur ob wir jetzt **A** oder **B** als 'Lösung' brauchten :D Sollen wir beides in meinem PR mal lassen oder soll **A** erstmal wieder rausnehmen? Potentiell nehmen wir dann ja zu oft an das ein Eingang **nicht** verschattet ist.

spcqike, 2024-07-07:
Gute Frage 

ich werde berichten ob er jetzt stärker schwingt als vorher :) bei meinen Paneelen liegt ein String morgens quasi immer im Schatten und braucht ne gute Stunde länger, als die anderen. Gucken ob sich A irgendwie auswirkt :)

Dozer-hh, 2024-07-08:
Ich habe auch einen HMS-2000 und habe ja ebenfalls das Fehlverhalten bereits beobachtet.

Habe gerade auch ein Limit von 14% = 280 W per DPL und es kommen nur 131 W an, obwohl im Haus 513 W verbraucht werden. 

Von 4 Modulen sind aktuell 3 verschattet und 1 voll in der Sonne.
Hebe ich das Limit auf und setze auf 2.000 W liefert der WR und die Module 352 W.

Settings im DPL-Modus:
Angestrebter Netzbezug: -10 W
Hysterese: 5
Minimales Leistungslimit: 50 W
Grundlast: 250 W
Minimales Leistungslimit: 2.000 W (Test!)

spcqike, 2024-07-08:
@Dozer-hh kannst du die Firmware als dem neuen PR installieren und einen log Auszug beisteuern, in dem der DPL nicht hoch skaliert? 

Wenn sich dein HMS im niedrig Last Bereich ähnlich verhält, wie meiner, kann ich deinen Fehler sogar nachvollziehen. Allerdings sollte er mit der PR Version behoben werden. 

Ich vermute auch bei dir erzeugt der WR nicht genügend Leistung, wenn du nur 14% als Limit hast. 14% bzw 280W wären 70W pro Eingang. 

deine 3 verschatteten Eingänge erzeugen in Summe ~ 70W. Dazu ~70W des nicht verschatteten Eingangs. Ergeben deine ~130W AC. 

Allerdings vermute ich, dass der nicht verschattete ebenso wie bei mir einfach nicht die erwarteten 70W liefert, sondern nur so 60W. Und daher vom Algorithmus auch als verschattet interpretiert wird. 

Teste mal mit der Version aus #1089

Dozer-hh, 2024-07-08:
@spcqike Hab sie schon runtergeladen, werde ich heute Nachmittag zu Hause installieren.
Sollte dann morgen früh um die gleiche Zeit hoffentlich ein Ergebnis bingen. Evtl. auch heute Abend, wenn die beiden SO-Module aus der Sonne sind und die beiden SW-Module noch Sonne haben.

Ich werde berichten...

spcqike, 2024-07-09:
hier meine Tabelle

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

</head>

<body link="#467886" vlink="#96607D">

Limit % | Limit absolut | DC Eingang | AC Ausgang
-- | -- | -- | --
5 | 100 | 88 - 101 | 83 - 95
10 | 200 | 191 - 201 | 181 - 192
11 | 220 | 224 | 213
12 | 240 | 224 | 213
13 | 260 | 262 | 249
14 | 280 | 262-272 | 249-259
15 | 300 | 283-322 | 267-306
16 | 320 | 343-354 | 326-336
17 | 340 | 354-364 | 336-347
18 | 360 | 375-386 | 357-367
19 | 380 | 408-418 | 388-397
20 | 400 | 418-429 | 398-408
25 | 500 | 526-536 | 500-510
30 | 600 | 626-644 | 598-612

</body>

</html>

wie man sieht, zappeln die Werte überall. (mehr als die hier genannten % Stufen habe ich nicht getestet) oder das Limit wird nicht erreicht.

hier eine rudimentäre Aufzeichnung der Stufen (hab mir leider erst im Nachgang gedacht "mensch, das wäre dochvieleicht gut das zu sehen :D, daher passt die Skalierung leider nicht so super ...) Grün - Limit, grau - AC Out. Auch da sieht man das gezappel etwas.
[image]

ab 15% erreicht er das AC Limit, vorher nicht. und um einige Watt schwankt er immer. Unter berücksichtigung des Messbereichs ist das aber denke ich OK, da es nicht mal 1% schwankt. (10W von 2000W sind halt nur 0.5%)

spcqike, 2024-07-09:
> Die Frage ist jetzt nur ob wir jetzt A oder B als 'Lösung' brauchten :D Sollen wir beides in meinem PR mal lassen oder soll A erstmal wieder rausnehmen?

ich hab es jetzt 2 Tage so laufen gehabt und kein verstärktes Schwingen festgestellt. das Limit wird genauso skaliert und gehalten, wie vorher. (nur, dass er jetzt halt aus dem 250W Loch wieder raus kommt :) )

von daher würd ich es vielleicht erstmal so lassen und weiter beobachten. 🤷‍♂️

Dozer-hh, 2024-07-09:
Also mit kurzen Kontrollen aus dem Büro per VPN sieht es mit der aktuellen Version bisher gut aus.
Zumindest liefern die Panel nicht wie bisher weniger Leistung, als sie zum Zeitpunkt, als ich geschaut habe, hätten liefern können.
Habe, als die Einspeiseleistung unter dem Hausverbrauch lag und nur 1 Panel in der Sonne stand, mal DPL deaktiviert und die Werte aller 4 Panel waren identisch zu aktiviertem DPL.