hoylabs/OpenDTU-OnBattery #1094: Anzeige erzeugter Tagesertrag falsch

GitMichaM, 2024-07-08:
### What happened?

Leider kommt es sporadisch vor, dass ein falscher Tagesertrag angezeigt wird. 
z.B. heute 131,0 kWH (bei 2 x 410W Modulen an einem HM 600) statt 3,5 kWH
Im Hoymiles-Portal wurde eine Erzeugung von etwa 3,51 KWH angezeigt.
Anbei ein Screenshot von OpenDTU-OnBattery
[image]

### To Reproduce Bug

-

### Expected Behavior

[image]

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

	2024.06.03

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

spcqike, 2024-07-09:
Lässt du die openDTU und die originale DTU dauerhaft laufen? Das kann zu Problemen bei der Signalübertragung und Daten Fehlern führen. 

Mir sind solche viel zu großen Werte noch nicht unter gekommen. 

Wenn möglich lass mal die Originale DTU ein paar Tage aus und gucke, ob das Problem besehen bleibt oder doch verschwindet.

schlimmchen, 2024-07-10:
Das wäre dann etwas für das Upstream-Projekt, da dieser Teil 1:1 übernommen ist. Wenn da ein Fehler ist, dann kommt der aus dem Upstream. Dort wird man dir sagen, was @spcqike schon schrieb: Jeder WR darf nur mit einer DTU verbunden sein, egal ob OpenDTU oder Original-DTU oder OpenDTU-OnBattery.

Bitte prüfe, ob das Problem auch auftritt, wenn du nur eine DTU (die OpenDTU-OnBattery) laufen hast.

GitMichaM, 2024-07-10:
danke für die Tipps, ich hatte in der Tat beide am laufen, das hat immer ganz gut funktioniert. 
Ich werde es mal ohne Original DTU testen. 

Hatte das so am laufen, da ich meine OpenDTU nur lokal über die Webseite abfragen kann. 
Extern über eine App fällt leider raus, da nur Vodafone-Anschluss mit DS-Lite-Tunnel vorhanden.

AndreasBoehm, 2024-07-10:
@GitMichaM Schau dir mal 'Cloudflare Tunnel' an falls du es noch nicht kennst, damit kannst du trotz DS-Lite und ohne DynDNS eine OpenDTU oder andere beliebige Dienste von extern erreichbar machen: https://bootcamp.uxdesign.cc/how-to-setup-a-cloudflare-tunnel-and-expose-your-local-service-or-application-497f9cead2d3

schlimmchen, 2024-08-18:
@GitMichaM Ich unterstelle, dass dein Problem nach dem Abschalten der Original DTU nicht mehr aufgetreten ist. Ein gleichzeitiger Betrieb von mehreren DTUs (egal welcher Geschmacksrichtung, AhoyDTU, OpenDTU, OpenDTU-OnBattery, Hoymiles DTU...) wird ausdrücklich nicht unterstützt, weil bekannt ist, dass es dann zu Problemen kommt.

Bitte melde dich wieder, falls dein Problem weiterhin besteht.

closed 2024-08-18 by schlimmchen