hoylabs/OpenDTU-OnBattery #1095: [Request] Powerqueen Charger Support

itscz-org, 2024-07-09:
### Is your feature request related to a problem? Please describe.

No problem

### Describe the solution you'd like

I use two of these 30A bluetooth powerqueen chargers: https://www.ipowerqueen.de/products/power-queen-12-24v-30amp-mppt-solarladeregler-und-bluetooth-adapter

It would be awesome if they could be supported as well in addition to the Victrons. However, there is currently no documentation of the protocol used. I already aksed Powerqueen but they didn't provide any infos. Their smartphone app is the only piece of software right now that can communicate with these chargers. I am not familiar with bluetooth traffic sniffing to do some reverse engineering on my own. I anyone can guide me i may contribute more infos, but that is all for now.

### Describe alternatives you've considered

None

spcqike, 2024-07-09:
Bluetooth may be complicated but the charger should have a rs232 or rs485 port. so that’s something to work with. 

Can you capture data from this port and provide it here?

itscz-org, 2024-07-09:
It has rs485 but i don't know how to utilize it

itscz-org, 2024-08-04:
I ordered an USB-RS485 cable i will try if i can get any infos from it.

itscz-org, 2024-10-07:
Just an update: The cable didn't fit, it had an RJ45 connector whereas the charger has something smaller (6-pin?). So no progress here yet.

drahdiwaberl, 2024-10-25:
it's an OEM product, can be found with various different brand names on it. 

https://www.litime.de/cdn/shop/files/LiTime30AMPPT12V24V-1_ad234e60-8b69-4f3a-bb5d-baf899726c50.jpg

https://www.redodopower.de/cdn/shop/files/DerMPPTSolarladereglervonRedodobestehtauseinerAluminiumlegierung_387567dc-09e8-466a-ad66-63e83d2c66c7_720x.jpg

chances are high there is some library and documentation of the connection (it's an RJ12 plug) already out there, but one would have to find out under which name..