hoylabs/OpenDTU-OnBattery #1106: Inverter Gesamtertrag wird nicht immer zurückgesetzt (sporadisch gehts)

fellpower, 2024-07-14:
### What happened?

Setup: 

- HM-400
- 2 x 12V 100Ah LiFePo4 in Reihe
- Victron 75/15 MPPT 
- 150W Panel

Egal ob ich den HM-400 neu starte per Befehl, oder ihm die DC Spannung nehme (Hängt an einem Akku), der Tagesertrag wird nur sporadisch, oder gar nicht zurückgesetzt.

DPL ist nicht aktiv, steuere alles per Node-Red. Auch den mitternächtlichen Neustart, den der WR auch brav macht, jedoch ohne erfolgreichem Zurücksetzen des Tagesertrages.

Nutze ich die "Neustarten" Funktion von OpenDTU (egal ob per Webinterface oder MQTT), startet der WR brav neu, es wird aber der Ertrag nicht zurückgesetzt, auch nicht, wenn ich dem Inverter die DC Spannung nehme (5min gewartet).

Starte ich im gleichen Atemzug die DTU neu, wird der Ertrag zurückgesetzt. Immer. Vielleicht hilft das ja.

Version: 2024.06.03

Bei meinem HM-800 - nur OpenDTU (ohne Battery), funktioniert das Zurücksetzen - per Webinterface und MQTT Neustart).

### To Reproduce Bug

Einfach den WR neu starten - Ertrag bleibt gleich

### Expected Behavior

Der Ertrag sollte zurückgesetzt werden, beim Neustart

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.06.03

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

spcqike, 2024-07-14:
Welche Einstellungen hast du denn beim WR hinterlegt? 

Ich hab folgende. Mein Ertrag wird Mitternacht zurück gesetzt, wie erwartet. 

[image]

fellpower, 2024-07-14:
[image]

Hab ich auch drin, passiert aber nix ^^

spcqike, 2024-07-14:
Wenn du möchtest, dass der Ertrag beim Neustart des WR zurück gesetzt werden soll, probiere es doch mal ohne Tagesertrag Korrektur. 

Ich dachte, die sollte ein zurücksetzen verhindern, was die WR wohl machen, wenn sie Neustarten.

schlimmchen, 2024-08-18:
> Nutze ich die "Neustarten" Funktion von OpenDTU (egal ob per Webinterface oder MQTT), startet der WR brav neu, es wird aber der Ertrag nicht zurückgesetzt, auch nicht, wenn ich dem Inverter die DC Spannung nehme (5min gewartet).

Weil Tagesertragkorretktur eingeschaltet ist. Dafür ist der Schalter da.

Die erweiterten Funktionen für die Inverter funktionieren bei batteriegespeisten Invertern nicht wie erwartet, weil die Inverter um Mitternacht herum nicht aus sind, sondern durch die Batterie weiterlaufen. Bitte *alle* diese erweiterten Funktionen ausschalten, dann wird auch der Tagesertrag zuverlässig zurückgesetzt zur Zeit des automatischen Neustarts durch den DPL.

schlimmchen, 2024-08-18:
Verwandt (gleiches Problem): #1128.

fellpower, 2024-08-18:
Jetzt kommt aber das eigentliche Problem: Ich nutze den DPL nicht, weil ich mir selbst was gebaut habe, was meine Batterie verwaltet. Und darum wird er nicht zurückgesetzt. Vielleicht könnte man den Neustart des WR mal vom DPL abkoppeln? Ist der DPL an, geht, ist er aus, nicht - mit dem Neustart ^^

schlimmchen, 2024-08-18:
> weil ich mir selbst was gebaut habe, was meine Batterie verwaltet. Und darum wird er nicht zurückgesetzt.

Dann kann diese Software auch den Wechselrichter neustarten, z.B. über MQTT mit "cmd/restart", siehe [Doku](https://opendtu-onbattery.net/firmware/mqtt_topics/#inverter-limit-specific-topics).

> Vielleicht könnte man den Neustart des WR mal vom DPL abkoppeln?

Das könnte man machen. Wer macht das denn? Und warum? Und wie sieht das dann aus?

Bitte starte den WR über MQTT oder die WebApi neu durch die von dir eingesetzte Software und melde dich, falls das dein Problem nicht löst. Jedenfalls sehe ich keinen Handlungsbedarf in OpenDTU-OnBattery. Die erwähnten Advanced-Schalter funktionieren nicht an batteriegespeisten WR und der DPL macht wunderbar einen Neustart des von ihm geregelten Inverters und da wird dann auch der Daily Yield zurückgesetzt.

closed 2024-08-18 by schlimmchen