hoylabs/OpenDTU-OnBattery #1111: Battery-Limiter-Mode (Alternative solution of the "Load correction factor“)

SW-Niko, 2024-07-16:
### Is your feature request related to a problem? Please describe.

Die aktuelle Berechnung des Batterie-Stop-Schwellenwertes mit Hilfe des „Load correction factors“ ist ein gutes Feature hat aber einige Schwachstellen:

- Der Anwender muss für sein System den richtigen „Load correction factor“ ermitteln.
- Der „Load correction factor“ ist nicht statisch. Er unterliegt langsamen Änderungen. (Temperatur, Alterung, Ladungszustand)
- Der „Load correction factor“ ändert sich, wenn sich die Position der Spannungsmessung ändert. (Inverter / MPPT / SmartShunt / BMS)
- Die Berechnung hat einen Fehleranteil, wenn die Energie ganz oder teilweise vom MPPT oder vom AC-Charger geliefert wird.
- Die Unterspannungs-Abschaltung verwendet für die Berechnung die aktuelle Leistung und die aktuelle Spannung. Sie ist damit immer einen Regelschritt zu spät.

Ich denke das keine der oben aufgeführten Schwachstellen ein ernstes Problem verursacht. Wer ein BMS hat, hat auch eine weitere Sicherheit, die eine Tiefentladung der Batterie verhindert. Üblicherweise wird die Batterie-Stop-Schwellenwerte auf 10% - 20% SOC eingestellt und man bekommt dadurch einen zusätzlichen Puffer.

### Describe the solution you'd like

Bei der Suche nach einer Lösung bin ich auf einen etwas anderen Ansatz gekommen, der die Schwachstellen umgeht und darüber hinaus weitere Vorteile bietet.

Der Battery-Limiter begrenzt nahe der Batterie-Stop-Schwelle die maximal entnehmbare Leistung. Diese Grenze wird vorausschauend und dynamisch aus der aktuellen Batteriespannung berechnet.

**Dynamisch:** Die Leistungsgrenze folgt der Batteriespannung. Je näher man der  Batterie-Stop-Schwelle kommt desto kleiner wird die entnehmbare Leistung.
**Vorausschauend:** Es werden nur Leistungen zugelassen die nicht zu einer Unterschreitung des Batterie-Stop-Schwellenwertes führen würden.

Erst wenn die Leistung bis zum minimalen Leistungslimit reduziert wurde und der Batterie-Stop-Schwellenwert unterschritten ist wird die Energieentnahme aus der Batterie gestoppt. (Abschaltung des Inverters)

Damit werden alle oben aufgeführten Schwachstellen geschlossen.

### Describe alternatives you've considered

Ein andere Möglichkeit, die zu mindestens einen Hauptteil der Schwachstellen löst, wäre die dynamische Berechnung des aktuellen „Load correction factors.“

PS: Der Batterie-Limiter hat das als Unterfunktion quasi schon im Gepäck.

### Additional context

Wie schon angedeutet kommen mit dem Battery-Limiter Ansatz automatisch noch weitere Vorteile:

- Sanftes Anfahren der Leistungsentnahme bei Spannungen nahe der Batterie-Stop-Schwelle. Zum Beispiel: morgens, wenn die Batterie sehr entladen ist und die Sonne wieder Energie liefert.
- Tagsüber wenn wenig Energie von der Sonne kommt wird der Inverter quasi auf die Sonnenenergie geregelt. Eine Art „Solar-Passthrough“ an der Batterie-Abschalt-Schwelle.
- Durch das „sanfte“ runter und hoch regeln kann die Batterie-Stop-Schwelle und die Batterie-Start-Schwelle sehr nahe zusammengelegt werden. Mehrfaches Ein- und Ausschalten bei stark wechselnder Sonne und wechselnder Last wird unterbunden.
- Der „Load correction factor“ fällt weg und es werden keine zusätzlichen Konfigurationsparameter benötigt.
- Der Battery-Limiter stellt sich selbstständig auf die individuelle Anlage ein. Ein nachjustieren wegen sich ändernder Temperatur, Alterung, Ladungszustand usw. ist nicht notwendig.
- Damit wird auch #676 gelöst

Der Battery-Limiter läuft auf meiner Anlage seit ein paar Wochen und nun bin ich dabei aus der spezial Lösung für meine Anlage eine Lösung für möglichst viele Anlagen zu generieren.

SW-Niko, 2024-07-16:
Jetzt habe ich gleich mal eine Frage zu Hardware die ich nicht habe. Welche Spannungsauflösung haben die Geräte?
Die Victron MPPTs haben 10mV
Die Victron SmartShunts vermutlich ebenfalls 10mV
Die BMS liegen vermutlich bei 1mV? Oder gibt es auch welche mit 10mV?
Die Hoymiles Inverter vermutlich alle bei 100mV?

Squint5471, 2024-09-03:
> ### Is your feature request related to a problem? Please describe.
> 
> Die aktuelle Berechnung des Batterie-Stop-Schwellenwertes mit Hilfe des „Load correction factors“ ist ein gutes Feature hat aber einige Schwachstellen:
> 
>     * Der Anwender muss für sein System den richtigen „Load correction factor“ ermitteln.
> 
>     * Der „Load correction factor“ ist nicht statisch. Er unterliegt langsamen Änderungen. (Temperatur, Alterung, Ladungszustand)
> 
>     * Der „Load correction factor“ ändert sich, wenn sich die Position der Spannungsmessung ändert. (Inverter / MPPT / SmartShunt / BMS)
> 
>     * Die Berechnung hat einen Fehleranteil, wenn die Energie ganz oder teilweise vom MPPT oder vom AC-Charger geliefert wird.
> 
>     * Die Unterspannungs-Abschaltung verwendet für die Berechnung die aktuelle Leistung und die aktuelle Spannung. Sie ist damit immer einen Regelschritt zu spät.
> 
> 
> Ich denke das keine der oben aufgeführten Schwachstellen ein ernstes Problem verursacht. Wer ein BMS hat, hat auch eine weitere Sicherheit, die eine Tiefentladung der Batterie verhindert. Üblicherweise wird die Batterie-Stop-Schwellenwerte auf 10% - 20% SOC eingestellt und man bekommt dadurch einen zusätzlichen Puffer.
> ### Describe the solution you'd like
> 
> Bei der Suche nach einer Lösung bin ich auf einen etwas anderen Ansatz gekommen, der die Schwachstellen umgeht und darüber hinaus weitere Vorteile bietet.
> 
> Der Battery-Limiter begrenzt nahe der Batterie-Stop-Schwelle die maximal entnehmbare Leistung. Diese Grenze wird vorausschauend und dynamisch aus der aktuellen Batteriespannung berechnet.
> 
> **Dynamisch:** Die Leistungsgrenze folgt der Batteriespannung. Je näher man der Batterie-Stop-Schwelle kommt desto kleiner wird die entnehmbare Leistung. **Vorausschauend:** Es werden nur Leistungen zugelassen die nicht zu einer Unterschreitung des Batterie-Stop-Schwellenwertes führen würden.
> 
> Erst wenn die Leistung bis zum minimalen Leistungslimit reduziert wurde und der Batterie-Stop-Schwellenwert unterschritten ist wird die Energieentnahme aus der Batterie gestoppt. (Abschaltung des Inverters)
> 
> Damit werden alle oben aufgeführten Schwachstellen geschlossen.
> ### Describe alternatives you've considered
> 
> Ein andere Möglichkeit, die zu mindestens einen Hauptteil der Schwachstellen löst, wäre die dynamische Berechnung des aktuellen „Load correction factors.“
> 
> PS: Der Batterie-Limiter hat das als Unterfunktion quasi schon im Gepäck.
> ### Additional context
> 
> Wie schon angedeutet kommen mit dem Battery-Limiter Ansatz automatisch noch weitere Vorteile:
> 
>     * Sanftes Anfahren der Leistungsentnahme bei Spannungen nahe der Batterie-Stop-Schwelle. Zum Beispiel: morgens, wenn die Batterie sehr entladen ist und die Sonne wieder Energie liefert.
> 
>     * Tagsüber wenn wenig Energie von der Sonne kommt wird der Inverter quasi auf die Sonnenenergie geregelt. Eine Art „Solar-Passthrough“ an der Batterie-Abschalt-Schwelle.
> 
>     * Durch das „sanfte“ runter und hoch regeln kann die Batterie-Stop-Schwelle und die Batterie-Start-Schwelle sehr nahe zusammengelegt werden. Mehrfaches Ein- und Ausschalten bei stark wechselnder Sonne und wechselnder Last wird unterbunden.
> 
>     * Der „Load correction factor“ fällt weg und es werden keine zusätzlichen Konfigurationsparameter benötigt.
> 
>     * Der Battery-Limiter stellt sich selbstständig auf die individuelle Anlage ein. Ein nachjustieren wegen sich ändernder Temperatur, Alterung, Ladungszustand usw. ist nicht notwendig.
> 
>     * Damit wird auch [Load corrected voltage is too high in some scenarios #676](https://github.com/helgeerbe/OpenDTU-OnBattery/issues/676) gelöst
> 
> 
> Der Battery-Limiter läuft auf meiner Anlage seit ein paar Wochen und nun bin ich dabei aus der spezial Lösung für meine Anlage eine Lösung für möglichst viele Anlagen zu generieren.

Super Vorschlag, an diese Begrenzung hatte ich auch schon gedacht. Es kommt immer wieder vor, dass ein großer Verbraucher die Abschaltung bei niedrigem SOC auslöst und kurz danach läuft das System wieder mit kleineren Leistungen an.

spcqike, 2024-09-03:
Passt das irgendwie in #1198 ?

SW-Niko, 2024-09-03:
Hallo @Squint5471,
ja genau das ist eines der Szenarien die dadurch verhindert werden. 

Hallo @spcqike,
ja, das geht in die gleiche Richtung. Aber mein Lösungsansatz arbeitet auch mit dummen Batterien, die keine Stromgrenze liefern.
Ich regle variabel und vorausschauend die Inverter Leistung runter wenn ich mich dem unteren Abschaltlimit nähere.
Aber ... ich muss #1198 zuerst mal komplett durchlesen, dann kann ich die Unterschiede auflisten

schlimmchen, 2024-11-16:
Abstrakt gesprochen macht das Sinn. Der load correction factor ist mir schon immer ein Dorn im Auge (weil er trotz Doku zu kompliziert ist). Heute habe ich ein weiteres Problem formuliert in #1399.

Gibt's denn schon Code hierzu, den man mal beäugeln könnte?

Wie stellst du denn sicher, dass der Ansatz bei jedem System sinnvoll funktioniert? Also dass er grundsätzlich gut funktionieren kann macht Sinn, aber was zum jeweiligen Zeitpunkt eine vernünftige Grenze für die Entnahme an Leistung aus der Batterie ist, hängt doch von einer Reihe Faktoren ab...

SW-Niko, 2024-11-17:
@schlimmchen ,
Jetzt wirst du mir langsam unheimlich. Ich bin seit dieser Woche wieder an der Geschichte dran.
Mal ganz kurz ein paar Stichpunkte was mein Battery-Manager schon alles macht.
- Rechnet den Ri der Batterie mit
- Rechnet die Leerlaufspannung der Batterie mit
- Limitiert den Inverter wenn die Batteriespannung nahe der Abschaltspannung ist
- Sorgt dafür das die Batterie immer wieder mal vollgeladen wird (z.B einmal im Monat)

Das funktioniert auf meinem System schon ganz prima. Aber ich muss noch einiges einbauen damit es auf möglichst vielen (allen) Systemen laufen kann. Ich mache auch einen PR wenn ich soweit bin.

>Wie stellst du denn sicher, dass der Ansatz bei jedem System sinnvoll funktioniert?
Genau das ist der kritische Punkt. 
Ich kann die benötigten Parameter selbstständig ermitteln und laufend anpassen, wenn ich ausreichende Messdaten bekomme.
Der SmartShunt kann das, aber ob alle Battery-Provider (BMS) das auch können weiß ich nicht und deshalb arbeite ich noch an einer Notstrategie.

SW-Niko, 2024-12-03:
Hallo @schlimmchen ,
ich habe mich jetzt einige Zeit mit der Thematik auseinander gesetzt und bin zu folgendem Ergebnis gekommen.

- Der "load correction factor" ändert sich stark. Viele Faktoren spielen da mit.
- Vernünftig funktioniert das nur unter allen Bedingungen wenn Spannung und Strom/Leistung mit ausreichender Genauigkeit an der Batterie gemessen werden (z.B. mit einem Smart Shunt)
- Mein Ansatz mit der Reduzierung der Leistung ist auch davon abhängig wo die Spannung gemessen wird aber er arbeitet genauer und bietet weitere Vorteile.

>Wie stellst du denn sicher, dass der Ansatz bei jedem System sinnvoll funktioniert? 

Ich erkenne wenn die Regelung über die Reduzierung der Leistung nicht funktioniert und schalte dann den Inverter auf das untere Power Limit. Man könnte aber dann auch als automatisches Fallback zurück auf die Lösung mit dem "load correction factor" gehen.

Wenn noch Interesse besteht dann fasse ich meine Erfahrungen zusammen und mache einen PR.

schlimmchen, 2025-01-20:
Ich freue mich immer noch auf dieses Feature! Die Batteriespannung erholt sich, wie man hier sieht, auf 51.5V, nachdem das Entladen beendet wurde. Entladen wird zwar bis unter der Stop-Threshold (51.0V), aber dennoch wird eigentlich früher als gewünscht gestoppt.

[image]

Dein Ansatz, den Entladestrom nahe des Stop-Schwellenwertes zu begrenzen, macht für mich immer noch Sinn. Im Szenario oben wäre der "Rebounce" nicht so stark gewesen, wenn er weiter mit der Leistung entladen hätte wie zwischen 8 und 10 Uhr.

SW-Niko, 2025-01-23:
Ja, dieses Verhalten sehe ich auch. 
Je kleine die Leistung desto kleiner der "Rebounce".

Mal rechnen ..
Bei dir, 48V, 10A, Rebounce ca 500mV
Bei mir, 24V, 1.2A, Rebounce ca 80mV

Man wird den Rebounce nicht ganz los. Aber wenn man immer bei der gleichen Leistung / Spannung abschaltet dann ist der Rebounce ziemlich konstant und das kann man bei der Konfiguration berücksichtigen.

Das ist ein log von heute.
Im wesentlichen steht da:
Das der Limiter um 13:39 bei 80W angefangen hat zu limitieren. In 5 Schritten von 80W auf 28W reduziert hat. 
Und um 14:02 wurde angeschaltet, weil bei 28W das untere Limit von 26V unterschritten wurde.

16:33:13.507 > [Battery-Guard] 2) Feature: Low voltage power limiter
16:33:13.554 > [Battery-Guard] Status: Idle, Limiting active: No, Actual limit: 80W (Range: 28W - 80W)
16:33:13.599 > [Battery-Guard] Quality of limiting: Excellent, Quality counter: 1.00 (Min: 1, Max: 1, Amount: 5)
16:33:13.645 > [Battery-Guard] Quality of limiting: Maximum safety counter: 1, Forced to use the lower power limit: 0
16:33:13.693 > [Battery-Guard] Last power limiting start - stop time: 13:39 - 14:02
16:33:13.871 > [Battery-Guard] Last stop data - Time: 23.01.2025 14:02, Reason: Below Limit, Voltage: 25.997V
16:33:13.926 > [Battery-Guard] Target voltage range for power limit regulation: 26.000V - 26.020V

Es wird ...... langsam.

hhoebel, 2025-04-10:
Ich finde dieses Feature auch sehr spannend und würde mich darüber freuen. Kann man denn was dazu beitragen/helfen/testen?