hoylabs/OpenDTU-OnBattery #1120: MQTT powermeter reading wrong values (1 phase)

AndreasBoehm, 2024-07-22:
### What happened?

I did setup my powermeter via MQTT because it was unreliable when i hit it with a interval of 1 second via HTTP.

During the initial setup i already noticed that there is some issue when you use a MQTT powermeter with a total value (1 phase) instead of providing one value for each phase. 

But today something else happened, the sum of all phases was `-NaN`

### To Reproduce Bug

Use the following settings and let it sit for a bit.

[image]

### Expected Behavior

Powermeter works even if only one value is provided

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

	b32382d

### Relevant log/trace output

```Shell
12:35:29.853 > [PowerMeterMqtt] Topic 'tele/smart_meter_lights/SENSOR': new value: 180.00, total:  -nan
```

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

schlimmchen, 2024-07-22:
@DerHavey1994 has experienced this as well: https://github.com/helgeerbe/OpenDTU-OnBattery/pull/1077#issuecomment-2211682579

I have not experienced this before. The differences are that I use three topics and no path, i.e., no JSON parsing. I switched to a single MQTT topic (I have the total available in a different topic) and it broke immediately:

```
15:12:09.889 > [PowerMeterMqtt] Topic 'grid/power/total/watts': new value: -174.82, total: 3158.08
```

It seems that the old values are somehow accumulated?! I will look into it.

DerHavey1994, 2024-07-22:
I currently have it set without the optional json path & don't get any strange values ​​displayed. The numbers after the decimal point also work.

schlimmchen, 2024-07-22:
I made two mistakes:
1. The values in the array of values the MQTT power meter can manage were not initialized (to zero). The sum would consist of uninitialized values/data. :man_facepalming: 
2. Access to the target variable was not protected using a mutex in all cases. In particular, the previous use of the mutex was only sufficient when using no JSON path, no unit other than Watts, and no sign inversion, and also no debug logging...

Issue 1. explains one type of findings by you and me: Incorrect values. It also explains why I did not observe issues until now, as I was using three topics, so all values are eventually set to valid values.

Issue 2. explains well that @DerHavey1994 problems went away after stopping to use JSON encoded values. It can also explain why you observed that the total only degraded after a while rather than instantly.

schlimmchen, 2024-07-22:
@DerHavey1994 If you like, check out the firmware of the [PR build run of #1122](https://github.com/helgeerbe/OpenDTU-OnBattery/actions/runs/10043705640). I am confident that your issue (total is NaN) will not occur again.

@AndreasBoehm I merged #1122 already as I assumed that is the best way for you to integrate them on your side?

Please re-open the issue if the changes in #1122 are insufficient.

PR #1122 (2024-07-22): MQTT powermeter fixes
* initialize values/memory
* protect all access to common memory with mutex

closes #1120.
closed 2024-07-22 by schlimmchen