hoylabs/OpenDTU-OnBattery #1123: Stop Threshold for Battery Discharging never reached.

greymda, 2024-07-22:
### What happened?

Set up Stop Threshold for Battery Discharging at 16% but won't get there. For some reason when reaching below the Start Threshold for Battery Discharging which is set to 25% (in the screenshot it's 24% currently), battery usage stops.

[image]

[image]

### To Reproduce Bug

1. Set Stop Threshold for Battery Discharging at 16%
2. Set Start Threshold for Battery Discharging at 25%
3. Let battery drain to 24%
4. Observe the battery stops being drained, although it's not 16% yet, but only below Start Threshold for Battery Discharging value.

### Expected Behavior

To drain battery until Stop Threshold for Battery Discharging value is reached.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.03.23

### Relevant log/trace output

```Shell
18:37:39.748 > [DPL::loop] ******************* ENTER **********************
18:37:39.750 > [DPL::loop] battery interface enabled, SoC: 24 %, StartTH: 25 %, StopTH: 16 %, SoC age: 0 s, ignore: no
18:37:39.804 > [DPL::getBatteryVoltage] BMS: 52.25 V, MPPT: 52.38 V, inverter: 52.40 V, returning: 52.25V
18:37:39.807 > [DPL::loop] dcVoltage: 52.25 V, loadCorrectedVoltage: 52.35 V, StartTH: 50.00 V, StopTH: 49.00 V
18:37:39.809 > [DPL::loop] StartTH reached: no, StopTH reached: no, SolarPT enabled, use at night: yes
18:37:39.859 > [DPL::calcPowerLimit] battery use prevented, solar power (DC): 104 W
18:37:39.863 > [DPL::calcPowerLimit] power meter: 383 W, target consumption: 10 W, inverter output: 97 W, solar power (AC): 97
18:37:39.866 > [DPL::calcPowerLimit] limited to solar power: 97 W
18:37:39.869 > [DPL::setNewPowerLimit] input limit: 97 W, lower limit: 10 W, upper limit: 800 W, hysteresis: 10 W
18:37:39.872 > [DPL::setNewPowerLimit] inverter max: 800 W, inverter is producing, requesting: 97 W, reported: 104 W, diff: 7 W
18:37:39.874 > RX Period End
```

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

spcqike, 2024-07-22:
Out of curiosity. Is it possible that your DTU resets and restarts?

greymda, 2024-07-22:
> Out of curiosity. Is it possible that your DTU resets and restarts?

wouldn’t it be visible in the logs or in Info page? if si, then uptime is 21h ..

Manos1966, 2024-07-23:
Full Solar Passthrough 0% 😨

spcqike, 2024-07-23:
>  if so, then uptime is 21h ..

yeah its only visible in the info screen.

i thought, maybe it restarts and therefore forgets, where it is. like,  if you start at 19%, are you charging up to 25% or are you allowed to discharge to 16%?

i dont have a battery, so i'm not to sure with its DPL settings. maybe its because "discharge at night"? as 18:37 isn't night. in your log you are only using solarPT power. what happens at like 21:00 or whatever time your NTP night starts? (or was it if solar power is < 20W?)

Manos1966, 2024-07-23:
- Full Solar Passthrough Start Threshold  auf 80% stellen
- speichern
- Aktiviere Solar-Passthrough Deaktivieren
- speichern

...und nochmals testen

greymda, 2024-07-23:
@Manos1966 I don't need Full Solar Passthrough, hence the 0% (from the Wiki, it means it's deactivated). I'm not in Germany, we are not allowed here to feed into grid.

Manos1966, 2024-07-23:
I know, as I said above, make the changs, deaktivate it and see if it solved the issue.

Manos1966, 2024-07-23:
@greymda sorry, I was writting German before...

Just as a side-information when the battery is reaching its 100% SOC:
The Victron-MPPT is charging the battery according to the Voltage.
When the battery is reaching the full 100%, the Victron stops charging the battery any further (provided you have setup your Victron MPPT correctly) = The Victron MPPT stops producing.
Full Solar Passthrough works by taking the production (Watt) of the Victron MPPT and sets the inverter to produce the same Wattage (plus/minus some adjustments).
Since the Victron MPPT is producing nothing, the Full Solar Passthrough will also tell the inverter to produce nothing.

All in all, what I am trying to say is: You do not need to worry about feeding into grid because of the Full Solar Passthrough.

schlimmchen, 2024-07-23:
> All in all, what I am trying to say is: You do not need to worry about feeding into grid because of the Full Solar Passthrough.

That is only true if the charge controller always already entered absorption *before* the SoC is 100%. Not reliable.

Setting the full solar passthrough start SoC to 0 % and the respective voltages to >60 V is the correct approach, as documented in the wiki and as @greymda pointed out.

> maybe its because "discharge at night"? [...] (or was it if solar power is < 20W?)

@spcqike is on the right track. Yes, "night" is when the solar output drops below 20W (this is still hardcoded).

And the issue @greymda experiences is indeed a bug: If solar passthrough is enabled, and "use battery at night" is also enabled, we shall **enable** battery discharging **once** it becomes night (solar output < 20W). However, we set "use battery" to "it is night" if those two switches are enabled. What we want is to update "use battery" only if "use battery" is currently false.

PR #1126 (2024-08-15): Fix DPL nighttime discharging
the switch "always start discharging battery at night" would cause to stop discharging the battery when there was solar power and the battery was discharged below the start threshold.

this change introduces a nighttime discharging boolean variable, which is enabled the instant we decide to start a battery discharge cycle due to nighttime havin arrived. we reset this variable as soon as it is daytime (solar power available). in that case, we allow discharging the battery if the start threshold was reached. this can actually be the case if the battery is charged wi [... cut]
closed 2024-08-15 by schlimmchen