hoylabs/OpenDTU-OnBattery #1124: No URL input field available in settings/powermeter <option value="3">HTTP(S) + JSON</option>

ibr, 2024-07-23:
### What happened?

try to set connection to Shelly 3em, nice example box, but cannot enter URL, no field 

### To Reproduce Bug

 open settings/powermeter

Form option:
<option value="3">HTTP(S) + JSON</option>

### Expected Behavior

URL Field visible

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

v24.6.29-9-g0013b2c

### Relevant log/trace output

```Shell
<div class="alert alert-secondary mt-5" role="alert"><h2>URL examples:</h2><ul><li>http://admin:secret@shelly3em.home/status</li><li>https://admin:secret@shelly3em.home/status</li><li>http://tasmota-123.home/cm?cmnd=status%208</li><li>http://12.34.56.78/emeter/0</li></ul><h2>JSON path examples:</h2><ul><li>total_power - { "othervalue": "blah", "total_power": 123.4 }</li><li>testarray/[2]/myvalue - { "testarray": [ {}, { "power": 123.4 } ] }</li></ul> More info: <a href="https://github.com/mobizt/FirebaseJson">https://github.com/mobizt/FirebaseJson</a></div>

Message after submit:
URL must either start with http:// or https://!
```

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

spcqike, 2024-07-23:
Ist es vielleicht ähnlich meinem Problem von hier? 

https://github.com/helgeerbe/OpenDTU-OnBattery/pull/1077#issuecomment-2200817594

Das sollte eigentlich gefixt wurden sein. Zumindest ging es nach schlimmchens Anpassungen dann wie erwartet.

schlimmchen, 2024-07-23:
> URL Field visible

What does it look like? Please share a screenshot?

> Self-Compiled

So you probably forgot to build the webapp beforehand? https://www.opendtu.solar/firmware/compile_webapp/ If possible, use a binary from one of the build action runs of the development branch, like https://github.com/helgeerbe/OpenDTU-OnBattery/actions/runs/10046067064.

ibr, 2024-07-23:
I used vscode on MacOSX, compiled and flashed OpenDTU on fusion board, connected to HMS800, found everything working, then compiled and flashed OpenDTUOnBattery, so maybe old config is still around??

[image]

MQTT settings are also not stored...

schlimmchen, 2024-07-23:
So you probably forgot to build the webapp beforehand? https://www.opendtu.solar/firmware/compile_webapp/

ibr, 2024-07-24:
Sorry, new to the project, tried yarn dev, which shows all input fields, will try to flash again.

closed 2024-07-24 by ibr