hoylabs/OpenDTU-OnBattery #1128: "Inverter Total Yield Day" does not reset on schedule

greymda, 2024-07-24:
### What happened?

Inverter Total Yield Day is not being reset to zero, although the setting in DPL is set and from the Event log it seems all the prerequisites were done. Attaching screensots:

[image]

[image]

[image]

### To Reproduce Bug

Set all the parameters as in screenshots.

### Expected Behavior

The Inverter Total Yield Day should be reset to zero at 00:00 (as per the setting).

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.03.23

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

spcqike, 2024-07-24:
whats your inverter setting?

[image]

schlimmchen, 2024-07-24:
How many inverters are polled by your DTU? Only the one managed by the DPL will be restarted at midnight.

greymda, 2024-07-24:
@schlimmchen  I have only one inverter.

greymda, 2024-07-24:
@spcqike 
[image]

spcqike, 2024-07-24:
Well, maybe you can try to use the latest firmware.

greymda, 2024-07-24:
I don't see any mentions of a fix in the release notes for this particular functionality...

spcqike, 2024-07-24:
maybe its just a missmatch / bad  timing.

DPL restarts the inverter. when the inverter restarts, it resets its internal daily yield value. 

as the reset happens everytime the inverter restarts, there is the "yield day correction" in opendtu software. so a very dark cloud or DC disconnect doesn't reset your software daily yield (but it still does your inverters internal yield)

the day yield correction probably also resets at midnight. so if your inverter resets like 5s after "day yield correction" resets "day yield", software "day yield" may be catch the previous inverter yield value.

you can try to restart your inverter like 23:58 and check, if it helps.

but TBH, idk. i don't really use day yield values. i see them in opendtu webui, but i don't care, as all monitoring and reporting is based on totalYield anyway.

greymda, 2024-07-25:
the issue continues.

the Event log shows the inverter restart at 00:00 but the Daily Yield is incorrect (I wish i had such yield by night in reality).

Moreover it somehow messes up my HA Energy stats as well :(

[image]

[image]

greymda, 2024-08-04:
unfortunately this behavior still continues, although i set “restart inverter at 01:00”.
one thing i noticed is that this occurs when i have no battery left and the inverter is yellow during the night.

does it make any difference?
[image]

schlimmchen, 2024-08-18:
I tried to reproduce this given your additional info that this happens when the inverter is in standby, but I failed to reproduce it.

You have "yield day correction" enabled. Why? This is intended to keep the daily yield even if the inverter restarts, which is in conflict with what you want to achieve. Disable all of these advanced switches, as they are not working as expected with battery-powered inverters. In this case, the yield correction values buffered in OpenDTU are indeed reset at midnight, but as the inverter is still reachable, because it is battery-powered, the next statistics answer from the inverter will simply overrule any corrections.

greymda, 2024-08-18:
thanks for looking into this, still all these settings are by default- i haven’t changed anything here.
will try disabling it and report back.

schlimmchen, 2024-08-18:
Oh boy... Those settings enabled would be something we need to fix.

The code says that the default values are "false", i.e., all those features are disabled by default:

https://github.com/helgeerbe/OpenDTU-OnBattery/blob/51624669b0d1e9977722f1dc9e815d4b60014136/src/Configuration.cpp#L506-L509

Maybe the web app enables them by default when adding a new inverter? Hm, does not seem like it does. That leaves me confused as to why these settings should have become enabled without you actively enabling them.

Now that you disabled them, we'll wait for you to report back and then we'll see about these settings being on by default.

greymda, 2024-08-29:
@schlimmchen reporting back that since i've disabled the "yield day correction" option, the stats seems to be correct. thanks for pointing out the solution!

closed 2024-08-29 by schlimmchen