hoylabs/OpenDTU-OnBattery #1129: [Request] Live Data screen: State of charge in Voltage instead of SoC

greymda, 2024-07-24:
### Describe the solution you'd like

Should a user enable in Dynamic Power Limiter Settings to Ignore Battery SoC, the Live Data screen should reflect Voltage instead of Percentile.

[image]

[image]

AndreasBoehm, 2024-07-24:
What if we add the `Voltage` to live data for all configurations?
Would also help users with a MQTT battery to see all available data of their battery.

What do you think @schlimmchen ?

schlimmchen, 2024-07-24:
Yes, absolutely. Even though it breaks with the current layout (there is no total card with two values yet), I would like to suggest that the "State of charge" card is changed "Battery" and it lists SoC and/or Voltage like so:

xx %

or

xx % / yy.z V

or

yy.z V

Depending on the info available (MQTT Battery might have only one of those).

Reason: There are too many cards already.

Maybe we can cram in discharge power as well.

swingstate, 2024-07-24:
agree

greymda, 2024-07-24:
Although I like the idea to have both % and V, at the same time I find a bit frustrating seing permanently a number that does not brings me any value per se - in my case SoC from JK BMS is all over the place. 
Hence, I'd like to have the opportunity to chose what I want to have it displayed, maybe some checkboxes like:
✓ SoC
✓ Voltage

schlimmchen, 2024-07-24:
> Hence, I'd like to have the opportunity to chose what I want to have it displayed, maybe some checkboxes like:

Sorry, no. That kind of fine-grained personalization will require you to build your own firmware, as adding a configuration switch for that is overkill.

AndreasBoehm, 2024-07-24:
[image]
I can create an MR later today to also add a discharge power card, or do you want to put that into the same card?

schlimmchen, 2024-07-24:
> But the switch already exists- the Ignore SoC.

@greymda You are absolutely right! I did not think of that. I stand by my opinion that an extra switch "please do not show me the battery SoC in the live view" is too much, but the "Ignore SoC" switch could indeed also control whether or not to show the SoC in the totals :heart: 

> I can create an MR later today that also includes discharge power

@AndreasBoehm :heart: 

How do you decide whether or not to show SoC and/or voltage when the MQTT batter provider is used? I think the code is lacking... The BatteryStats assume that all providers deliver SoC, but in case of MQTT, now that voltage can also/alternatively be processed, it is not clear whether the BatteryStats has a valid SoC?

schlimmchen, 2024-07-24:
> But the switch already exists- the Ignore SoC.

On the other hand... Not sure that I want to muss the SoC in the total :grin: In my case, the SoC is not too far off, at least not now in the summer where the battery regularly hits 100% (and 0%). It gives a good and quick indication about the SoC, whereas the voltage need to be interpreted. Still, I want to have the voltages control the DPL. And the switch is also part of the DPL, as it tells the DPL to ignore the SoC. Maybe I *should* learn to interpret the voltage? Hm. Not sure about this. @swingstate ? @AndreasBoehm ?

AndreasBoehm, 2024-07-24:
I think that its counter intuitive that a setting for the DPL affects if the SoC is shown on the liveview or not.

AndreasBoehm, 2024-07-24:
> How do you decide whether or not to show SoC and/or voltage when the MQTT batter provider is used? I think the code is lacking... The BatteryStats assume that all providers deliver SoC, but in case [...]

BatteryStats provides methods to check if the SoC or voltage is valid based on a last updated timestamp for each value, i will work based on that and add the same mechanics for current.

greymda, 2024-07-24:
> ![Screenshot 2024-07-24 at 14 58 27](https://private-user-images.githubusercontent.com/1270749/351723130-ef0fd484-5daf-48fd-9e61-80ced5cddc61.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE4Mjk5MzEsIm5iZiI6MTcyMTgyOTYzMSwicGF0aCI6Ii8xMjcwNzQ5LzM1MTcyMzEzMC1lZjBmZDQ4NC01ZGFmLTQ4ZmQtOWU2MS04MGNlZDVjZGRjNjEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MjRUMTQwMDMxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODBjMDQ2YmUyNTQwYmY4ZTRiNTcyNjNhMjEyMGUzY2UxMzFhM2MzNDg2N2ZjMDRjMTYwNjFiM2M3ZjY0YmIzOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.LEzxvWT_BxO2nRW6z9zRD_U5Aw88iJyLYhgrokZUlbc) I can create an MR later today that also includes discharge power

By the way, your screen reflects why i put this request in the first place.
How can 52.62V represent only 51% SoC? Maybe I'm missing something out...

AndreasBoehm, 2024-07-24:
> By the way, your screen reflects why i put this request in the first place. How can 52.62V represent only 51% SoC? Maybe I'm missing something out...

Don't trust the data that are shown in my screenshot, my battery setup is not done yet and i am still using my mocked CAN messages to test/develop OpenDTU OnBattery.

closed 2024-07-27 by schlimmchen