hoylabs/OpenDTU-OnBattery #1154: [Request] AC Ladegerät, angestrebter Netzbezug Limit erhöhen

woollfi, 2024-08-07:
### Describe the solution you'd like

höheres Limit als -3000 für angestrebten Netzbezug bei AC Ladegerät

Solarteur, 2024-08-07:
An welches AC-Ladegerät hast du denn gedacht? 

Die bisher möglichen 3000W (an einer Phase) sind schon "sportlich".

woollfi, 2024-08-07:
Leistung passt schon, ich meinte den angestrebten Netzbezug, der ist bei -3000W begrenzt, gut wäre da kein Limit nach oben, das man mehr variieren kann.

Lg

AndreasBoehm, 2024-08-08:
@woollfi Könntest du einen Screenshot anhängen der erklärt um welches Eingabefeld es geht? Danke

woollfi, 2024-08-08:
[image]
[image]
[image]
[image]

closed 2024-08-08 by schlimmchen