hoylabs/OpenDTU-OnBattery #1164: Pro PCB: 3x MPPT + Pylontech gleichzeitig funktioniert nicht

DerHavey1994, 2024-08-12:
### What happened?

@schlimmchen: In #1018 & #1029 wurde bereits darüber berichtet.
Ich benutze seit neustem das Pro PCB von swingstate. Ich bekomme es aber nicht hin 3 MPPT sowie die Pylontech gleichzeitig zum laufen zu bekommen.
Sobald man ein Pin Mapping mit den 4 Geräten aktiviert stürzt die DTU ab. (Im Display "OpenDTU!")

Es funktioniert:
- 2x MPPT + Pylontech
- 3x MPPT
- Pin Mapping mit allen 8 Pins vergeben, wenn vorher die Batterie Schnittstelle deaktiviert wurde

Ich habe die DTU am Schreibtisch fertig geflasht und vorkonfiguriert, ohne ein Endgerät einzustecken, dort ist der Fehler bereits aufgetaucht.

### To Reproduce Bug

Pin Mapping mit Battery (rx/tx) sowie Victron (rx-rx3/tx-tx3) hochladen und auswählen. Nach dem klick auf speichern ist die DTU nicht mehr erreichbar..

### Expected Behavior

Benutzung von 3x MPPT sowie Pylontech gleichzeitig.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

04513d3 / generic_esp32s3_usb

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

schlimmchen, 2024-08-12:
Wenn die Hardware bei dir am Schreibtisch liegt und das Problem dort auftritt, dann kannst du bitte mal die seriellen Ausgaben mitschneiden und teilen, oder? Deine Pin-Mapping benutzt auch sicher keinen Pin doppelt?

@swingstate Kommt dir das bekannt vor?

DerHavey1994, 2024-08-12:
ELF file SHA256: 6dba02b8d476a04a

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x42121472
SPIWP:0xee
mode:DOUT, clock div:2
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
SHA-256 comparison failed:
Calculated: ca4adcc736b98868c479048883edac1969cfd9dde0951723205cf1625628f181
Expected: 27accf466d6a3ffbdd2fe4a9dcdf2a85d67a20eb4b1f2dced8b0c8a88771aee2
Attempting to boot anyway...
entry 0x403c98d0
E (448) esp_core_dump_flash: No core dump partition found!
E (448) esp_core_dump_flash: No core dump partition found!

Starting OpenDTU
Initialize FS... done
Reading configuration... done
Reading PinMapping... found valid mapping done
Initialize Network... done
Setting Hostname... Configuring WiFi STA using new credentials... done
Initialize NTP... done
Initialize SunPosition... done
Initialize MqTT... done
Initialize WebApi... done
Initialize Display... done
Initialize LEDs... done
Check for default DTU serial... done
done
Initialize Hoymiles interface... NRF: Connection successful
  Setting radio PA level...
  Setting DTU serial...
  Setting poll interval...
  Setting verbosity...
  Adding inverter: 116184891815 - HM-1500 done
done
[VictronMppt Instance 1] rx = 14, tx = 13
[SerialPortManager] HW UART 0 now in use by 'Victron MPPT 1'
[VictronMppt Instance 2] rx = 12, tx = 11
[SerialPortManager] HW UART 0 already in use by 'Victron MPPT 1'
[SerialPortManager] HW UART 1 now in use by 'Victron MPPT 2'
[VictronMppt Instance 3] rx = 10, tx = 17
[SerialPortManager] HW UART 0 already in use by 'Victron MPPT 1'
[SerialPortManager] HW UART 1 already in use by 'Victron MPPT 2'
[SerialPortManager] HW UART 2 now in use by 'Victron MPPT 3'
Initialize Huawei AC charger interface...
Invalid pin config
[Pylontech] Initialize interface...
[Pylontech] Interface rx = 18, tx = 7

abort() was called at PC 0x4037d5eb on core 1

Backtrace: 0x40377e5e:0x3fcebc30 0x4037d5f5:0x3fcebc50 0x403841b9:0x3fcebc70 0x4037d5eb:0x                    3fcebcf0 0x42077df1:0x3fcebd10 0x42004055:0x3fcebd60 0x4202f78c:0x3fcebdc0 0x42003d01:0x3f                    cebde0 0x42003e4f:0x3fcebe10 0x4204d7ee:0x3fcebe50 0x4207437a:0x3fcebeb0

und im Anhang die Pin_Mapping dazu
[3MPPT_1Pylontech.json](https://github.com/user-attachments/files/16588086/3MPPT_1Pylontech.json)

swingstate, 2024-08-12:
Kommt mir nicht bekannt vor bzw. habe ich dieses Setup so bisher nicht testen können.

schlimmchen, 2024-08-12:
Interessant. Ich schau mir mal an, worauf der Backtrace hindeutet, sobald ich kann. Der in der Beschreibung von dir angegebene Git-Hash ist in der Tat jene Version, die du installiert hast? Das ist wichtig, weil der Backtrace sonst nicht passt (ist eh schon nur "ungefähr", wenn ich die firmware nachbauen muss).

DerHavey1994, 2024-08-12:
Ja ist die Version, hab ich copy & paste aus den Systeminformationen gezogen beim erstellen des Issues.

tobi0171, 2024-08-17:
Kann ich bestätigen, ich habe auch das neue DTU.Pro PcB von Swingstate... mein Setup ist ähnlich. HM1500 Pylontech MPPT150-45 und MPPT100-20.

Sobald ich eine Pinmap mit mehr als einem MPPT auswähle und den Pylontech auswähle stürzt die DTU ab und ich muss Sie neu flashen. Der Fehler tritt jedoch nur auf, wenn ich mehr als 1xMPPT auswähle. 
Das Pinmap an Port 2 oder das Pinmap an Port 3 laufen problemlos. Nur wenn 2 oder 3 MPPTs und der Pylontech ausgewählt werden gibt es Probleme. 

Ich würd mich freuen wenn sich hier etwas tut, das ich auch gerne 2 oder 3 MPPTs verwenden würde. 

Ich nutze allerdings diese Version von oDTUoB https://github.com/helgeerbe/OpenDTU-OnBattery/actions/runs/9750467053 
(Danke Schlimmchen... ich hoffe die tolle Option kann in das künftige Update mit einfließen,sie läuft bei mir gut)

Danke

DerHavey1994, 2024-08-18:
[onBattery Pro 1.7 Mapping 2xMPPT 1xPylontech.json](https://github.com/user-attachments/files/16648220/onBattery.Pro.1.7.Mapping.2xMPPT.1xPylontech.json)
@tobi0171 hier einmal für zwei. Laufen an Anschluss 1 & 2. Läuft bei mir ohne Probleme. Wenn es bei dir nicht laufen sollte, dann flashe dieselbe Build-Run Version, damit muss es dann funktionieren.

tobi0171, 2024-08-18:
@DerHavey1994  Dankeschön, ja das läuft bei mir in der Trockenübung ohne angeschlossene Geräte jetzt auch mit deinem Pinmap. DANKE! 
Ich hatte ein Pinmap (1,7) mit MPPT an Port 2+3 . Dort gab`s Probleme/Absturz. "_OB-Pro - Inverter: HMS/T & HM; BMS: CAN-Bus; MPPT-Ports: 2 & 3".... Wobei das Pinmap bei dem nur Port 2 oder das andere mit nur Port 3 also jeweils einzeln auch funktioniert hatten, nur in kombination 2+3 nicht. 
Das Pinmap von dir ist nun mit Port 1+2, das klappt und scheint zu funktionieren... 
Frage/Feststellung : Problem an Port 3 in Verbindung mit irgendwas?

DerHavey1994, 2024-08-18:
Also bei mir funktionieren auch 3 MPPT, nur halt nicht in Verbindung mit Pylontech.

Bei einer Mapping von @swingstate ist aber auch ein PIN doppelt vergeben (glaube 10 war es). Denke das dein Fehler dadurch kommt.

Ich habe tatsächlich alle Leiterbahnen nochmal per Hand verfolgt, um sicher zustellen, dass alle PINs richtig sind in der Mapping.

schlimmchen, 2024-08-27:
```
0x40377e5e: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
0x4037d5f5: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:137
0x403841b9: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c:46
0x4037d5eb: _esp_error_check_failed at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_err.c:43
0x42077df1: twai_driver_install at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/twai.c:465 (discriminator 1)
0x42004055: BatteryCanReceiver::init(bool, char const*) at /home/beki/Documents/OpenDTU-OnBattery/src/BatteryCanReceiver.cpp:34
0x4202f78c: PylontechCanReceiver::init(bool) at /home/beki/Documents/OpenDTU-OnBattery/src/PylontechCanReceiver.cpp:10
0x42003d01: BatteryClass::updateSettings() at /home/beki/Documents/OpenDTU-OnBattery/src/Battery.cpp:69
0x42003e4f: BatteryClass::init(Scheduler&) at /home/beki/Documents/OpenDTU-OnBattery/src/Battery.cpp:31
0x4204d7ee: setup() at /home/beki/Documents/OpenDTU-OnBattery/src/main.cpp:197
0x4207437a: loopTask(void*) at /home/beki/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```

https://github.com/espressif/esp-idf/blob/a9d0f22193acdf47a5a4db36832ae7068818962b/components/driver/twai.c#L465

Ich fürchte wir sind leider an eine weitere Limitierng gestoßen, nämlich sind nun womöglich keine Interrupts mehr frei: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/intr_alloc.html

Das würde heißen: Egal welches Peripheriegerät du aus dem Pin Mapping raus nimmst, der Rest wird dann funktionieren. Daher würde bei dir, wie auch in #1202 helfen, wenn du das Display deaktivierst, als Beispiel.

DerHavey1994, 2024-08-27:
Limitierung ist natürlich nicht schön..
Aber gut, ist zwar schade, dass das Display nicht funktioniert, da ich es im neuen Gehäuse extra eingebaut habe.. 

Dafür habe ich jetzt aber die Übersicht über all meine Geräte & die Gesamt MPPT Leistung.

Ist ein Kompromiss, mit dem ich leben kann!

Vielen Dank schon mal für die Hilfe :)

AndreasBoehm, 2024-08-27:
@schlimmchen könnte das helfen was skippermeister hier erklärt hat?

https://github.com/helgeerbe/OpenDTU-OnBattery/pull/1144#issuecomment-2294393712

schlimmchen, 2024-08-27:
Ich wollte das "mal schnell" belegen, weil ich gesehen habe, dass es eine Funktion `esp_intr_dump()` gibt, aber die ist noch nicht in der verwendeten version von esp-idf vorhanden, d.h. es würde lästig.

> und im Anhang die Pin_Mapping dazu [3MPPT_1Pylontech.json](https://github.com/user-attachments/files/16588086/3MPPT_1Pylontech.json)

Wenn ich die auf einen jungfräulichen ESP32-S3 lade und sowohl VE.Direct als auch Batterieschnittstelle aktiviere, kommt das Ding hoch.
```
Starting OpenDTU
Initialize FS... done
Reading configuration... done
Reading PinMapping... found valid mapping done
[SPIPortManager] SPI0 and SPI1 reserved by 'Flash and PSRAM'
Initialize Network... done
Setting Hostname... Configuring WiFi STA using new credentials... done
Initialize NTP... done
Initialize SunPosition... done
Initialize MqTT... done
Initialize WebApi... done
Initialize Display... done
Initialize LEDs... done
Check for default DTU serial... done
done
Initialize Hoymiles interface... 
[SPIPortManager] SPI0 already in use by 'Flash'
[SPIPortManager] SPI1 already in use by 'PSRAM'
[SPIPortManager] SPI2 now in use by 'NRF24'
NRF: Connection error!!
  Setting radio PA level... 
  Setting DTU serial... 
  Setting poll interval... 
  Setting verbosity... 
done
[VictronMppt Instance 1] rx = 14, tx = 13
[SerialPortManager] HW UART 0 now in use by 'Victron MPPT 1'
[VictronMppt Instance 2] rx = 12, tx = 11
[SerialPortManager] HW UART 0 already in use by 'Victron MPPT 1'
[SerialPortManager] HW UART 1 now in use by 'Victron MPPT 2'
[VictronMppt Instance 3] rx = 10, tx = 17
[SerialPortManager] HW UART 0 already in use by 'Victron MPPT 1'
[SerialPortManager] HW UART 1 already in use by 'Victron MPPT 2'
[SerialPortManager] HW UART 2 now in use by 'Victron MPPT 3'
Initialize Huawei AC charger interface... 
Invalid pin config
[Pylontech] Initialize interface...
[Pylontech] Interface rx = 18, tx = 7
[Pylontech] Twai driver installed
[Pylontech] Twai driver started
Switch to WiFi mode
Setting Hostname... done
Configuring WiFi STA using new credentials... done
Configuring WiFi STA DHCP IP... done
[DPL::announceStatus] waiting for valid date and time to be available
WiFi connected
WiFi got ip: 192.168.16.136
```
Es ist also auch nicht trivial zu reproduzieren. Vielleicht muss ein NRF24 angeschlossen sein, weil der Treiber erst dann einen Interrupt reserviert, wenn er das Modul entdeckt hat? Wäre schlüssig.

Also hier handelt es sich scheinbar um ein lästiges Problem... Erstmal muss es wirklich verstanden werden. Meine Vermutung finde ich ziemlich gut. Mit ein bisschen Mühe kann man die prüfen. Aber ich möchte jetzt erstmal was anderes voran treiben. 

Bis dahin ist meine Arbeitshypothese, dass die verfügbaren Interrupts ein (weiterer) limitierender Faktor sind, den wir jetzt durch immer mehr Peripherie, die man gleichzeitig anschließen kann (durch SPI und UART Port Manager z.B.), sprengen.

Ach, da fällt mir noch was ein... @skippermeister hat vermutlich schon die passende Lösung in petto: https://github.com/helgeerbe/OpenDTU-OnBattery/pull/1144#issuecomment-2294393712

Also, auf dem Fusion Board kann ich es reproduzieren. Dann geht es vermutlich auch auf dem OnBattery Pro PCB. Hier die pin_mapping.json, die auf dem Fusion das Problem reproduziert: 
[pin_mapping.json](https://github.com/user-attachments/files/16767736/pin_mapping.json)

Mit @skippermeister 's vorgeschlagener Anpassung verschwindet das Problem... Mega! Vielen Dank @skippermeister!

Ich mach einen PR.

schlimmchen, 2024-08-27:
> @schlimmchen könnte das helfen was skippermeister hier erklärt hat?

Ne, jetzt is zu spät, ich hab zwei Minuten bevor du geschrieben hast in während ich meinen Kommentar formuliert habe schon seinen Kommentar verlinkt gehabt, ätsch! :rofl:

PR #1204 (2024-08-28): Fix: Pylontech/Pytes CAN driver: user interrupts at level 2
we are running out of level 1 interrupts on ESP32-S3 boards. for that reason, until this prooves to be another sort of problem, we allocate the TWAI's interrupt for the battery CAN implementation at level 2.

closes #1164. relates to #1202.
closed 2024-08-28 by schlimmchen