hoylabs/OpenDTU-OnBattery #1170: [Request] Add Notifications for Unselected Hardware Profile and Incomplete Setup in LiveView

swingstate, 2024-08-17:
### Is your feature request related to a problem? Please describe.

During recent troubleshooting sessions, we've observed that some users proceed with troubleshooting steps even after uploading a mapping file without selecting a hardware profile. This oversight often leads to repeated configuration attempts (e.g., reconfiguring their battery or inverter), resulting in confusion and wasted time.

### Describe the solution you'd like

To enhance the user experience and reduce these errors, the LiveView should display a clear notification if no hardware profile has been selected. A simple message like "No hardware profile selected" should be shown to prompt the user to take the necessary action.

Additionally, this functionality could be extended to notify users if critical elements of the basic setup are incomplete. For instance, a message like "Basic Setup not finished" could be displayed if essential components, such as the PowerMeter, Inverter, Battery, or mapping profile, have not been defined or activated.

### Describe alternatives you've considered

-

Matti07, 2024-08-25:
@swingstate That's a really good idea! The missing activation cost me a few hours.
I had read about it in the wiki or so, but by the time it was necessary I had already forgotten...

schlimmchen, 2024-12-03:
Completed in 1a0a9887:

[image]

closed 2024-12-03 by schlimmchen

swingstate, 2024-12-04:
abgefahren. 👍🏽