hoylabs/OpenDTU-OnBattery #1178: [Request] Add UpperPowerLimit and TargetPowerConsumption HASS autodiscovery

ButterBetzi, 2024-08-19:
### Is your feature request related to a problem? Please describe.

Those two newly introduced values are available via MQTT, but not auto discovered by HomeAssistant.
Refering to those commits mentioned in the changelog:
https://github.com/helgeerbe/OpenDTU-OnBattery/commit/1f6fdb7fc0be7b493273c6b21422f970cb1d0c3d
https://github.com/helgeerbe/OpenDTU-OnBattery/commit/ccbaf558087d2c481e148b9f9b1bce125d3ba07c

### Describe the solution you'd like

Make those two and any other values to have been added autodioscoverable by HomeAssistant.

ferdlgogosch, 2024-09-12:
Hallo ButterBetzi, ist das Problem mit dem neuen Update vom 11.9. behoben?

ButterBetzi, 2024-09-12:
Hi,
habe ich noch nicht installiert, kann ich also noch nicht sagen

ButterBetzi, 2024-09-28:
Hi,
@ferdlgogosch nein ist es nicht, auch nicht mit dem neusten develop build.
@schlimmchen irgendwelche Updates bzgl. des Themas? :)

ferdlgogosch, 2024-10-25:
Hallo, ich glaube es wurde in der neuen Version (Oktober)noch nicht berücksichtigt.  Habe bezüglich keinen Eintrag in HA gefunden.

schlimmchen, 2024-12-07:
Inspect all subscribed DPL MQTT topics and decide if we can make them HASS auto-discoverable, i.e., all the thresholds. I guess all of them should be available in HASS. Wait... All the thresholds are all available in HASS. So this is really only about UpperPowerLimit and TargetPowerConsumption.

ChaotenKurt, 2025-03-04:
Hallo, ich wollte nur Bescheid geben, dass es bei mir das gleiche ist. Über einen MQTT Explorer kann ich die Werte in MQTT sehen, jedoch werden diese als einzige nicht in Home Assistant angezeigt.

schlimmchen, 2025-04-27:
Completed by #1910, thanks @cmdnymous!

closed 2025-04-27 by schlimmchen