hoylabs/OpenDTU-OnBattery #1181: [Request] DPL Reaktion optional nach Zeit oder Zyklen

Dani1802, 2024-08-21:
### Is your feature request related to a problem? Please describe.

Mein Setup:
Tasmota SmartMeter Reader mit Teleperiod=10 überträgt seine Daten per MQTT zur OpenDTU-OnBattery. Dort regelt der DPL mit angestrebtem Bezug von 50 und Hysterese von 10 den HM-800. An dem Hub hängt ein Zendure Batteriesystem (SolarFlow / Hub2000)

Das funktioniert soweit auch tadellos, jedoch haben wir ein Wasserbett, was zu folgendem Verhalten führt:
Alle 7 Minuten schaltet sich die Heizung des Betts kurz (ca. 10 Sekunden) ein und zieht ca. 250W.

Der DPL bekommt das mit und erhöht den Ausgang des Wechselrichters, doch bis die Leistung ins Hausnetz gelangt ist der Peak wieder vorbei und der Strom geht ins öffentliche Netz.

### Describe the solution you'd like

Mein Wunsch wäre nun folgende optionale Funktion:

Der DPL reagiert erst, wenn das Hysterese Delta nach x Sekunden oder y Aktualisierungen des aktuellen SmartMeter Verbrauchs noch erfüllt ist.

### Additional context

Hier ist das "Problem" schön zu sehen.

[image]

spcqike, 2024-08-21:
alternativ zu Anpassungen an der Firmware kannst du auch probieren, den Tasmota per HTTP(S) Request abzufragen. Dann musst du nicht 10s auf einen neuen Zählerwert warten.
oder
du passt dein Tasmota Skript an, dass es Änderungen schneller sendet, und nicht erst zur nächsten Teleperiod. wie im Wiki beschrieben https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Power-Meter#important-for-tasmota-users

schlimmchen, 2024-08-21:
Du willst #1083, stimmt's. Wenn du zustimmst, dann ist dieses Issue ein Duplikat und es möge dann geschlossen werden.

Dani1802, 2024-08-21:
@spcqike Danke für den Hinweis, habe mein Tasmota Script angepasst und die Werte kommen nun Sekündlich.
Netz Bezug und Netz Einspeisung sind power_curr vom Tasmota und werden je nach Vorzeichen zu Bezug oder Einspeisung. Das Wasserbett wird über einen Plus1PM gemessen.
Für mich sieht der Ablauf so aus:
Bett heizt -> Verbrauch steigt -> DTU regelt WR hoch -> Verbrauch sinkt nach kurzer zeit wieder -> WR speist ein -> DTU regelt WR runter.
Schon klar, dass wir hier über Kleinvieh reden, aber wenn mein Akku nicht über die Nacht hält, dann zählt jede Wh ;-)

Habe nochmal etwas getestet und mit den Abfrageintervallen gespielt:
Ab linker blauer senkrechter Linie:
-- Tasmota: 1s
-- DTU->WR: 10s
Ab mittlerer blauer senkrechter Linie:
-- Tasmota: 1s
-- DTU->WR: 1s
Ab rechter blauer senkrechter Linie:
-- Tasmota: 1s
-- DTU->WR: 30s

Alles in allem keine große Veränderung zu sehen :-/

[image]

@schlimmchen
Sorry, hatte den anderen Issue nicht gesehen -> ja, kann geschlossen werden.
PS: Ich denke das Thema würde sich "relativ leicht" lösen lassen, wenn der DPL erst aktiv wird, wenn die Hysterese länger als X Power Abfragen zutrifft.

spcqike, 2024-08-21:
> Alles in allem keine große Veränderung zu sehen :-/

ich hab powermeter und DPL auf 1s Intervall und auch da sehe ich diese Peaks. klar. die müssen ja auch da sein. 
Wichtig ist doch nur, wie lange diese sind.

bei 10s teleperiod und 10s DPL Intervall kannst du halt maximal 10s zu viel einspeisen. könntest aber auch im worst case die Produktion erst nach 10s Netzbezug starten.

bei tasmota instant (er sendet ja sofort, nicht erst nach 1s) und 1s DPL Interval, kann der DPL halt "schnellstmöglich" reagieren. also nach max. 1s hochregeln und nach max 1s runter regeln. 

Wie lang sind denn die Einspeisepeaks? Ich vermute, du siehst dort keine großen Unterschiede, da du nur 10s Werte sampelst und darstellst. beobachte mal den Tasmota im Live-View (oder heb die Powermeter WErte der DTU auf, die sie ja sofort per MQTT empfängt) und gucke, ob du im <10s Bereich unterschiede findest.

Dani1802, 2024-08-21:
Das ist ein Screenshot aus HomeAssistant mit einer Sampling Rate von 1s bei den Netz Werten und 5s beim WR.

[image]

Mit DPL Intervall meinst Du schon das Abfrageintervall unter Einstellungen -> DTU, oder gibts für den DPL ein extra Intervall?

Am liebsten wäre es mir, wenn die DTU den Peak überlesen/ignorieren würde, denn so "kostet" mich das Bett aktuell das doppelte an Strom. Den ersten Peak kaufe ich, den zweiten verschenke ich.

Manos1966, 2024-08-21:
@Dani1802 "Tasmota SmartMeter Reader mit Teleperiod=10" ist nicht gut, deswegen aber gibt es
 https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Power-Meter#important-for-tasmota-users

spcqike, 2024-08-21:
> Mit DPL Intervall meinst Du schon das Abfrageintervall unter Einstellungen -> DTU

ja richtig.

> denn so "kostet" mich das Bett aktuell das doppelte an Strom.

wie meinst du das? aktuell "kostet" es dich laut diesem Bild ~10s im Einkauf, dann ~25s in der Eigenerzeugung und ~10s im Überschuss verschenken. würde die DTU das nicht ausregeln, hättest du halt 25s Einkauf.

Das Einzige, was ich versuchen würde zu ändern, ist, die Abfallzeit zu beschleunigen. Scheinbar reagiert der WR erst nach ~10s. Zumindest hast du ~10s den Export, der ja erst kommt, wenn die Last weg ist. Das ist bei mir nicht so.

Ich protokolliere meinen WR und den Shelly im 5s Takt und sehe auch höchstens ein Abtastintervall als aktive Zeit
[image]

Auch wenn der WR in Punkt 2 erst nach 2 Punkten, also 10s, mehr meldet, liefert er es doch schon nach den ersten 5s. zu erkennen an der orangen Linie, die eben nur für einen Messpunkt "steht". beim Abfallen das gleiche. die Last (blau) fällt weg, der Netzbezug (orange) fällt in den Keller, der Messwert 5s später zeigt bereits, dass der WR ausgeregelt hat. weitere 5s später meldet auch die openDTU die neuen Leistungen vom WR.

In Wirklichkeit hat der WR viel schneller ausgeregelt, nur taste ich dies nicht schnell genug ab. Es kann gut und gerne sein, dass der echte Netzbezug bereits nach 1 oder 2s ausgeglichen ist. das sehe ich in der Aufzeichnung nur nicht, weil ich nur die Momentanleistungen aller 5s betrachte.

Beobachte deinen Tasmota doch einfach mal in live und gucke, ob du wirklich für ~10s einen Export stehen hast. mich würde es wundern, wenn es so wäre. vorausgesetzt, du bekommst sekündliche aktuelle Werte vom Powermeter und hast nicht unerwartet viele Paket Verluste zum WR.

Dani1802, 2024-08-24:
Nicht mehr relevant

closed 2024-08-24 by Dani1802