hoylabs/OpenDTU-OnBattery #1187: Can/ISo Pylontech

rudolfp1979, 2024-08-22:
### What happened?

Nur schonmal im Voraus im neuesten Development schein ein Bug zu sein wenn ich Ihn ein spiele weil er überall sagt es gibt eine neue version dann erkennt er den Pylonen nicht mehr am AllianceFusion Board. Zurück auf der 18.8 wird er sofort wieder erkannt nur falls dir das weiter hilft.

### To Reproduce Bug

neueste Version installiert development branch

### Expected Behavior

sollte funktionieren^^

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

df53f34

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

schlimmchen, 2024-08-23:
> dann erkennt er den Pylonen nicht mehr am AllianceFusion Board

Ist hier eine Pylontech Batterie gemeint?

Du hast also ein Fusion Board. Was hängt denn da dran, bzw. wie ist denn die Pylontech Batterie angeschlossen?

rudolfp1979, 2024-08-23:
Genau eine pylontech Batterie über das can iso shield zusammen mit einem victron 150/45 am tx1/rx1

schlimmchen, 2024-08-23:
Dass das im letzten Release funktionieren soll aber im development branch nicht kann ich mir nicht so recht erklären, die Änderungen sind schließlich überschaubar.

Hast du die Firmware selbst kompiliert oder hast du einen Build aus einem der Github Action Runs genommen?

rudolfp1979, 2024-08-23:
Wurde selbstkompiliert für einen esp32s3 ging aber fehlerfrei durch

schlimmchen, 2024-08-23:
Aha. Was ist denn das Symptom? Wie stellst du denn fest, dass die Pylontech nicht mehr "geht"? Weil in der WebApp etwas fehlt? Hast du die WebApp gebaut? Siehe [Anleitung](https://opendtu-onbattery.net/firmware/compile_vscode/) Schritt 7.

rudolfp1979, 2024-08-26:
Du hattest Recht bei der WebApp lief was schief jetzt läuft es wieder

closed 2024-08-26 by rudolfp1979