hoylabs/OpenDTU-OnBattery #1188: PYTES CAN protocol support doesn't work with E-BOX-4850P-C

ranma, 2024-08-23:
### What happened?

Problem: When using the new PYTES CAN protocol support, the battery status doesn't update. The battery works fine in PYLONTECH mode.

Expectation: PYTES CAN protocol support works with Pytes E-BOX-4850P-C

After upgrading OpenDTU to version`2024.08.18`, I tried switching to Pytes protocol by:

1. Switching OpenDTU from "Pylontech via CAN bus" to "Pytes via CAN bus"
2. Then switching the battery to Pytes protocol using
```
login config
setprt PYTES
logout
```

Observed behaviour:
Battery status continued to update after switching OpenDTU (with additional Pytes fields not present since the messages sent were still Pylontech messages), then after switching the Battery itself I didn't see any further updates in the Live view (but CAN messages can be see in the log).

I also tried restarting both OpenDTU and the Battery, but this didn't help.

For now I've switched back to Pylontech mode.

```
Hostname	OpenDTU-3480B8
SDK-Version	v4.4.7-dirty
Konfigurationsversion	0.1.28
Firmwareversion / git Hash	[2024.08.18](https://github.com/helgeerbe/OpenDTU-OnBattery/releases/tag/2024.08.18)
Firmware Branch	master
PIO Umgebung	generic_esp32s3_usb
Firmware-Aktualisierung	
Reset Grund CPU 0	Software reset CPU
Reset Grund CPU 1	Software reset CPU
Anzahl der Konfigurationsspeicherungen	53
Betriebszeit	0 Tage 00:17:00
```

```
PYTES_config>info
@
Device address      : 1
Manufacturer        : PYTES
Device name         : E-BOX-4850P-C
Board version       : SQBMSV15
Main Soft version   : SPBMS15SPH2203V1.5.33.C8
Iterate version     : SPBMS15SPH2203V1.5.33.C8.T4
Soft  version       : V1.5
Boot  version       : V1.14
Comm version        : V2.0
Release Date        : 23-12-15
Barcode             : LB0A010002450068
Board ID            : GD33353155000F36373436B39D
Board SRAM Size     : 96KB
Board FLASH Size    : 256KB
Specification       : 48V/50AH
Cell Number         : 15
Max Dischg Curr     : -60000mA
Max Charge Curr     : 52000mA
Console Port rate   : 115200
Command completed successfully
```

CAN messages from log after `setprt PYTES`:
```
09:57:49.729 > [Pytes] Received CAN message: 0x0400 - 14 02 FA 00 7D 00 BD 01
09:57:49.933 > [Pytes] Received CAN message: 0x0401 - C5 0C BB 0C 07 00 04 00
09:57:50.053 > [Pytes] Received CAN message: 0x0402 - E6 00 DC 00 00 00 0A 00
09:57:50.060 > [Pytes] Received CAN message: 0x0403 - 04 84 80 02 04 04 80 00
09:57:50.070 > [Pytes] Received CAN message: 0x0404 - 12 00 64 00 01 00 1C 00
09:57:50.075 > [Pytes] Received CAN message: 0x0405 - 1F 13 00 00 E2 00 0F 00
09:57:50.082 > [Pytes] Received CAN message: 0x0406 - 00 00 00 00 01 00 00 00
09:57:50.090 > [Pytes] Received CAN message: 0x0407 - 3B 1B 7F 00 3B 1B 00 00
09:57:50.095 > [Pytes] Received CAN message: 0x0408 - 01 01 00 00 00 00 00
09:57:50.105 > [Pytes] Received CAN message: 0x0409 - 1E C3 00 00 3E 22 00 00
09:57:50.111 > [Pytes] Received CAN message: 0x040A - 50 59 54 45 53
09:57:50.115 > [Pytes] Received CAN message: 0x040B - 18 08 17 09 39 2C 01 00
09:57:50.124 > [Pytes] Received CAN message: 0x040C - 01 00 00 00 01 00 00 00
09:57:50.129 > [Pytes] Received CAN message: 0x040D - 00 00 00 00 00 00 00 00
09:57:50.133 > [Pytes] Received CAN message: 0x041E - A6 02 00 00 B2 02 00 00
```

CAN messages from log after `setprt PYLON`:
```
09:58:48.629 > [Pytes] Received CAN message: 0x0359 - 00 00 00 00 01 50 59
09:58:48.642 > [Pytes] Received CAN message: 0x0351 - 14 02 FA 00 7D 00 BD 01
09:58:48.647 > [Pytes] chargeVoltageLimit: 53.200001 chargeCurrentLimit: 25.000000 dischargeCurrentLimit: 12.500000 dischargeVoltageLimit: 44.500000
09:58:48.657 > [Pytes] Received CAN message: 0x0355 - 12 00 64 00
09:58:48.662 > [Pytes] soc: 18 soh: 100
09:58:48.673 > [Pytes] Received CAN message: 0x0356 - 1F 13 00 00 E2 00
09:58:48.678 > [Pytes] voltage: 48.950001 current: 0.000000 temperature: 22.600000
09:58:48.687 > [Pytes] Received CAN message: 0x035C - C0 00
09:58:48.701 > [Pytes] Received CAN message: 0x035E - 50 59 54 45 53
09:58:48.707 > [Pytes] Manufacturer: PYTES
```

### To Reproduce Bug

Attempt to use a Pytes E-BOX-4850P-C in Pytes protocol mode.

### Expected Behavior

PYTES CAN protocol support works with Pytes E-BOX-4850P-C

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.08.18

### Relevant log/trace output

```Shell
09:57:49.729 > [Pytes] Received CAN message: 0x0400 - 14 02 FA 00 7D 00 BD 01
09:57:49.933 > [Pytes] Received CAN message: 0x0401 - C5 0C BB 0C 07 00 04 00
09:57:50.053 > [Pytes] Received CAN message: 0x0402 - E6 00 DC 00 00 00 0A 00
09:57:50.060 > [Pytes] Received CAN message: 0x0403 - 04 84 80 02 04 04 80 00
09:57:50.070 > [Pytes] Received CAN message: 0x0404 - 12 00 64 00 01 00 1C 00
09:57:50.075 > [Pytes] Received CAN message: 0x0405 - 1F 13 00 00 E2 00 0F 00
09:57:50.082 > [Pytes] Received CAN message: 0x0406 - 00 00 00 00 01 00 00 00
09:57:50.090 > [Pytes] Received CAN message: 0x0407 - 3B 1B 7F 00 3B 1B 00 00
09:57:50.095 > [Pytes] Received CAN message: 0x0408 - 01 01 00 00 00 00 00
09:57:50.105 > [Pytes] Received CAN message: 0x0409 - 1E C3 00 00 3E 22 00 00
09:57:50.111 > [Pytes] Received CAN message: 0x040A - 50 59 54 45 53
09:57:50.115 > [Pytes] Received CAN message: 0x040B - 18 08 17 09 39 2C 01 00
09:57:50.124 > [Pytes] Received CAN message: 0x040C - 01 00 00 00 01 00 00 00
09:57:50.129 > [Pytes] Received CAN message: 0x040D - 00 00 00 00 00 00 00 00
09:57:50.133 > [Pytes] Received CAN message: 0x041E - A6 02 00 00 B2 02 00 00
```
```

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

ranma, 2024-08-23:
Hmm, looking at my old traces from https://github.com/helgeerbe/OpenDTU-OnBattery/pull/1088#issuecomment-2222142161, I need to check why these have different CAN message ids now, maybe the `setprt PYTES` didn't actually work correctly.

ranma, 2024-08-23:
Though I think the command should be correct:
```
PYTES_config>setprt PYLONT
@

 Support Protocol : VICTRON   GROWATT   GOODWE    MEGAREVO  SRNE      DEYE      
                    VOLTRONIC LUXPOWER  STUDER    KEHUA     AFORE     SCHNEIDER 
                    SOLAX     SOLIS     SMA       PYTES     MUST      ASW       
                    INGETEAM  SOLARK    SENERGY   LIVOLTEK  NEP     
```

shows that actually `setprt PYLON` doesn't seem to be offically supported in the config interface help text (but does work for me).

ranma, 2024-08-23:
So it looks like the protocol implemented in OpenDTU is actually the Victron protocol, not the Pytes protocol.

It now works after configuring the protocol to Victron using:
```
login config
setprt VICTRON
logout
```

with this the received messages now look like this:
```
12:19:52.727 > [Pytes] Received CAN message: 0x0351 - 14 02 FA 00 FA 00 BD 01
12:19:52.762 > [Pytes] chargeVoltageLimit: 53.200001 chargeCurrentLimit: 25.000000 dischargeCurrentLimit: 25.000000 dischargeVoltageLimit: 44.500000
12:19:52.773 > [Pytes] Received CAN message: 0x0355 - 19 00 64 00
12:19:52.782 > [Pytes] soc: 25 soh: 100
12:19:52.792 > [Pytes] Received CAN message: 0x0356 - 86 13 65 00 EC 00
12:19:52.805 > [Pytes] voltage: 49.980000 current: 10.100000 temperature: 23.600000
12:19:52.815 > [Pytes] Received CAN message: 0x035A - 00 00 00 00 00 00 00 00
12:19:52.824 > [Pytes] Alarms: 0 0 0 0 0 0 0 0 0 0
12:19:52.834 > [Pytes] Warnings: 0 0 0 0 0 0 0 0 0 0
12:19:52.842 > [Pytes] Received CAN message: 0x035E - 50 59 54 45 53
12:19:52.850 > [Pytes] Manufacturer: PYTES
12:19:52.861 > [Pytes] Received CAN message: 0x035F - 01 00 69 00 0C 00
12:19:52.869 > [Pytes] fwversion: v105.0 availableCapacity: 12 Ah
12:19:52.878 > [Pytes] Received CAN message: 0x0360 - 00
12:19:52.887 > [Pytes] Received CAN message: 0x0372 - 01 00 00 00 00 00 00 00
12:19:52.896 > [Pytes] moduleCountOnline: 1 moduleCountBlockingCharge: 0 moduleCountBlockingDischarge: 0 moduleCountOffline: 0
12:19:52.907 > [Pytes] Received CAN message: 0x0373 - 01 0D 08 0D 28 01 29 01
12:19:52.917 > [Pytes] lowestCellMilliVolt: 3329 highestCellMilliVolt: 3336 minimumCellTemperature: 23.000000 maximumCellTemperature: 24.000000
12:19:52.926 > [Pytes] Received CAN message: 0x0374 - 30 31 30 30 00 00 00 00
12:19:52.935 > [Pytes] cellMinVoltageName: 0100
12:19:52.946 > [Pytes] Received CAN message: 0x0375 - 30 35 30 30 00 00 00 00
12:19:52.952 > [Pytes] cellMaxVoltageName: 0500
12:19:52.962 > [Pytes] Received CAN message: 0x0376 - 31 30 30 30 00 00 00 00
12:19:52.969 > [Pytes] cellMinTemperatureName: 1000
12:19:52.978 > [Pytes] Received CAN message: 0x0377 - 30 30 30 30 00 00 00 00
12:19:52.987 > [Pytes] cellMaxTemperatureName: 0000
12:19:52.995 > [Pytes] Received CAN message: 0x0378 - A8 02 00 00 B2 02 00 00
12:19:53.004 > [Pytes] chargedEnergy: 68.000000 dischargedEnergy: 69.000000
12:19:53.013 > [Pytes] Received CAN message: 0x0379 - 31 00
12:19:53.021 > [Pytes] totalCapacity: 49 Ah
12:19:53.028 > [Pytes] Received CAN message: 0x0380 - 4C 42 30 41 30 31 30 30
12:19:53.039 > [Pytes] snPart1: LB0A0100
12:19:53.047 > [Pytes] Received CAN message: 0x0381 - 30 32 34 35 30 30 36 38
12:19:53.058 > [Pytes] snPart2: 02450068
```

schlimmchen, 2024-08-23:
@AndreasBoehm WTF?

ranma, 2024-08-23:
I've updated the wiki document at https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Pytes-Battery accordingly

ranma, 2024-08-23:
> @AndreasBoehm WTF?

Presumably this is the default protocol the Battery is often configured to when it is delivered. Mine apparently came pre-configured for Victron protocol as well, at least according to my CAN traces that I did before switching it to PYLONtech mode.

AndreasBoehm, 2024-08-23:
That is very interesting, i wasn't aware that there is actually a `PYTES` CAN protocol variant as i do not own the Pytes Console Cable. As this should also be usable for people that don't want to deep dive i keep my battery at default settings.

What we should add to the wiki is the setting of the DIP Switches instead of describing how to play with the console here. It is not needed to adjust anything on the battery to make this work.

> DIP Switches should be in their default position (Victron setting) and nothing should be adjusted in the console.

[image]

ranma, 2024-08-23:
> That is very interesting, i wasn't aware that there is actually a `PYTES` CAN protocol variant as i do not own the Pytes Console Cable.

FWIW the Pytes console Cabel is Cisco-compatible, I bought a cheap Cisco to USB cable on digitec.ch and it works for me to access to Battery console.

AndreasBoehm, 2024-08-23:
I would be very interested in a full log of all possible can messages in the PYTES mode, maybe there is more data available than what we can show right now :)

AndreasBoehm, 2024-08-23:
> FWIW the Pytes console Cable is Cisco-compatible, I bought a cheap Cisco to USB cable on digitec.ch and it works for me to access to Battery console.

Like this?
https://www.ebay.de/itm/186361655539?_nkw=Cisco+USB&itmmeta=01J5ZDK5J9S6AS2NZVQZNJX4HC&hash=item2b640530f3:g:mtcAAOSwOTZmAUjn&itmprp=enc%3AAQAJAAAA4HoV3kP08IDx%2BKZ9MfhVJKmqUu1EweW8lX7y6cBqa2WSBz2daBiEi2Cr3zvX%2FdI7qalbFHA5PluQYofVsdTVsQ%2Fm7nADpf3cOQCGCpn1ikly%2BzPoDgFEUrR9MQcMThuY8NjkzC%2FItpW59iKgSZ8kspSiU3tsaLg8uespf1vwKdhuJ8TOyM1t7K%2FQ35vEUGi%2FoiCntJO2w4i453tY0%2B8LTT6JoDstnz4%2FJrXIETv%2FCP8dOrB76nqSd%2FaBFV6BZuQRBSxZgX8ZRDh8LSq5IwprZPqaWgaDUKI0j2o%2BF6V3M%2BR2%7Ctkp%3ABk9SR6TZzO2vZA

ranma, 2024-08-23:
> Like this? [...]

Yes, that looks very similar to the one I have: https://www.digitec.ch/de/s1/product/delock-usb-zu-rj-45-konsolenkabel-3m-3-m-netzwerkkabel-10119762?supplier=406802

ranma, 2024-08-23:
> I would be very interested in a full log of all possible can messages in the PYTES mode, maybe there is more data available than what we can show right now :)

This should be complete, those get repeated regularly:
```
09:57:49.729 > [Pytes] Received CAN message: 0x0400 - 14 02 FA 00 7D 00 BD 01
09:57:49.933 > [Pytes] Received CAN message: 0x0401 - C5 0C BB 0C 07 00 04 00
09:57:50.053 > [Pytes] Received CAN message: 0x0402 - E6 00 DC 00 00 00 0A 00
09:57:50.060 > [Pytes] Received CAN message: 0x0403 - 04 84 80 02 04 04 80 00
09:57:50.070 > [Pytes] Received CAN message: 0x0404 - 12 00 64 00 01 00 1C 00
09:57:50.075 > [Pytes] Received CAN message: 0x0405 - 1F 13 00 00 E2 00 0F 00
09:57:50.082 > [Pytes] Received CAN message: 0x0406 - 00 00 00 00 01 00 00 00
09:57:50.090 > [Pytes] Received CAN message: 0x0407 - 3B 1B 7F 00 3B 1B 00 00
09:57:50.095 > [Pytes] Received CAN message: 0x0408 - 01 01 00 00 00 00 00
09:57:50.105 > [Pytes] Received CAN message: 0x0409 - 1E C3 00 00 3E 22 00 00
09:57:50.111 > [Pytes] Received CAN message: 0x040A - 50 59 54 45 53
09:57:50.115 > [Pytes] Received CAN message: 0x040B - 18 08 17 09 39 2C 01 00
09:57:50.124 > [Pytes] Received CAN message: 0x040C - 01 00 00 00 01 00 00 00
09:57:50.129 > [Pytes] Received CAN message: 0x040D - 00 00 00 00 00 00 00 00
09:57:50.133 > [Pytes] Received CAN message: 0x041E - A6 02 00 00 B2 02 00 00
```

At least some of these are clearly the same as in Victron mode, just with different IDs (haven't checked in detail).

(Note that in general my access to this battery is limited as it is installed at my moms place and I haven't set up remote access to the Battery console)

ranma, 2024-08-23:
> What we should add to the wiki is the setting of the DIP Switches instead of describing how to play with the console here. It is not needed to adjust anything on the battery to make this work, no `PYLON` or `PYTES` setting needed, the default will work fine.

SGTM

schlimmchen, 2024-08-23:
When is the DIP switch setting used? Does the console command override the DIP switch setting? How can a user make sure that the DIP switch defined the protocol in use?

This sounds to me that the feature is named improperly. What Andreas implemented is not a protocol (or register set etc.) specific to PYTES, but specific to how batteries (of any manufacturer) behave, that are supposed to run in a Victron ecosystem. I feel that this expands the scope of the feature in its current state, but also calls for improvement in the wording and docs?

AndreasBoehm, 2024-08-23:
> When is the DIP switch setting used? Does the console command override the DIP switch setting? How can a user make sure that the DIP switch defined the protocol in use?

Maybe @ranma can help us here?
I did order a console cable today and could try the combination of DIP switch settings and protocol config via console probably next week. 

> This sounds to me that the feature is named improperly. What Andreas implemented is not a protocol (or register set etc.) specific to PYTES, but specific to how batteries (of any manufacturer) [...]

I partially agree with you, its weird what pytes is doing here. My first thought was that the pytes Victron protocol is here to make pytes compatible with Victron by replicating the protocol used by pylontech, but the protocol is only partially similar.
If the protocol is based on the Inverter it doesn't make any sense for me that pytes and pylontech are sending different data.
From a quick google search i could not find any option to adjust which protocol the pylontech batteries use.

ranma, 2024-08-23:
> When is the DIP switch setting used? Does the console command override the DIP switch setting? How can a user make sure that the DIP switch defined the protocol in use?

That's an excellent question, I wanted to ask myself :)

I would assume that the DIP switch setting is applied when the DIP switches are changed, possibly on the next reboot?

Console changes take effect immediately and override the DIP switch setting, these do survive a reboot of the Battery.

> This sounds to me that the feature is named improperly. What Andreas implemented is not a protocol (or register set etc.) specific to PYTES, but specific to how batteries (of any manufacturer) [...]

I think the Pytes battery is particularly flexible in allow to override the protocol to use the one of various manufacturer ecosystems.

Victron is themselves very flexible and the Victron GX gateway may be auto-detecting the battery protocol based on the messages it sees?

https://www.victronenergy.com/live/battery_compatibility:pylontech_phantom

Victrons own CAN protocol seems to be documented at https://www.victronenergy.com/upload/documents/VE.Can-registers-public.pdf

In particular this seems to be the protocol of Victrons own VictronEnergy Lynx Ion BMS.

ranma, 2024-08-23:
> Maybe @ranma can help us here?

I played a bit with the DIP switches on my battery, but it is unclear under which conditions the protocol is actually updated based on the DIP switch state.

I can see that on boot-up the battery logs the current DIP switch state, eg.:
```
 Timer Test Passed
 Addr Init
 DeviceAddr 1
 DIP 0
 Group Address 1
```

or
```
 Timer Test Passed
 Addr Init
 DeviceAddr 1
 DIP 4
 Group Address 1
```

But this didn't seem to modify the protocol in my tests. Maybe there are some extra conditions like "The CAN bus messages are not getting ACKs for the in-use protocol".

If someone wants to poke into the Pytes firmware (STM32-like), I have a dump.

schlimmchen, 2024-08-23:
Interesting. That's something we should figure out and add to the docs (opendtu-onbattery.net needs a respective doc for connecting a battery using CAN and in particular for the PYTES batteries as well).

I am curious whether this also means that a new-style JK Inverter BMS could be connected to OpenDTU-OnBattery (or a CAN bus in general) using CAN if the JK inverter BMS is setup to use the victron protocol.

ranma, 2024-08-23:
> I would be very interested in a full log of all possible can messages in the PYTES mode, maybe there is more data available than what we can show right now :)

`0x0400 - 14 02 FA 00 7D 00 BD 01`

532 250 125 445

53.2V is the charge end voltage
44.5V is the discharge end voltage
25.0A is the current charge current limit  (0.5C)
12.5A is the current discharge current limit (0.25C)

`0x0401 - C5 0C BB 0C 07 00 04 00`

3269 3259 7 4

Voltage and cell number for highest/lowest voltage.

`0x0402 - E6 00 DC 00 00 00 0A 00`

230 220 0 10

Temperature and cell number for highest/lowest temperature.

`0x0403 - 04 84 80 02 04 04 80 00`

No idea

`0x0404 - 12 00 64 00 01 00 1C 00`

18 100 1 28
18 is state of charge, 100 might be "state of health" (not aged yet), 28 might be cycle count

`0x0405 - 1F 13 00 00 E2 00 0F 00`

4905 226 15

Probably current voltage, current and maybe state of charge in percent

`0x0406 - 00 00 00 00 01 00 00 00`

No idea

`0x0407 - 3B 1B 7F 00 3B 1B 00 00`

No idea

`0x0408 - 01 01 00 00 00 00 00`

No idea

`0x0409 - 1E C3 00 00 3E 22 00 00`

49950 and 8766
Full capacity and current state of charge in mAh?
(State of charge 17.55% matches presumed SOC in message 0x0404) 

`0x040A - 50 59 54 45 53` MFG name "PYTES"

`0x040B - 18 08 17 09 39 2C 01 00`

2072 2327 11321 1
No idea

`0x040C - 01 00 00 00 01 00 00 00`
No idea

`0x040D - 00 00 00 00 00 00 00 00`

No idea

`0x041E - A6 02 00 00 B2 02 00 00`

678 and 690
Charged and discharged energy in kWh

ranma, 2024-08-24:
With partial implementation:
```
15:07:53.477 > [Pytes] Received CAN message: 0x0400 - 14 02 00 00 FA 00 BD 01
15:07:53.482 > [Pytes] chargeVoltageLimit: 53.200001 chargeCurrentLimit: 0.000000 dischargeCurrentLimit: 25.000000 dischargeVoltageLimit: 44.500000
15:07:53.516 > [Pytes] Received CAN message: 0x0401 - 21 0E 84 0D 05 00 09 00
15:07:53.551 > [Pytes] lowestCellMilliVolt: 3460 highestCellMilliVolt: 3617 cellMinVoltageName: 0009 cellMaxVoltageName: 0005
15:07:53.552 > [Pytes] Received CAN message: 0x0402 - 22 01 22 01 00 00 00 00
15:07:53.553 > [Pytes] minimumCellTemperature: 29.000000 maximumCellTemperature: 29.000000 cellMinTemperatureName: 0000 cellMaxTemperatureName: 0000
15:07:53.555 > [Pytes] Received CAN message: 0x0403 - 04 24 80 04 04 04 80 00
15:07:53.556 > [Pytes] Received CAN message: 0x0404 - 64 00 64 00 01 00 1D 00
15:07:53.590 > [Pytes] soc: 100 soh: 100
15:07:53.592 > [Pytes] Received CAN message: 0x0405 - C3 14 3D 00 22 01 0F 00
15:07:53.593 > [Pytes] voltage: 53.149998 current: 6.100000 temperature: 29.000000
15:07:53.595 > [Pytes] Received CAN message: 0x0406 - 00 00 00 00 01 00 00 00
15:07:53.596 > [Pytes] Received CAN message: 0x0407 - 3B 1B 7F 00 3B 1B 00 00
15:07:53.597 > [Pytes] Received CAN message: 0x0408 - 00 01 00 00 00 01 00
15:07:53.598 > [Pytes] Received CAN message: 0x0409 - 1E C3 00 00 1E C3 00 00
15:07:53.599 > [Pytes] soc: 100.000000 totalCapacity: 49.950001 Ah availableCapacity: 49.950001 Ah 
15:07:53.646 > [Pytes] Received CAN message: 0x040A - 50 59 54 45 53
15:07:53.648 > [Pytes] Manufacturer: PYTES
15:07:53.650 > [Pytes] Received CAN message: 0x040B - 18 08 18 0F 07 30 01 00
15:07:53.652 > [Pytes] Received CAN message: 0x040C - 01 00 00 00 01 00 00 00
15:07:53.653 > [Pytes] Received CAN message: 0x040D - 00 00 00 00 FF 7D 00 00
15:07:53.654 > [Pytes] Received CAN message: 0x041E - C3 02 00 00 BA 02 00 00
15:07:53.659 > [Pytes] chargedEnergy: 70.700005 dischargedEnergy: 69.800003
```

ranma, 2024-08-24:
WIP: [pytes-protocol+discharge-limiter.patch.txt](https://github.com/user-attachments/files/16737202/pytes-protocol%2Bdischarge-limiter.patch.txt)

ranma, 2024-09-03:
DIP switch to protocol type mapping:
```
int DIP_config_FUN_08024bfc(int dip_switch_value)

{
                    /* Only settable via `setprt` console command in FW V1.5: SOLAX, PYLON */
  switch(dip_switch_value) {
  case 0:
                    /* VICTRON */
    PTR_struct extcan_config_08024cd8->protocol_type = 8;
    break;
  case 1:
                    /* VOLTRONIC */
    PTR_struct extcan_config_08024cd8->protocol_type = 0xe;
    break;
  case 2:
                    /* DEYE */
    PTR_struct extcan_config_08024cd8->protocol_type = 0xd;
    break;
  case 3:
                    /* GOODWE */
    PTR_struct extcan_config_08024cd8->protocol_type = 10;
    break;
  case 4:
                    /* GROWATT */
    PTR_struct extcan_config_08024cd8->protocol_type = 9;
    break;
  case 5:
                    /* SRNE */
    PTR_struct extcan_config_08024cd8->protocol_type = 0xc;
    break;
  case 6:
                    /* STUDER */
    PTR_struct extcan_config_08024cd8->protocol_type = 0x11;
    break;
  case 7:
                    /* MUST */
    PTR_struct extcan_config_08024cd8->protocol_type = 0x19;
    break;
  case 8:
                    /* AFORE */
    PTR_struct extcan_config_08024cd8->protocol_type = 0x13;
    break;
  case 9:
                    /* SCHNEIDER */
    PTR_struct extcan_config_08024cd8->protocol_type = 0x14;
    break;
  case 10:
                    /* SOLIS */
    PTR_struct extcan_config_08024cd8->protocol_type = 0x16;
    break;
  case 0xb:
                    /* SMA */
    PTR_struct extcan_config_08024cd8->protocol_type = 0x17;
    break;
  case 0xc:
                    /* LUXPOWER */
    PTR_struct extcan_config_08024cd8->protocol_type = 0xf;
    break;
  case 0xd:
                    /* ASW */
    PTR_struct extcan_config_08024cd8->protocol_type = 0x1a;
    break;
  case 0xe:
                    /* KEHUA */
    PTR_struct extcan_config_08024cd8->protocol_type = 0x12;
    break;
  case 0xf:
                    /* MEGAREVO */
    PTR_struct extcan_config_08024cd8->protocol_type = 0xb;
    break;
  case 0x10:
                    /* INGETEAM */
    PTR_struct extcan_config_08024cd8->protocol_type = 0x1b;
    break;
  case 0x11:
                    /* SOLARK */
    PTR_struct extcan_config_08024cd8->protocol_type = 0x1c;
    break;
  case 0x12:
                    /* SENERGY */
    PTR_struct extcan_config_08024cd8->protocol_type = 0x1d;
    break;
  case 0x13:
                    /* LIVOLTEK */
    PTR_struct extcan_config_08024cd8->protocol_type = 0x1e;
    break;
  case 0x14:
                    /* NEP */
    PTR_struct extcan_config_08024cd8->protocol_type = 0x1f;
    break;
  default:
                    /* PYTES */
    PTR_struct extcan_config_08024cd8->protocol_type = 0x18;
  }
  printf_or_puts(s__DIP_%d_08024cdc,dip_switch_value);
  return 0;
}
```

ranma, 2024-09-03:
Also interesting in the calling function:
```
  if (dip_switch_value == 0x3f) {
    printf_or_puts(s__Enter_engineering_mode_08024bcc);
    *DAT_08024b5c = 0;
  }
```

AndreasBoehm, 2024-09-16:
Have you been able to find out if there is a way to override the protocol using dip-switches after you set it via the console?
Would love to be able to get my battery back to "stock" after playing with the console.

ranma, 2024-09-20:
> Have you been able to find out if there is a way to override the protocol using dip-switches after you set it via the console? Would love to be able to get my battery back to "stock" after playing with the console.

So I found that it sets it from the switches before it reads the eeprom configuration, but I need to do some more reverse-engineering to find where exactly the console configuration overrides the dip-switch setting.

ranma, 2024-09-20:
Try `setprt DIP`.

ranma, 2024-09-20:
BTW can you check which DIP switch combination logs "DIP 30" or "DIP 1e" on startup? That should set PYTES protocol via DIP switches. Presumably that's "down up up up up down". (Can't test myself currently since I don't have remote access to the console)

AndreasBoehm, 2024-10-23:
> Try `setprt DIP`.

```
PYTES_config>setprt DIP
@
Command completed successfully
$$
PYTES_config>setprt
@
 CAN Comm protocol : PYTES
 CAN Group address : 1
 RS485 Comm protocol : PYTES
 RS485 Group address : 1
Command completed successfully
$$
```

This command does not give an error but the protocol is not changed. I will try with a restart of my battery.

AndreasBoehm, 2024-10-25:
> ```
> PYTES_config>setprt DIP
> @
> Command completed successfully
> $$
> PYTES_config>setprt
> @
>  CAN Comm protocol : PYTES
>  CAN Group address : 1
>  RS485 Comm protocol : PYTES
>  RS485 Group address : 1
> Command [...]

Restarting my battery made it switch the protocol to whatever is set via the DIP switches.

I can also confirm that a DIP switch setting of "down up up up up down" switches the battery to PYTES protocol without doing anything on the console 🚀 

Could you add this to the wiki/docs @ranma ?

Thanks a lot for everything!

closed 2024-10-25 by AndreasBoehm

AndreasBoehm, 2024-11-21:
Hey @ranma, now that its getting cold in the room where my battery is stored i am wondering if you have found anything in the code or CAN messages that indicates of the heating element of the V5 battery is currently active? Thanks.

stefan123t, 2024-11-21:
@ranma I had a bit of play with the information and output you provided:

<details><summary>CAN Protocols via DIP Switches on the Sticker</summary>
<p>

```
Victron 000000
Voltronic 100000
Deye 010000
Goodwe 110000
Growatt 001000
SRNE 101000
Studer 011000
MUST 111000
Afore 000100
100100
Solis 010100
SMA 110100
Luxpower 001100
SMT 101100
Kelong 011100
Megarevo 111100
Hoymiles 000011
``` 

</p>
</details> 

<details><summary>Supported CAN Protocols according to setptr HELP</summary>
<p>

```
VICTRON
GROWATT
GOODWE
MEGAREVO
SRNE
DEYE
VOLTRONIC
LUXPOWER
STUDER
KEHUA
AFORE
SCHNEIDER
SOLAX
SOLIS
SMA
PYTES
MUST
ASW
INGETEAM
SOLARK
SENERGY
LIVOLTEK
NEP
``` 

</p>
</details> 

<details><summary>* CAN Protocols not shown on the DIP switch Sticker</summary>
<p>

```
KEHUA
SCHNEIDER
SOLAX
PYTES
ASW
INGETEAM
SOLARK
SENERGY
LIVOLTEK
NEP
``` 

</p>
</details> 

Three of them were not defined explicitly with a case in the function you extracted from the STM32 code.

* SOLAX ?
* PYLON ?
* PYTES ? (default)

CAN Protocol | DIP Switches | Hex Value | Decimal Value
--- | --- | --- | ---
Victron   | 000000 | 0x00 | 00
Voltronic | 100000 | 0x01 | 01
Deye      | 010000 | 0x02 | 02
Goodwe    | 110000 | 0x03 | 03
Growatt   | 001000 | 0x04 | 04
SRNE      | 101000 | 0x05 | 05
Studer    | 011000 | 0x06 | 06
MUST      | 111000 | 0x07 | 07
Afore     | 000100 | 0x08 | 08
Schneider | 100100 | 0x09 | 09 *
Solis     | 010100 | 0x0A | 10
SMA       | 110100 | 0x0B | 11
Luxpower  | 001100 | 0x0C | 12
SMT / ASW | 101100 | 0x0D | 13 *
Kelong / KEHUA | 011100 | 0x0E | 14 *
Megarevo  | 111100 | 0x0F | 15
INGETEAM  | 000010 | 0x10 | 16 *
SOLARK    | 100010 | 0x11 | 17 *
SENERGY   | 010010 | 0x12 | 18 *
LIVOLTEK  | 110010 | 0x13 | 19 *
NEP       | 001010 | 0x14 | 20 *
n/a       | 101010 | 0x15 | 21
n/a       | 011010 | 0x16 | 22
n/a       | 111010 | 0x17 | 23
n/a       | 000110 | 0x18 | 24
n/a       | 100110 | 0x19 | 25
n/a       | 010110 | 0x1A | 26
n/a       | 110110 | 0x1B | 27
n/a       | 001110 | 0x1C | 28
n/a       | 101110 | 0x1D | 29
n/a       | 011110 | 0x1E | 30
n/a       | 111110 | 0x1F | 31
n/a       | 000001 | 0x20 | 32
n/a       | 100001 | 0x21 | 33
n/a       | 010001 | 0x22 | 34
n/a       | 110001 | 0x23 | 35
n/a       | 001001 | 0x24 | 36
n/a       | 101001 | 0x25 | 37
n/a       | 011001 | 0x26 | 38
n/a       | 111001 | 0x27 | 39
n/a       | 000101 | 0x28 | 40
n/a       | 100101 | 0x29 | 41
n/a       | 010101 | 0x2A | 42
n/a       | 110101 | 0x2B | 43
n/a       | 001101 | 0x2C | 44
n/a       | 101101 | 0x2D | 45
n/a       | 011101 | 0x2E | 46
n/a       | 111101 | 0x2F | 47
Hoymiles  | 000011 | 0x30 | 48
n/a       | 100011 | 0x31 | 49
n/a       | 010011 | 0x32 | 50
n/a       | 110011 | 0x33 | 51
n/a       | 001011 | 0x34 | 52
n/a       | 101011 | 0x35 | 53
n/a       | 011011 | 0x36 | 54
n/a       | 111011 | 0x37 | 55
n/a       | 000111 | 0x38 | 56
n/a       | 100111 | 0x39 | 57
n/a       | 010111 | 0x3A | 58
n/a       | 110111 | 0x3B | 59
n/a       | 001111 | 0x3C | 60
n/a       | 101111 | 0x3D | 61
n/a       | 011111 | 0x3E | 62
n/a       | 111111 | 0x3F | 63

ranma, 2024-11-25:
> Hey @ranma, now that its getting cold in the room where my battery is stored i am wondering if you have found anything in the code or CAN messages that indicates if the heating element of the V5 battery is currently active? Thanks.

I only looked at the E-BOX-4850 code, which doesn't have a heating element. Though there is some reference in the firmware to the BMS potentially supporting one: `Contrl IO - ctrl [cfet/dfet/buzz/heat] [on/off]`

ranma, 2024-11-25:
> Three of them were not defined explicitly with a case in the function you extracted from the STM32 code.
> 
> * SOLAX ?
> * PYLON ?
> * PYTES ? (default)

Indeed in my fw version PYLON (internal id 0x10) is only settable using the setprt command AFAICS. setprt uses this helper:

```
int get_protocol_idx_from_name_FUN_08029780(char *name)

{
  int iVar1;
  
  iVar1 = strcmp_FUN_080072ec(s_VICTRON_08029918,name);
  if (iVar1 == 0) {
    return 8;
  }
  iVar1 = strcmp_FUN_080072ec(s_GOODWE_08029920,name);
  if (iVar1 == 0) {
    return 10;
  }
  iVar1 = strcmp_FUN_080072ec(s_GROWATT_08029928,name);
  if (iVar1 == 0) {
    return 9;
  }
  iVar1 = strcmp_FUN_080072ec(s_STUDER_08029930,name);
  if (iVar1 == 0) {
    return 0x11;
  }
  iVar1 = strcmp_FUN_080072ec(s_AFORE_08029938,name);
  if (iVar1 == 0) {
    return 0x13;
  }
  iVar1 = strcmp_FUN_080072ec(s_LUXPOWER_08029940,name);
  if (iVar1 == 0) {
    return 0xf;
  }
  iVar1 = strcmp_FUN_080072ec(s_KEHUA_0802994c,name);
  if (iVar1 == 0) {
    return 0x12;
  }
  iVar1 = strcmp_FUN_080072ec(s_MEGAREVO_08029954,name);
  if (iVar1 == 0) {
    return 0xb;
  }
  iVar1 = strcmp_FUN_080072ec(s_SOLAX_08029960,name);
  if (iVar1 == 0) {
    return 0x15;
  }
  iVar1 = strcmp_FUN_080072ec(s_PYLON_08029968,name);
  if (iVar1 == 0) {
    return 0x10;
  }
  iVar1 = strcmp_FUN_080072ec(s_SOLIS_08029970,name);
  if (iVar1 == 0) {
    return 0x16;
  }
  iVar1 = strcmp_FUN_080072ec(s_SCHNEIDER_08029978,name);
  if (iVar1 == 0) {
    return 0x14;
  }
  iVar1 = strcmp_FUN_080072ec(s_SMA_0802990c,name);
  if (iVar1 == 0) {
    return 0x17;
  }
  iVar1 = strcmp_FUN_080072ec(s_PYTES_08029984,name);
  if (iVar1 == 0) {
    return 0x18;
  }
  iVar1 = strcmp_FUN_080072ec(s_VOLTRONIC_0802998c,name);
  if (iVar1 == 0) {
    return 0xe;
  }
  iVar1 = strcmp_FUN_080072ec(s_MUST_08029998,name);
  if (iVar1 == 0) {
    return 0x19;
  }
  iVar1 = strcmp_FUN_080072ec(s_ASW_08029910,name);
  if (iVar1 == 0) {
    return 0x1a;
  }
  iVar1 = strcmp_FUN_080072ec(s_INGETEAM_080299a0,name);
  if (iVar1 == 0) {
    return 0x1b;
  }
  iVar1 = strcmp_FUN_080072ec(s_SOLARK_080299ac,name);
  if (iVar1 == 0) {
    return 0x1c;
  }
  iVar1 = strcmp_FUN_080072ec(s_SRNE_080299b4,name);
  if (iVar1 == 0) {
    return 0xc;
  }
  iVar1 = strcmp_FUN_080072ec(s_DEYE_080299bc,name);
  if (iVar1 == 0) {
    return 0xd;
  }
  iVar1 = strcmp_FUN_080072ec(s_SENERGY_080299c4,name);
  if (iVar1 == 0) {
    return 0x1d;
  }
  iVar1 = strcmp_FUN_080072ec(s_LIVOLTEK_080299cc,name);
  if (iVar1 == 0) {
    return 0x1e;
  }
  iVar1 = strcmp_FUN_080072ec(s_NEP_08029914,name);
  if (iVar1 == 0) {
    return 0x1f;
  }
  return -1;
}
```

which is called here:
```
int print_or_set_protocol_FUN_08029e60
              (undefined4 param_1,undefined4 param_2,uint8_t argc,char **argv)

{
  struct extcan_config *psVar1;
  char *pcVar2;
  uint32_t uVar3;
  int iVar4;
  uint uVar5;
  uint uVar6;
  uint local_18;
  char **ppcStack_14;
  
  psVar1 = PTR_struct extcan_config_08029fc8;
  local_18 = 0;
  ppcStack_14 = argv;
  if (argc == '\x01') {
    [...setprt without arg should print current settings...]
  else if (argc == '\x02') {
    pcVar2 = argv[1];
    iVar4 = strcmp_FUN_080072ec(pcVar2,s_DIP_08029fbc);
    if (iVar4 == 0) {
      BSP_EEPROM_write_data_FUN_08007a6a(0x787,&local_18,1);
      return 0;
    }
    uVar5 = get_protocol_idx_from_name_FUN_08029780(pcVar2);
    if ((int)uVar5 < 0) {
      uVar5 = 0;
      printf(s__%-20s_0802a06c,s__Support_Protocol_:_0802a074);
      for (uVar6 = 8; uVar6 < 0x20; uVar6 = uVar6 + 1) {
        pcVar2 = get_can_prt_name_FUN_08016674(uVar6);
        if (pcVar2 != (char *)0x0) {
          printf(s_%-10s_0802a08c,pcVar2);
          if (uVar5 % 6 == 5) {
            printf(s__%-20s_0802a06c,0x8029fc0);
          }
          uVar5 = uVar5 + 1;
        }
      }
      printf(s__08029fc4);
      return 1;
    }
    PTR_struct extcan_config_08029fc8->protocol_type = uVar5;
    local_18 = local_18 & 0xffffff00 | uVar5 & 0xff;
    BSP_EEPROM_write_data_FUN_08007a6a(0x787,&local_18,1);
    can_extcan_comm_protocol_init_FUN_08015c34();
  }
  return 0;
}    
```