hoylabs/OpenDTU-OnBattery #1189: Bug in getBatteryPower()

CyberDem0n, 2024-08-23:
### What happened?

It could happen that `_batteryDischargeEnabled` will not be reset in the morning, as a result battery will be used while it is supposed to be charged.

### To Reproduce Bug

DPL settings
========
Target Grid Consumption: -10W
Hysteresis: 8W
Target Inverter: HM-1500
Inverter is powered by solar modules : false
Minimum Power Limit: 30W
Base Load: 110W
Maximum Power Limit: 800W
PowerMeter reading includes inverter output: true
Enable Solar-Passthrough: true
Use battery at night even if only partially charged: true
(Full) Solar-Passthrough Losses: 7%
Ignore Battery SoC: false
Start Threshold for Battery Discharging: 80%
Stop Threshold for Battery Discharging: 15%
Full Solar-Passthrough Start Threshold: 0% (disabled).

Analysis
=====
We (re)start OpenDTU-OnBattery during the day, while `SOC` is between start and stop thresholds.
In this case `_batteryDischargeEnabled` and `_nighttimeDischarging` are both `false`.
Later, during the day, `SOC` reaches `BatterySocStartThreshold`, and `_batteryDischargeEnabled` is flipped to `true`.
When the sunset happens, we have `isDayPeriod = false`, `_batteryDischargeEnabled = true`, and `_nighttimeDischarging = false`.
As a result, the following branch is not executed (`!_batteryDischargeEnabled` is `false`): https://github.com/helgeerbe/OpenDTU-OnBattery/blob/f8ad0a4487d13fcc54bf28d56656e23ce7a7c23e/src/PowerLimiter.cpp#L254-L259

And `_nighttimeDischarging` is not flipped to `true`.

During the night hours in my case `SOC` doesn't reach `BatterySocStartThreshold`.

In the morning we have `isDayPeriod = true`, `_batteryDischargeEnabled = true`, and `_nighttimeDischarging = false`, and as a result the following code is not executed either: https://github.com/helgeerbe/OpenDTU-OnBattery/blob/f8ad0a4487d13fcc54bf28d56656e23ce7a7c23e/src/PowerLimiter.cpp#L245-L248

### Expected Behavior

Battery discharge is stopped in the morning and resumed only when `SOC` reaches `Start Threshold for Battery Discharging`.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.08.18

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

AndreasBoehm, 2024-08-23:
I guess that this is what i observed here as well: https://github.com/helgeerbe/OpenDTU-OnBattery/issues/784#issuecomment-2292888243

CyberDem0n, 2024-08-23:
@AndreasBoehm yep, that's the reason why I first posted a message to that issue.

schlimmchen, 2024-08-27:
> Battery discharge is stopped in the morning and resumed only when SOC reaches Start Threshold for Battery Discharging.

This is the key issue, right? You guys expect that a discharge cycle ends (and a charge cycle starts) once a new day begins?

I guess currently many (most?) people assume that the charge and discharge cycles are somehow connected to the day period. I had this assumption myself while trying to understand the DPL, until I decided to throw it out of the window and set my start threshold to 52V while my stop threshold is 51V, so I am basically "always" allowing to use the battery. Howver, the battery cycles are de-coupled from time. They happen to coincide with sunrise and sunset (or nighttime) somehow, but that is by chance, not by design. And it depends on the system setup: A sufficiently large battery will "never" get out of a discharge cycle in the summer, it can just be used at will. A small battery will always be drained in the morning, so a charge cycle starts in any case for those setups.

@spcqike @Mano1966 @AndreasBoehm Are we sure we want to "fix" this, i.e., start a charge cycle once daytime begins? Should be fairly simple to implement. Need to update the docs, though. I guess the behavior is more intuitive then? And people like me who have the start threshold fairly low don't really care in any case, as the charge cycle is not really a thing for them. People who *want* to charge their battery during the day, each day, will profit if we changed this.

CyberDem0n, 2024-08-27:
> I guess currently many (most?) people assume that the charge and discharge cycles are somehow connected to the day period.

Well, yeah, maybe.

> A sufficiently large battery will "never" get out of a discharge cycle in the summer, it can just be used at will. A small battery will always be drained in the morning, so a charge cycle starts in any case for those setups.

The main question here is how battery counts these cycles. Or in other words - whether this behavior wearing it out or not.
TBH, I don't have an answer.

spcqike, 2024-08-27:
I guess I’m with @schlimmchen.
 I wouldn’t force a discharge stop in the morning unless the voltage or SoC is to low. Once the voltage reaches startthreshold, I would use the battery, regardless of the daytime. 

Why should I stop using the battery at like 8am and buy energy from the grid at this time? Especially with a dynamic tariff, where 8am is the most or second most expensive period of time to buy energy? 

> Are we sure we want to "fix" this, i.e., start a charge cycle once daytime begins?

I wouldn’t do it. At least not without yet another switch to let the user decide whether or not to have this behavior. 

And how does this work with solar passthrough? Daytime or when solar power > household need or whatever for the first time? 

> The main question here is how battery counts these cycles. Or in other words - whether this behavior wearing it out or not.

That’s a complex question. AFAIK the rule of thumb is that 100% discharge is 1 cycle. It „doesn’t matter“ if its 1x 100-0% or 10x 100-90%. But aging is faster at the very low and very high end. So staying within 20-80% is healthier. 

But still 2x 80-30% equals 100% and therefore 1 cycle. But if you always stay within 20-80 you can expect to get more cycles as if you use 0-100% all the time.

AndreasBoehm, 2024-08-28:
I think that we should fix this issue, but we should also add a new toggle that allows people like us to use the battery always, as long as it did not reach the stop threshold.

I am still new to the topic, and i was also pretty confused about charge cycles and if they are connected to a day period or not.

With the current toggle 'use battery always at night' i felt like it is implicitly saying that it will stop in the morning to start the 'battery protection'-mode again by allowing either charge or discharge, but not both at the same time.

schlimmchen, 2024-12-03:
I tend to document the behavior and not to change it: Charge cycles are decoupled from time in general and only depend on the charge state of the battery. Exception: If we started nighttime discharging due to the existing switch, then we need to stop it at sunrise. That's implemented.

However, if we started a discharge cycle at day n and it continues through day n+y because the stop threshold is never reached... That's fine. I think it should actually be like this, so users can make the most use of the stored energy. If we stop discharging at sunrise but the battery still has a significant charge, why would we stop using the battery just because the sun will be up shortly?

It's understandable that the battery should charge fully from time to time. That's on the order of weeks, not every day. And that is something for the battery watchdog #656 (which could prevent battery discharging after a week or so of use until the battery is once "fully" charged again).

Without objections, I am going to add some notes in the docs and close this issue in the next days or weeks.

AndreasBoehm, 2024-12-04:
I fully agree with you. Lets document it and not change anything.

schlimmchen, 2024-12-04:
[image]

https://github.com/hoylabs/OpenDTU-OnBattery-docs/commit/7572b7ff560c64c25873797e2bf8569d12e3ce97

closed 2024-12-04 by schlimmchen

stefan123t, 2024-12-05:
While I fully agree with your decision regarding charge cycles in the morning/evening, i somehow sense that I am a bit late to the party.

> Why should I stop using the battery at like 8am and buy energy from the grid at this time? Especially with a dynamic tariff, where 8am is the most or second most expensive period of time to buy energy?

I just read about legal requirements to support dynamic tariffs within a period of fifteen minutes. A set of up to / at least four different tariffs has to be supported by Smart Meters and Energy providers. I am currently considering a larger setup for a WEG / community project where you might be considered an energy provider and you may have to adhere to those regulations too.

i will try to look up a reference and date for this, it was a german whitepaper for “Mieterstrom” and different market models to implement such a project.

i understand that these requirements are more related to “monetizing” the power stored for the community and not directly connected to the technical logic of charge cycles. I still think those may have an impact on the considerations for one or the other.

it is probably best to continue that related (off-)topic in a separate feature request in depth.