hoylabs/OpenDTU-OnBattery #1194: [Request] Logging console message to syslog

ranma, 2024-08-24:
### Is your feature request related to a problem? Please describe.

Grabbing log messages from the web UI is cumbersome and I'd like to have a continuously running log sink.

### Describe the solution you'd like

It would be nice if a syslog address could be configured and console message could be sent to a syslog-compatible sink in addition to the web UI.
https://datatracker.ietf.org/doc/html/rfc5424

### Describe alternatives you've considered

Send log messages over MQTT

### Additional context

Syslog is a commonly-supported logging destination for embedded systems

AndreasBoehm, 2024-08-24:
I was thinking about the same thing but i dont see that a lot of people will use that to be honest.

ranma, 2024-09-15:
Upstream request: https://github.com/tbnobody/OpenDTU/issues/1819

stefan123t, 2024-09-16:
@ranma thanks for implementing this. I did not spot your earlier issue downstream, sorry.

schlimmchen, 2024-09-16:
> I was thinking about the same thing

Me, too.

> but i dont see that a lot of people will use that to be honest.

I think that four people is far enough :grin:

The upstream PR is ready, is it not? Can you rebase that branch onto our development branch and open a PR? If this "just works", I am happy to merge it. I guess you have to put in some work as MessageOutputClass looks very different in the upstream?

I would be oven more interested in logging the serial console itself, as it (and only it) includes backtraces, but even the log should be interesting.

schlimmchen, 2024-09-16:
I did some very limited research and it turns out that, despite its drawbacks, syslog seems to be the best option for this purpose. I wan concerned that we should not introduce such old and untrusted stuff, but I guess it's the best option? Please tell me if you think otherwise.

ranma, 2024-09-20:
> I would be oven more interested in logging the serial console itself, as it (and only it) includes backtraces, but even the log should be interesting.

You can hook the system log messages by calling `esp_log_set_vprintf`, but then you have to be careful to avoid deadlocks in case `esp_log_set_vprintf` is called again during your logging handler.

ranma, 2024-09-20:
> The upstream PR is ready, is it not? Can you rebase that branch onto our development branch and open a PR? If this "just works", I am happy to merge it.

Done.

ranma, 2024-09-28:
Fixed in 20159f341e44a5ea5c25e010f49ef4b0ea475189

closed 2024-09-28 by ranma