hoylabs/OpenDTU-OnBattery #1206: Power Meter digest mode not working anymore on [2024.08.18] and [Shelly 20240822-121054] (auth mode "none" works still)

gitisgreat2023, 2024-08-28:
### What happened?

Just flashed a board on 2024.08.18. The power meter gives then errors while trying to read out the Shelly in digest mode. Auth "none" still works though.

The errors are various, most common is:
1: HTTP Error: connection lost

I tried many times in "Test configuration" to see other errors. I saw once a power value (in digest mode!) out of 40 times trying. I'm not sure however anymore, couldn't get it a second time.

### To Reproduce Bug

Shelly 20240822-121054/1.4.3-g143ff62, 240725095835-ebee9893-HEAD
Set Authentication to Enable password protected device.

Then set in OpenDTU-On-Battery Auth mode to digest.
It worked perfectly in 2024.06.03.

### Expected Behavior

Digest mode works like in in 2024.06.03.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.08.18

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

schlimmchen, 2024-08-28:
Please test [this firmware](https://github.com/helgeerbe/OpenDTU-OnBattery/actions/runs/10497191892?pr=1183).

gitisgreat2023, 2024-08-28:
Tested it, it works! Thanks alot, that was lightning fast!!
What was the issue? What broke it?

schlimmchen, 2024-08-28:
You haven't noticed the PR I linked :wink: See #1183.

Thanks for testing. I will merge the PR, which will close this issue.

closed 2024-08-28 by schlimmchen

gitisgreat2023, 2024-08-28:
> You haven't noticed the PR I linked 😉 See #1183.
> 
> Thanks for testing. I will merge the PR, which will close this issue.

I quickly scanned the commits [here](https://github.com/helgeerbe/OpenDTU-OnBattery/compare/b0ac3f4...HEAD) , didn't see any so thought better report a bug...
Missed the PR link indeed, yeah looked at it, that was a little work it seems. Great you fixed it, thanks!

schlimmchen, 2024-08-28:
Hm, let me clarify to make sure you know that I wasn't complaining: I mentioned again that I linked the respective PR because you said the fix was lightning fast. The fix took several hours, but I had completed it already.

I was **not** mentioning it because I thought you should have noticed before you opened this issue. On the contrary: I was waiting for someone (else) to report this :wink:

gitisgreat2023, 2024-08-28:
Okay got it... even with having it implemented already you were pretty fast in replying to my issue and referring to the right firmware... ;-)