hoylabs/OpenDTU-OnBattery #1209: Dynamic Power Limiter Änderungen werden nicht übernommen.

openKimchi, 2024-08-30:
### What happened?

Nachdem ich 2 von 4 PV-Modulen vom Inverter (HM-1500) getrennt hatte, wollte ich das maximale Leistungslimit von 1500W auf 800W herabsetzen. Der neue Wert wird zwar in den Einstellungen übernommen aber nicht am Gerät. Es wird mir zB. immer noch 325W mit 25% angezeigt obwohl 25% jetzt 200W sein müssten.

### To Reproduce Bug

Einstellung für maximales Leistungslimit ändern.

### Expected Behavior

Neue Einstellung wird übernommen

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.08.18

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

spcqike, 2024-08-30:
ich kann dein Problem nicht ganz nachvollziehen.

Wo wird was angezeigt? sind die 325W die erzeugte Leistung oder das gesetzte Limit, oder was ganz anderes?

und: welche Eingänge hast du nun belegt? 1 und 2 oder 1 und 3 oder .... ? Das ist für die Interpretation des Verhaltens des DPL beim HM-1500 wichtig, da er geteilte MPPT hat.

SW-Niko, 2024-08-30:
Nein das ist kein Bug. Die Prozentangabe bezieht sich immer auf die maximale Inverter Leistung! 
Darum ändert sich auch nichts an der Prozentangabe wenn du das Leistungslimit änderst.
Du wirst feststellen, das der DPL auf die 800W begrenzt. Die Anzeige sollte dann 800W / 53,3% sein.

openKimchi, 2024-08-30:
[image]

Dort müsste aber 246 | 30,5% stehen, wenn ich im DPM 800W eingestellt habe.
Die verbleibenden Panels hängen an String 1 und 3

openKimchi, 2024-08-30:
@SW-Niko : Ich habe mir schon so etwas gedacht. Ich finde es aber dennoch etwas verwirrend.

spcqike, 2024-08-30:
ah, jetzt hab ichs auch verstanden.

ne, das ist eigentlich nicht verwirrend. die Leistung und die % beziehen sich halt immer auf den Wechselrichter, nicht auf das selbst eingestellte maximale Limit im DPL.

Im Gegenteil: Je nach Setup kann das tatsächlich übertragene Limit sogar größer sein, als das eingestellte maximale Limit. Dabei stellt der DPL dennoch sicher, dass das maximale Limit AC seitig eingehalten wird.

banales Beispiel: ein HM-600 wird auf 300W max eingestellt, aber nur 1 Eingang belegt. Um nun die 300W AC seitig zu bekommen, muss der DPL 600W als Limit übertragen, da das Limit Hoymiles intern immer auf alle MPPT gleichmäßig aufgeteilt wird. Ist nun einer nicht belegt, muss das zu übertragene Limit verdoppelt werden, damit der verbleibende belegte Eingang die Leistung auch bereitstellen kann.

openKimchi, 2024-08-30:
Alles klar, danke! Dein Beispiel ist einleuchtend und jetzt wird für mich auch ein Schuh draus.

spcqike, 2024-08-30:
wenn das Thema für dich damit geklärt ist, kannst du bei Gelegenheit das Issue schließen 😃

closed 2024-08-30 by openKimchi