hoylabs/OpenDTU-OnBattery #1211: [Feature Request] Signalstärkeanzeige

schlimmchen, 2024-08-30:
@spcqike 
Eine Möglichkeit den "Empfangspegel" (so ungenau er auch sein mag) als Live-Wert angezeigt zu kriegen wäre vielleicht schon nicht schlecht. Ich könnte mir vorstellen diesen Wert auf der Web-Console im 1-2-3 Sekunden Abstand zu loggen, natürlich über einen Menüpunkt an- und abschaltbar. Ich kann mich der Argumentation daß das auf der Haupt-Info-Seite nichts zu suchen hat voll anschliessen. (und ich müsste mal schauen ob die Web-Console schon im OpenDTU Projekt drin ist, wenn ja wäre das ein Feature Request für OpenDTU).

_Originally posted by @madmartin in https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/1180#discussioncomment-10497865_

Es ist durchaus denkbar, dass man den Wi-Fi und den NRF und den CMT2300 Pegel auf einer Stelle in der Web UI anzeigt.

Sicherlich nicht im Display, das ist aufwändig.

Probleme:
* NRF und CMT Infos stehen aktuell unter Info -> System, während Wi-Fi Signalstärke unter Inter -> Netzwerk zu finden ist. Wo wird also die Stelle sein, wo alles auf einmal zu sehen ist?
* Damit das Feature Sinn macht, müssen die Werte über einen Websocket regelmäßig ausgespuckt werden.
* Wie kommt man in den oberen Softwareschichten an diese Signalqualität der Inverter-Funkmodule ran? Gibt es überhaupt bei beiden Modellen einen Indikator, oder nur beim NRF?

gitisgreat2023, 2024-09-03:
NRF und CMT signal strength (dBm) anzeigen im Web UI wäre super! Ich logge fast nur in der Konsole ein damit ich die signal strengths sehen kann...

Zum Punkt 3: für NRF und CMT gibt's in der Konsole signal strength bei mir...

DanielR92, 2024-09-04:
NRF24 ist hier zwecks signal strength extrem ungenau! Ist mehr so ein:"joa irgendwas höre ich... oder nicht?"
` p.rssi = mNrf24->testRPD() ? -64 : -75;`

CMT kann hier die Signal Stärke viel genauer ausgelesen werden. 
`*rssi = mSpi.readReg(CMT2300A_CUS_RSSI_DBM) - 128;`

DanielR92, 2024-09-07:
Gerade gesehen, bei openDTU gibt es dazu bereits ein Issue!
https://github.com/tbnobody/OpenDTU/issues/2254

schlimmchen, 2024-09-07:
*gibt es dazu inzwischen auch ein Issue.

Falls wir früher dran sein sollten, werde ich es sicherlich auch per PR fürs Upstream-Projekt vorschlagen. Das sollten wir grundsätzlich so handhaben, wann immer sich Features für beide Projekte eignen. Zuletzt gabs aber wenig Überschneidungen. #1193 werde ich noch vorschlagen, aber das eilt nicht, Thomas wird es nicht mögen, unterstelle ich 😉

spcqike, 2024-09-08:
> Falls wir früher dran sein sollten

#2255

SciLor, 2024-09-30:
Jup, ich habe das für meinen Einsatzzweck bereits implementiert. Leider gibt es weder zum Issue noch zum PR eine Rückmeldung. Mittlerweile gibt es auch Konflikte im PR.

schlimmchen, 2024-10-06:
Meiner Meinung nach erledigt mit edfe06e31eae286fbb88da69e9b65420cfa8628a bzw. dem Merge in dieses Projekt durch b2913f24d300c6a991bdcac2b071cefc181dbc8d.

closed 2024-10-06 by schlimmchen