hoylabs/OpenDTU-OnBattery #1227: [Request] use victron battery sense if available

spcqike, 2024-09-05:
### Is your feature request related to a problem? Please describe.

as discussed here 
https://github.com/helgeerbe/OpenDTU-OnBattery/pull/1214#issuecomment-2329737784

### Describe the solution you'd like

> Wenn jemand einen SmartShunt hat und den als BatteryProvider einbindet, aber auch einen BatterySense hat, dann bevorzugen wir zur Zeit die Spannung des SmartShunt, obwohl der des BatterySense besser wäre.

it would be good to use the voltage the charger provides, if a battery sense module is connected.

so order of reliable voltage measurement would be

1. BMS
2. Victron solar charger if batterysense is connected
3. smart shunt
4. victron charger without batterysense
5. inverter input

> Darum frage ich auch zusätzlich immer erst die Batterietemperatur ab um sicher zu gehen das ein Smart Batterie Sense angeschlossen ist.

we should check whether or not a battery sense is available and use the voltages accordingly.

### Describe alternatives you've considered

leave everything as it is and use the shunt voltage instead, which can be a little less accurate.

iamdevnull, 2024-10-13:
Ich bin aktuell dabei meinen Victron Smart Battery Sense zurückzusenden.

Welchen Zweck sollte der Battery Sense erfüllen?

- Messung der Spannung  zur Ermittlung des Ladestand der Batterie (SoC)

Warum sendest du es zuück?

- Weil der Battery Sense die Spannung an den Polen misst, diese Spannung kann jedoch bei Last/Aufladung/Entladung abweichen. Einerseits durch Spannungsabsfall oder bei MPPT sogar in dem es variiert.

Warum würdest du den Smart Shunt bevorzugen?

- Weil der Smart Shunt zumindest auch den Stromfluss bei der Kalkulation in Betracht zieht.

Was würdest du als die zuverlässigste Quelle zur Ermittlung der Spannung/SoC in Betracht ziehen?

- Daten aus dem BMS auslesen und über MQQT zur Verfügung stellen und diese dann in OpenDTU-OnBattery verwerten.

Ich bin Informatiker und kein Elektrotechniker. Verzeiht mir, falls ich eine falsche Annahme aufbauen sollte.

@schlimmchen wie siehst du meine Aussagen?

schlimmchen, 2024-10-13:
> diese Spannung kann jedoch bei Last/Aufladung/Entladung abweichen

Das geht dem SmartShunt doch genauso?!

> Weil der Smart Shunt zumindest auch den Stromfluss bei der Kalkulation in Betracht zieht.

Nur bei der Kalkulation des SoC, oder?

> Daten aus dem BMS auslesen und über MQQT zur Verfügung stellen und diese dann in OpenDTU-OnBattery verwerten.

Wenn du ein BMS hast, an dessen daten du rankommst, braucht du weder smart shunt noch smart battery sense...

iamdevnull, 2024-10-13:
Ich gehe stark davon aus, dass deine rhetorischen Fragen nicht einfach so gestellt werden. Aus diesem werde ich deine Aussagen zustimmen 😄

Ich versuche aktuell mithilfe von Batmon + HA über MQTT die Werte der OpenDTUonBat zugänglich zu machen.

schlimmchen, 2024-10-13:
JBD BMS Support in OpenDTU-OnBattery kommt.

iamdevnull, 2024-10-14:
Gibt es eigentlich eine fertige Platine im Verkauf für den Anschluss eines BMS z.B JBD ?

schlimmchen, 2024-10-14:
https://opendtu-onbattery.net/3rd_party/opendtu_fusion/

Im Zweifel mit dem CAN/ISO Shield, wenn das BMS per UART angeschlossen werden soll. Wenn es per RS485 angeschlossen wird, wie ein JK BMS, dann ist die Grundplatine ausriechend, das ist schon ein RS485 drauf.

Wie das JBD angeschlossen wird, weiß ich gerade nicht auswendig -- ich hab keins, ich kenne nur den pull request, den ich plane einzupflegen.

schlimmchen, 2024-10-14:
> ich hab keins

Demnächst hab ich eins :wink: Angeschlossen wird es wohl per UART. Dann wäre das OpenDTU Fusion mit CAN/Iso Shield das richtige, um es direkt an OpenDTU-OnBattery einzubinden.

SW-Niko, 2024-12-09:
Hallo @spcqike 
Ich habe mir das Thema nochmal angeschaut.

Wenn man einen Smart-Shunt und ein Smart-Battery-Sense gleichzeitig über das VE-Smart Netzwerk mit einem Laderegler verbindet, dann verwendet der Laderegler die Daten vom Smart-Shunt.
Screenshot aus der Victron-App:

[image]

Vermutlich gibt Victron den Smart-Shunt Daten den Vorrang, weil diese Messung genauer ist (jedenfalls nach Datenblatt) und eine bessere Auflösung haben. (Smart-Shunt 1mV, Smart-Battery-Sense 10mV)

Die Batterie Spannungs-Daten vom Laderegler werden im Laderegler bereits durch die Spannungswerte vom Smart-Shunt oder Smart-Battery-Sense ersetzt. Das bedeutet ... das wenn wir Spannungsdaten vom Laderegler (mit  Smart-Battery-Sense) auslesen dann sind das automatisch die Daten von der Smart-Battery-Sense.

Darum denke ich das wir unsere derzeitige Prioritäten-Verarbeitung nicht verändern müssen und den Request schließen können.

1. Batterie Provider (BMS, Victron Smart Shunt)
2. Victron Laderegler (Laderegler, Smart-Battery-Sense, Smart Shunt)
4. Inverter

Spricht was dagegen?

schlimmchen, 2024-12-09:
Also die Annahme war folgende: Man hat einen SmartShunt und nutzt den als Battery Provider in OpenDTU-OnBattery, aber man hat auch einen Victron Solarladeregler, der per VE.Smart Netzwerk mit einem SmartBatterySense verbunden ist. Dann war das Argument, dass die Spannung vom SmartBatterySense über den Solarladeregler bevorzugt werden soll, obwohl normalerweise die Spannung vom Battery Provider den höchsten Rang hat. Das -- glaube ich -- hast du nicht richtig betrachtet: Das jetzige Schema wird den Spannungswert vom SmartShunt nehmen. Gefordert ist aber: Falls ein BatterySmartSense im System ist, dann soll die Ladereglerausgangsspannung den höchsten Rang haben.

Allerdings stelle ich fest, dass in diesem Setup der SmartBatterySense und der SmartShunt redundant sind. Der SmartShunt misst nicht nur den Strom durch die Batterie, sondern auch direkt an dessen Terminals, ohne Last, ihre Spannung. Warum sollte man also einen SmartBatterySense und einen SmartShunt gleichzeitig einsetzen? Bestenfalls um VE.Smart zu machen um eine genauere Spannung an den Solarladeregler zu bringen. Das ist aber ebenso redundant ist, weil der SmartShunt genau das gleiche tun kann -- siehe Niko' Screenshot oben.

Meiner Einschätzung nach macht der Request daher keinen Sinn.

Danke für deinen Input, Niko.

closed 2024-12-09 by schlimmchen