hoylabs/OpenDTU-OnBattery #1233: [Request] Optional setup for SmartMeter

DanielR92, 2024-09-06:
### Is your feature request related to a problem? Please describe.

I use Tibber and here it is connected to the battery. If too many requests happen, it will be queried unnecessarily and the batteries will be drained.

### Describe the solution you'd like

If communication with the inverter is not possible (night, etc..), the SmartMeter should no longer be spammed with requests.

### Describe alternatives you've considered

-

schlimmchen, 2024-09-06:
I rejected such a request before.

The important difference to that previous request is that the power meter now has a battery, which is indeed a limited resource.

This would have to be a switch, as not all users want to pause polling the power meter, even if the value is not needed by the DPL, because OpenDTU-OnBattery might be the only instance that publishes the reading via MQTT, which is desirable also during the night.

Why does that thing have a battery in the first place?! A battery-powered device in an electrical cabinet... So weird^^

spcqike, 2024-09-06:
as the bridge needs to query the pulse anyway, are we sure that querying the bridge does increase the communication between bridge and pulse? if so, i also support a toggle to let the user choose whether or not querying at night.

> Why does that thing have a battery in the first place?! A battery-powered device in an electrical cabinet... So weird^^

well, i guess because its easier to use a battery powered device. if you rent a flat, chances are high that you have a smart meter without an outlet in your cabinet. and having a grid powered device increases the effort to switch to tibber for many people as they need to call someone to install an outlet or 5V powersupply.

DanielR92, 2024-09-06:
Unfortunately, I live in a rented apartment.
In the basement, where most of the electrical boxes are, not every apartment has a socket available.

So the manufacturer did the right thing by equipping it with a battery. But yes, a USB interface would have been nice here too.

DanielR92, 2024-09-08:
You ask it every two seconds? Interesting! I have to slowly approach it.

closed 2025-02-16 by github-actions[bot]