hoylabs/OpenDTU-OnBattery #1238: ext Tempsensor DS18b20

pationair, 2024-09-08:
### Is your feature request related to a problem? Please describe.

Hello, is it possible to connect an external sensor for temperature monitoring?

The easiest would probably be a DS18B20.

### Describe the solution you'd like

The idea would be to put this in the battery box to measure the temperature of the battery.

I would appreciate feedback.

### Describe alternatives you've considered

I am currently using an Aqara Zigbee sensor

schlimmchen, 2024-09-08:
What BMS are you using and why does it not provide temperature data?

pationair, 2024-09-08:
Hello, I have a Power Queen 25.6V 100Ah battery, it has a BMS, but as far as I know I can't read it.

SW-Niko, 2024-09-28:
Hello @pationair ,
As an alternative solution you can maybe use the Victron "Smart Battery Shunt".
The Victron MPPT can use this information to stop charging the battery at low temperature and with PR #1292 you can see the temperature in the web UI.

schlimmchen, 2024-12-09:
As I don't even support this if the code was contributed from someone else, I am going to close this as "not planned".

closed 2024-12-09 by schlimmchen