hoylabs/OpenDTU-OnBattery #1247: add feature to charge akku with max AC power

onehub278, 2024-09-11:
### Is your feature request related to a problem? Please describe.

feature request: it would be nice to have a feature to charge akku with max AC power (Huawei R4850G2), when energy costs are very low

### Describe the solution you'd like

for example: add a max charging power field in watts and overrule auto power charger limiter via MQTT

### Describe alternatives you've considered

maybe i can do it manually under "AC Charger config" -> Disable "Auto power charger limiter" -> using "emergency charging" 

### Additional context

iam aWATTAr consumer and it is possible to get negative energycosts mostly at weekend, but also get very hight energycosts at early morning or at afternoon

gitisgreat2023, 2024-09-12:
See [here](https://github.com/MalteSchm/OpenDTU-OnBattery-Controller).

onehub278, 2024-09-14:
it seems all necessary commands are already implemented via API or MQTT:
https://opendtu-onbattery.net/firmware/web_api/#list-of-urls (/api/limit/config)
https://opendtu-onbattery.net/firmware/mqtt_topics/#inverter-total-topics (solar/powerlimiter/cmd/mode)

please close this request

thx

schlimmchen, 2024-09-14:
Okay, thanks for your feedback! Happy to see that you found a working solution :muscle:

closed 2024-09-14 by schlimmchen