hoylabs/OpenDTU-OnBattery #1253: [Request] Hide ETH pins on ESP32-S3

schlimmchen, 2024-09-13:
### Describe the solution you'd like

The pins/settings in the `eth` object of the `pin_mapping.json` are not applicable to ESP32-S3, as they don't have an Ethernet MAC. These pins should therefore be hidden in the device config of the web UI. We should also make sure that this type of Ethernet stack is not initialized on an ESP32-S3 even if the `eth` object is configured.

LennartF22, 2024-10-06:
FYI: https://github.com/tbnobody/OpenDTU/pull/2333

schlimmchen, 2024-10-06:
Yes, I just noticed this in the upstream changelogs. Thanks :heart: Will close this after merging the upstream into the development branch.

schlimmchen, 2024-10-06:
Completed by b2913f24d300c6a991bdcac2b071cefc181dbc8d and complimented by d6d5f32329731ffd16585c200bf6d424069a94c7.

closed 2024-10-06 by schlimmchen