hoylabs/OpenDTU-OnBattery #1254: [Request] Access serial console for Fusion Board

onehub278, 2024-09-14:
### Is your feature request related to a problem? Please describe.

It seems access to serial console for Fusionboard ist not possible
Serial output stops at:
E (348) esp_core_dump_flash: No core dump partition found!

### Describe the solution you'd like

please add same feature as described here:
https://opendtu-onbattery.net/firmware/howto/serial_console/

![opendtu-battery-on-FusionBoard](https://github.com/user-attachments/assets/45be311b-d1e4-4
![espresense-on-ESP32-S3-DevKitC-1](https://github.com/user-attachments/ass
[image]
ets/81f92324-9861-4495-a277-2b9df2db0501)
da7-a0d5-b6295b748ee6)

### Additional context

tested with several USB-Cables and other boards:
[image]
[image]
[image]

AndreasBoehm, 2024-09-14:
Which exact build did you flash on your fusion?
This information should be available on the system info page of the dtu.

onehub278, 2024-09-14:
> Which exact build did you flash on your fusion? This information should be available on the system info page of the dtu.

[image]

AndreasBoehm, 2024-09-14:
I think you should use the esp32s3_usb build instead

schlimmchen, 2024-09-14:
Yes, as is [documented)(https://opendtu-onbattery.net/firmware/#overview-of-firmware-variants).

@onehub278 You may use the `esp32s3` variant (without `_usb`), but you can't expect to see message output on the USB port in that case. The messages will appear on the UART header instead.

closed 2024-09-14 by schlimmchen

haegar33, 2024-10-07:
I got a openDTU-development branch in Platformio on Visual Code. There an enviroment **openfusionenv2** is available. Is this outdated or a new feature?

schlimmchen, 2024-10-07:
Neither. It's the same as `generic_esp32s3_usb`, except that it defines default values for some pins, which you can as well set using the recommended method: uploading a `pin_mapping.json`.