hoylabs/OpenDTU-OnBattery #1255: WebApp autocompile: Script does not detect yarn in vscode on Windows

schlimmchen, 2024-09-14:
### What happened?

People tried to build a custom firmware using a workspace that includes the webapp autocompile script and they failed as the script told them that yarn is not available even though they installed yarn/nodejs.

### To Reproduce Bug

Not sure, but probably this:
* Use Windows
* Install vscode
* Prepare the environment (clone repo, install PlatformIO)
* Install yarn/nodejs
* Confirm that the webapp can be build using cmd/powershell
* Remove webapp_dist/.hashes.pkl
* Try to build firmware inside vscode

### Expected Behavior

Pre-script finds yarn and uses it to compile the web app.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

123456

### Anything else?

See #1249.

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

Snoopy-HSS, 2024-09-15:
Hi,

yarn is usable inside vscode environment in a powershell terminal.
Only Phython does not regognize yarn.
Maybe Python does not use windows PATH variables???

I'm not that firm in Python under windows...

[image]

Same Problem is outside VSCODE in an Python CMDLINE.

Martin1275, 2024-09-19:
Hallo, ich habe heute das Python Script abgeändert, nun findet es yarn unter Windows.

Kann ich die geänderte Datei hier irgendwo hochladen damit ihr sie vergleichen könnt ? 
Bin komplett neu in VSCODE und GITHUB.

Was ist ein Commit oder Push ?

schlimmchen, 2024-09-19:
Äh, was? Du willst mir sagen ich verschwende gerade meine Zeit damit mir einen abzukrampfen das Problem wenigstens mal zu reproduzieren? :grin: Hätte ich das mal noch nen Tag liegen lassen :wink:

Klingt super! Deine Fragen deuten darauf hin, dass du den Inhalt der Datei bitte einfach mal als Kommentar postest, das wäre wohl das einfachste. Ich übernehme dann gerne. Danke!

Martin1275, 2024-09-19:
[image]

Martin1275, 2024-09-19:
ah hier kann man auch screenshots hochladen... hoffe das hilft

shell=True soll aber ein Sicherheitsrisiko sein.   

Die Info hab ich von hier: 
[https://stackoverflow.com/questions/24306205/file-not-found-error-when-launching-a-subprocess-containing-piped-commands](https://stackoverflow.com/questions/24306205/file-not-found-error-when-launching-a-subprocess-containing-piped-commands)

Martin1275, 2024-09-19:
Zum Testen hab ich im webapp_dist Verzeichnis dann die .hashes.pkl gelöscht

das Log sieht dann so aus:

[image]

spcqike, 2024-09-19:
Da hab ich gar nicht drauf geachtet. In meiner ursprünglichen Version des Skripts war Shell=True auch drin. 

https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/1151#discussioncomment-10239236

@schlimmchen hast du das raus genommen oder hab ich das in dem PR nicht mehr gehabt? Finde das nach den commits nicht mehr (:

schlimmchen, 2024-09-20:
Ich hab das absichtlich entfernt, weil ich fest der Meinung war, dass es nicht gebraucht wird, und weil ich weiß, dass es das grundsätzlich zu vermeiden gilt.

Da hier aber keine Benutzereingaben als Argumente verarbeitet werden, ist `shell=True` vermutlich kein Problem. Und wenn es das Skript für Windows-Benutzer braucht, dann machen wir das gerne rein.

Danke für die Hilfe beim Debuggen!

Warum `shell=True` nötig ist, ist mir nicht ganz klar. Vielleicht weil `yarn` eigentlich `yarn.cmd` ist? ChatGPT sagt, dass das Auflösen von Kommandos durch `subprocess.run()` auf Linux schon den `$PATH` berückksichtig, auf Windows aber nicht den `%PATH%`.

@Martin1275 Wie hast du denn yarn installiert? mit `npm install --global yarn` in einer Kommandozeile? Der Schritt fehlt nämlich in der Doku. Es scheint aber auch so zu sein, dass wir nicht den empfohlenen Weg gehen. Man soll `corepack enable` verwenden, um einen shim zu installieren für `yarn`, wobei `corepack` dann dafür sorgt, dass eine bestimmte, festgelegte Version von `yarn` verwendet wird, was mir sehr sinnvoll erscheint.

Martin1275, 2024-09-20:
Hmmm soweit ich mich erinnern kann, hab ich einen .msi Installer verwendet.

schlimmchen, 2024-09-20:
Vermutlich um Node.js zu installieren, was sinnvoll war. Aber dann hat man ja noch kein yarn?! Hast du `corepack enable` ausgeführt, weil du es irgendwo gelesen hast?

`shell=True` wurde in 789a07e288aa23717452c7849c341b00a3d293eb in branch `development` integriert.

Martin1275, 2024-09-20:
https://phoenixnap.com/kb/yarn-windows    hier gibt es den msi installer

schlimmchen, 2024-09-20:
Okay, verstanden. An alle anderen: Den offiziellen Installer gibts [auf Github](https://github.com/yarnpkg/yarn/releases/). @Martin1275 Ich habe deinen Link versteckt, weil die Leute bitte direkt auf die offizielle Github Seite sollen, falls sie diese Diskussion verfolgren.

Snoopy-HSS, 2024-09-20:
funktioniert wieder, auch in VS Code
Hat nur ein neueres Node.JS benötigt.

Danke für die hervorragende Arbeit, somit kann man die Platform.IO im Original lassen...

schlimmchen, 2024-09-21:
Das Thema ist mit
* 789a07e288aa23717452c7849c341b00a3d293eb
* a2092e66c435d1efc3be30d7e45978737df9b8b3
* 2f4eef47e900caa6d770fb0ff1b5be832229cfcb

nun vollumfänglich erschlagen, gehe ich fest davon aus. Auch @Snoopy-HSS hat ja freundlich bestätigt, dass es nun wieder tut auf Windows. Die letzten beiden commits sind neuer als seine Rückmeldung, aber die betreffen Linux und corepack, nicht das grundsätzliche Problem, dass die WebApp nicht gebaut werden konnte.

Außerdem wurde die [Doku](https://opendtu-onbattery.net/firmware/compile_webapp/) nochmal aufwändig aufpoliert (und die Anweisungen getestet), was auch zu Verbesserungen an den Github workflows geführt hat
* 0bcbec1baa1f8a8e823ba8eedd78294f23f4a9bd
* 58e01cdcc7dfe3cf7dd1851bfb3b635cb190c432
* c3c31e56945d7d6ecdc3810850f90ccce41fdd64
* d1aad0b8e058ca48c5b2afb2f6972c93eab344fe
* und andere

welche auch schon allesamt unaufgefordert im upstream von Thomas eingesammelt wurden.

closed 2024-09-21 by schlimmchen