hoylabs/OpenDTU-OnBattery #1256: [Request] MQTT Topic für Lastausgang Ein/Aus

openKimchi, 2024-09-15:
### Describe the solution you'd like

Viele Victron Laderegler haben einen Lastausgang, mit dem man zB. auch ein Relais steuern kann, um beispielsweise eine Batterie mit einem Inverter zu verbinden.
Dieser Lastausgang ist über den RX-Port des Ladegerätes steuerbar:
[image]
Es wäre schön, wenn ein MQTT Topic zu Verfügung gestellt würde, mit dem man den RX-Port steuern könnte.
Dann könnte man für den Lastausgang beispielsweise über ioBroker individuelle Zeitpläne erstellen oder den Lastausgang abschalten, wenn eine bestimmte SOC der Batterie erreicht wäre.
In einem anderen Projekt(Victron2mqtt) ist dieses Feature zB. implementiert.

openKimchi, 2024-09-16:
kannst du [hier](https://github.com/softwarecrash/Victron2MQTT) nicht einfach etwas abgucken...;)

schlimmchen, 2024-09-16:
Es scheint nicht aus meinem niedergeschriebenen Denken klar geworden zu sein: Abgucken muss ich dafür nichts, ich muss "nur" drei Stunden Zeit investieren, um es gescheit zu implementieren und zu testen... Und nochmal eine, um es zu dokumentieren.

Es ist kein Hexenwerk und die Implementierung ist mit meinem Kommentar schon skizziert. Aber es muss dann auch jemand Bock haben, es umzusetzen... Da du bisher der erste bist, der sich meldet das nutzen zu wollen, ist meine Motivation jedenfalls eher klein.

openKimchi, 2024-09-16:
Sorry, es sollte nicht so aussehen, als wenn ich an deinen Programmierkünsten gezweifelt hätte...mein Kommentar war wohl eher etwas dumm. Tut mir leid...
...und wenn ich tatsächlich der Einzige bin, der so ein Feature sinnvoll findet, ergibt es natürlich wenig Sinn das zu implementieren. Es gab aber in den Discussions mal eine ähnliche [Anfrage](https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/250)

schlimmchen, 2024-09-17:
> Sorry, es sollte nicht so aussehen, als wenn ich an deinen Programmierkünsten gezweifelt hätte...

Nene, kein Stress :wink:

Danke für den Link zu der Discussion, dass schonmal jemand danach gefragt hat ist durchaus relevant.

hoschiking, 2024-10-20:
Ich fände dieses Feature, wie wahrscheinich viele andere auch, sehr nützlich. Ich steuere meine 3 MPPT, die an der OpenDTUoB hängen auch. Man kann je nach Config den Lastausgang oder den Laderegler an oder ausschalten. Hierfür muß ich einen extra ESP verwenden und der Aufwand bei der Verdrahtung ist nicht ohne. Dies würde vieles einfacher machen.
Freie GPIO´s für bis zu 3 MPPT werden die meisten haben.
Wenn es einen MQTT Tpoic gibt, so lässt sich dann jeder MPPT beliebig per ioBroker oder Homeassistent steuern.
Eine tolle Sache.

SW-Niko, 2024-11-05:
Nur so zu Info.....
Den Lastausgang kann man auch über HEX-Nachrichten steuern. Vielleicht einfacher zu implementieren?

[image]

Allerdings bin ich mir nicht sicher ob ihr auch wirklich den Lastausgang meint. Am Lastausgang hängt nicht die Batterie!
Ihr wollt den Charger ausschalten damit die Batterie nicht weiter geladen wird, oder?

Das geht auch.
[image]

hoschiking, 2024-11-06:
Ich steuere über den RX Pin den Laderegler und schalte damit das Laden der Batterie aus. Ich denke, so war das auch bin den anderen Nutzern genannt. Was letztendlich über den RX Pin geschaltet werden soll, lässt sich jedoch mittels der victron App konfigurieren. Auch der Lastausgang ist da konfigurierbar. Da hätte man also freie Wahl. Bei der Implementierung mittels Hex wäre der Vorteil futsch, soweit ich es einschätzen kann.