hoylabs/OpenDTU-OnBattery #1257: [BUG] CMT2300A not working when using opendtufusionv2 with W5500 enabled

vaterlangen, 2024-09-15:
### What happened?

When I select the profile `OpenDTU Fusion v2 with CMT2300A and W5500 ethernet` on my OpenDTU Fusion v2 with PoE Shield, the CMT2300A is not detected and no communication to my inverter is possible (Ethernet is working perfectly fine). Seems like OpenDTU-OnBattery assumes, it should use the NRF24 (see screenshots)

When using the profile `OpenDTU Fusion v2 with CMT2300A and NRF24` on the same FW build, CMT3200A is working as expected. 

### To Reproduce Bug

Switch to profile `OpenDTU Fusion v2 with CMT2300A and W5500 ethernet` on a OpenDTU Fusion v2 with PoE Shield

### Expected Behavior

CMT2300A is usable

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

a6e7007

### Anything else?

Device Manager shows CMT2300A enabled 
[image]

DTU Settings show NRF24
[image]

System Info also shows NRF24
[image]

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

skippermeister, 2024-09-15:
@vaterlangen 
Die Chips ESP32 und ESP32S3 habennur 2 freie SPI Ports.
NRF24 + CMt + W5500 funktioniert nicht zusammen.
Wenn Du CMT + W5500 nutzen möchtest, dann musst im Pinmapping die Pins für das NRF24 Modul alle auf -1  setzen. Dann wird das RF Modul nicht initialisiert und der SPI Port wird frei.

vaterlangen, 2024-09-15:
@skippermeister  Danke für den Hinweis - hat geklappt!

Updated mapping for W5500 modes accordingly - see #1259

schlimmchen, 2024-09-16:
Am I assuming correctly that you built for the opendtufusionv2 environment?

vaterlangen, 2024-09-16:
Correct

PR #1259 (2024-09-16): Bugfix/deviceprofile dtufusion with cmt and w5500
Fixed device profile for OpenDTU Fusion v2 to allow usage of CMT2300A with W5500. Also added missing LED mapping and explicitly disabled ETH for W5500 profiles.

Fixes #1257
commit 27f5a943 (2024-09-16): Fix: device profile for OpenDTU Fusion with W5500 (#1259)

* explicitly disable NRF24 when using CMT + W5500. fixes #1257.
* explicitly disable CMT when using NRF + W5500
* added missing LED mapping for W5500 profiles
closed 2024-09-16 by schlimmchen