hoylabs/OpenDTU-OnBattery #1260: [Request] PSRAM aktivieren

Andre0be, 2024-09-15:
### Is your feature request related to a problem? Please describe.

Hallo,

bei einem ESP32-S3 wäre es gut den PSRAM zu aktivieren.

### Describe the solution you'd like

[env:esp32-s3-devkitc-1]
platform = espressif32
framework = arduino
board = esp32-s3-devkitc-1

# Configure options for the N16R8V variant
build_flags= -DBOARD_HAS_PSRAM
board_build.arduino.memory_type = qio_opi 
board_build.partitions = default_16MB.csv
board_upload.flash_size = 16MB

### Describe alternatives you've considered

Dieser Eintrag ist wohl dafür nötig:

board_build.arduino.memory_type = qio_opi

spcqike, 2024-09-16:
Welchen Vorteil bietet PSRAM bisher? 

Er würde als aktiv angezeigt werden, ok. Aber gibt es einen echten Mehrwert im Vergleich zu ohne?

schlimmchen, 2024-09-16:
Ich würde das anbieten wollen, als eine neue Firmware-Variante (auf keinen Fall in den bestehenden Varianten). Kann man schon machen. In der Doku ist es meiner Meinung nach hinreichend beschrieben, dass das Konsequenzen haben kann auf die Verwendung von GPIOs, die auf der Stiftleiste des DevKit verfügbar sind. Darauf würden wir dann garantiert im Wochentakt verlinken müssen... Aber das erscheint mir erträglich.

@spcqike Der konkrete Vorteil spielt weniger eine Rolle. Es ist das "ich hab PSRAM, ich will sehen, dass er erkannt wird und genutzt werden *könnte*". Und das sehe ich ein -- zumindest weil bekannt ist, was zu tun ist, um ihn zu aktivieren.

AndreasBoehm, 2024-09-16:
Sollen wir die anzeige des PSRAM aus dem web ui entfernen? Damit könnten wir PSRAM anfragen eventuell unterbinden.

schlimmchen, 2024-09-16:
Die Idee hatte ich auch schon. Ich bin mir nicht sicher. Wir würden aber Erfahrung sammeln mit PSRAM durch Leute, die die neuen Varianten einsetzen, wenn es sie denn gibt. Zusätzlich können wir die Anzeige aus der UI entfernen, sodass die Leute, die wirklich keinen PSRAM haben, die auch nicht sehen und sich fragen "was fehlt mir denn da".

Außer, dass ggf. Pins am I/O Header des DevKit "verloren" gehen (die auch bei 16MB Flash verloren sind, egal ob PSRAM aktiv ist oder nicht), was schlimmstenfalls Rückfragen auslöst, sehe ich kein Risiko, PSRAM einzuschalten. Ich nehme einfach mal an, dass das tut. Vielleicht tuts auch nicht und wir wissen dann, warum wir es nicht anbieten :wink:

Manos1966, 2024-09-17:
Mir ist der Satz (in der Doku) "You will not be able to leverage 16 MB of flash memory with OpenDTU-OnBattery any time soon", im Gedaechnis geblieben.

Die Nachteile "dass das Konsequenzen haben kann auf die Verwendung von GPIOs, die auf der Stiftleiste des DevKit verfügbar sind " sind auch beschrieben. Gibt es Vorteile?

"Wir würden aber Erfahrung sammeln mit PSRAM durch Leute": Wenn es nur um "wissentliches Interesse" geht, dann spricht natuerlich nichts dagegen PSRAM einzuschalten 😉

schlimmchen, 2024-10-03:
Danke, das ist richtig. Es muss wirklich nur noch aktiviert werden, es gibt auch schon PlatformIO Environments für dieses Projekt, wo PSRAM eingeschaltet ist.

ThomasCr, 2024-10-03:
Hier nochmal die Frage: wenn es aktiviert ist, wird es denn in dem Code von OpenDTU irgendwo angesprochen/verwendet? Was ich auch gestern noch irgendwo gelesen habe, dass der PSRAM nicht so performant sein soll, wie der normale RAM...

schlimmchen, 2024-10-04:
> wenn es aktiviert ist, wird es denn in dem Code von OpenDTU irgendwo angesprochen/verwendet? 

Ja.

https://github.com/helgeerbe/OpenDTU-OnBattery/blob/1812e6eb6a84d5cfda72f3bc7ee6ed3deb33882a/src/main.cpp#L45-L48

Korrekt, PSRAM kann nicht so performant sein wie SRAM.

Solar320, 2024-10-19:
[image]

Nutzen tut er scheinbar etwas davon, ob die Anzeige stimmt kann ich nicht sagen.

closed 2026-05-04 by AndreasBoehm