hoylabs/OpenDTU-OnBattery #1261: HUAWEI funktioniert nicht, wenn NRF und CMT gleichzeitig verwendet werden

woollfi, 2024-09-16:
### What happened?

Huawei Netzteil funktioniert nur wenn ein Chip konfiguriert ist, wenn beide Chips CMT und NRF konfiguriert und aktiv sind wird das Huawei nicht erkannt, keine Werte und Steuerung.
[image]

### To Reproduce Bug

Beide Sendechips aktivieren und Huawei betreiben

### Expected Behavior

Huawei regelt und zeigt Werte an

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.09.11

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

Manos1966, 2024-09-16:
This is not a bug.

"Made by design"

schlimmchen, 2024-09-16:
Da hat Manos den Nagel auf den Kopf getroffen, auch wenn es mir lieber gewesen wäre, er hätte [in die Doku verlinkt](https://opendtu-onbattery.net/hardware/limitations/#spi-busses) statt zu kopieren :wink:

Retten kann dich #999, aber da arbeitet gerade niemand dran.

Du kannst dich aufh eine beliebige Drahtlosschnittstelle beschränken, per pin mapping, und dafür das Huawei anschließen mit dem SPI<->CAN Adapter.

closed 2024-09-16 by schlimmchen